mjg59/python-broadlink #308: RM Mini 3 (0x5f36) Does Not Learn Codes

creemerica, 2020-01-09:
I purchased 2 RM Mini 3s off of Amazon and they both register as 0x5f36, however even after adding them to the rm list they do not learn any codes. I was able to enter learning mode however none of the codes are printed to the terminal window.

I am able to add/learn codes in the various apps but not through this library. Any idea why or does anyone have this specific device and got it working?

Nonobond007, 2020-01-09:
217/5000
Hello I just bought an RM-Mini-3 also on Amazon, broadlink_discovery returns me an rm device: 0x5f36.
I added this device to the __init__.py file, but it still doesn't work.
How can I do?

Thank you so much.

Siytek, 2020-01-28:
> I purchased 2 RM Mini 3s off of Amazon and they both register as 0x5f36, however even after adding them to the rm list they do not learn any codes. I was able to enter learning mode however none of [...]

Exactly the same for me, I just purchased two from eBay, device ID 0x5f36, can setup and discover but learning does not work. Has anyone looked inside these newer units?

Cioriz, 2020-02-07:
I'm having this issue as well in 2, just bought, RM Mini 3 devices (amazon.it and ebay.it).

`Discovering...
###########################################
Unknown
# broadlink_cli --type 0x5f36 --host  --mac
Device file data (to be used with --device @filename in broadlink_cli) : 
0x5f36`

Please dear skilled developers help us out, i can't really miss the local polling on this device!!

elafargue, 2020-02-07:
Are the reports above all for the red version of the RM3, or do some people experience this with the black version as well?

Siytek, 2020-02-07:
> Are the reports above all for the red version of the RM3, or do some people experience this with the black version as well?

Mine's the black version

martinrehfeld, 2020-02-10:
Crosslinking https://github.com/home-assistant/home-assistant/issues/30215 as some active investigation on said device is going on over there.

umarbutler, 2020-02-25:
Hi, any updates on a solution for this issue? I have the same firmware (v44057) and have been unable to connect my RM Mini 3 to HA as well as Broadlink Manager (keep getting the error 'Writing compatible device not detected!'). I have tried configuring the device to connect to Wi-Fi via both the IHC and Broadlink apps (using both AP and Bluetooth modes), to no avail.

felipediel, 2020-04-19:
Fixed by #317.

Snocksman, 2020-06-27:
Hi ! I´m using FHEM and have a Problem with one of the 0x5f36 RM3 mini Devices. The FHEM Module depends on this python script, but the fix for the new RM mini 3 Devices has not been implemented. Could you maybe have a look at the FHEM Module, maybe it´s only little work for you to fix this Module... You can send me a mail if i should send you the Module File: snocksman@t-online.de

felipediel, 2021-02-17:
Fixed with #317. Thank you!

closed 2021-02-17 by felipediel