mjg59/python-broadlink #310: Unstable behaviour - Error Connecting to Broadlink device

XSlaY, 2020-01-26:
Hi everyone,
Plugin is working perfectly...when it woks.
Sometimes domoticz is unable to connect to my BR RM Mini3 and somtimes no problem. I don't where the problem is.  I lose the BR connection generally after a reboot of my RPi. I know that it doesn't come from the BR itself because it keeps working fine with BR apps on my smartphone.

Here are my logs
Any idea of what could be the origin of this behaviour?

2020-01-26 18:10:10.172 (RM_MINI_SALON) Debug logging mask set to: PYTHON PLUGIN QUEUE IMAGE DEVICE CONNECTION MESSAGE ALL
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'HardwareID':'8'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'HomeFolder':'/home/pi/domoticz/plugins/BroadlinkRM2/'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'StartupFolder':'/home/pi/domoticz/'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'UserDataFolder':'/home/pi/domoticz/'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'Database':'/home/pi/domoticz/domoticz.db'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'Language':'fr'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'Version':'4.2.0'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'Author':'zak45'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'Name':'RM_MINI_SALON'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'Address':'192.168.1.51'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'Port':'0'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'Key':'BroadlinkRM2'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'Mode1':'34:ea:34:8f:83:02'
2020-01-26 18:10:10.172 (RM_MINI_SALON) 'Mode2':'/home/pi/domoticz/plugins/BroadlinkRM2'
2020-01-26 18:10:10.173 (RM_MINI_SALON) 'Mode3':'RM2'
2020-01-26 18:10:10.173 (RM_MINI_SALON) 'Mode4':'no'
2020-01-26 18:10:10.173 (RM_MINI_SALON) 'Mode5':'9000'
2020-01-26 18:10:10.173 (RM_MINI_SALON) 'Mode6':'Debug'
2020-01-26 18:10:10.173 (RM_MINI_SALON) 'DomoticzVersion':'4.10717'
2020-01-26 18:10:10.173 (RM_MINI_SALON) 'DomoticzHash':'b38b49e5'
2020-01-26 18:10:10.173 (RM_MINI_SALON) 'DomoticzBuildTime':'2019-05-09 13:04:08'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device count: 4
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device: 1 - ID: 104, Name: 'RM_MINI_SALON - Command', nValue: 0, sValue: 'Off'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device ID: '104'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device Name: 'RM_MINI_SALON - Command'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device nValue: 0
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device sValue: 'Off'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device LastLevel: 10
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device: 2 - ID: 106, Name: 'PAC_Temp21', nValue: 1, sValue: 'On-1'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device ID: '106'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device Name: 'PAC_Temp21'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device nValue: 1
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device sValue: 'On-1'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device LastLevel: 0
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device: 3 - ID: 107, Name: 'PAC_OnOff', nValue: 1, sValue: 'On-4'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device ID: '107'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device Name: 'PAC_OnOff'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device nValue: 1
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device sValue: 'On-4'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device LastLevel: 0
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device: 254 - ID: 105, Name: 'RM_MINI_SALON - Remote', nValue: 0, sValue: ''
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device ID: '105'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device Name: 'RM_MINI_SALON - Remote'
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device nValue: 0
2020-01-26 18:10:10.173 (RM_MINI_SALON) Device sValue: ''
2020-01-26 18:10:10.174 (RM_MINI_SALON) Device LastLevel: 0
2020-01-26 18:10:10.174 (RM_MINI_SALON) Pushing 'PollIntervalDirective' on to queue
2020-01-26 18:10:10.174 (RM_MINI_SALON) Connecting to: 192.168.1.51:34:ea:34:8f:83:02
2020-01-26 18:10:10.177 (RM_MINI_SALON) No ini file :/home/pi/domoticz/plugins/BroadlinkRM2/remote/plugin_remote_8.ini
2020-01-26 18:10:10.177 (RM_MINI_SALON) Custom Commands for Remote not managed
2020-01-26 18:10:10.177 (RM_MINI_SALON) Device Number begin to : 3
2020-01-26 18:10:10.177 (RM_MINI_SALON) Processing 'PollIntervalDirective' message
2020-01-26 18:10:10.177 (RM_MINI_SALON) Heartbeat interval set to: 30.
2020-01-26 18:10:10.168 Status: (RM_MINI_SALON) Entering work loop.
2020-01-26 18:10:10.169 Status: (RM_MINI_SALON) Initialized version 4.2.0, author 'zak45'
**2020-01-26 18:10:10.174 Error: (RM_MINI_SALON) Error Connecting to Broadlink device....192.168.1.51**

Merci !

xAsKeYx, 2020-02-19:
I have the same problem (python 3.7 and last beta version of domoticz)

Regards

dmartella, 2020-04-11:
same problem hefre...

tor-m6, 2020-04-17:
same with RM2 python3.7.7 and stable domoticz

erwanviolle, 2020-05-03:
same problem here with RM2 python3.7.7 and stable domoticz...
I can't find a solution :-(  Please

dxfantome, 2020-07-04:
allway same problem

felipediel, 2021-02-17:
Closing, since it is not a problem to be solved in our code base.

closed 2021-02-17 by felipediel