mjg59/python-broadlink #311: Usage impossible - PyCRC is now a different package

majuss, 2020-02-14:
The original PyCRC (http://pydoc.net/PyCRC/1.21/PyCRC.CRC16/) is not longer available on pypi and was replaced by: https://github.com/tpircher/pycrc.

So currently this software is not usable at all :(

hazbot72, 2020-03-07:
I ran into this same issue while installing on Raspberry PI 4, but was able to work around it by installing an archived copy of PyCRC 1.21 from piwheels:

https://www.piwheels.org/project/PyCRC/

tlc76, 2020-03-24:
I ran into the same issue...
How did you install the file PyCRC-1.21.zip ?
I cannot find any setup.py inside...

Thanks!

tlc76, 2020-03-24:
I managed to install the PyCRC-1.21-py2.py3-none-any.whl file.
All working fine now, thanks!

majuss, 2020-03-24:
We should remove this dependency, otherwise this lib will die!

majuss, 2020-03-25:
https://github.com/mjg59/python-broadlink/pull/327

PR is open :=

closed 2020-04-10 by Danielhiversen