mjg59/python-broadlink #319: TypeError: discover() got an unexpected keyword argument 'discover_ip_address'

MrksHfmn, 2020-03-14:
```
markus@MARKUS-DESKTOP:[cli]> ./broadlink_discovery
Discovering...
Traceback (most recent call last):
  File "./broadlink_discovery", line 14, in <module>
    devices = broadlink.discover(timeout=args.timeout, local_ip_address=args.ip, discover_ip_address=args.dst_ip)
TypeError: discover() got an unexpected keyword argument 'discover_ip_address'
```

MrksHfmn, 2020-03-14:
Unfortunately it stops right after the call.

wijnandtop, 2020-03-22:
Received the same error, apparently pypi doesn't have the latest broadlink library (or other way around the master branch of this project isn't a release). 

If you checkout the latest branch (0.12.0), everything works fine (as far as i can see).

closed 2020-03-23 by Danielhiversen