mjg59/python-broadlink #330: TypeError: adler32() argument 1 must be string or read-only buffer, not bytearray

knobo, 2020-03-30:
When I run `broadlink.setup`
I get

```
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.13.0-py2.7.egg/broadlink/__init__.py", line 998, in setup
    checksum = adler32(payload, 0xbeaf) & 0xffff
TypeError: adler32() argument 1 must be string or read-only buffer, not bytearray
```

hmagrini, 2020-04-01:
In case this helps you, I had the same issue and was able to fix by following the steps on this comment https://github.com/mjg59/python-broadlink/issues/274#issuecomment-517851366
You'll need to also install `mkvirtualenv`, I followed this https://virtualenvwrapper.readthedocs.io/en/latest/

ChristianChoi, 2020-04-01:
@hmagrini I tried follow your advice, but I still get the same error as @knobo:
```
[11:16:23] :~/Projects/python-broadlink$ mkvirtualenv broadlink
...
...
(broadlink) [11:16:45] ~/Projects/python-broadlink$ pip install -r requirements.txt
...
...
(broadlink) [11:16:57] ~/Projects/python-broadlink$ python setup.py install
...
...
(broadlink) [11:17:02] ~/Projects/python-broadlink$ cd cli/
(broadlink) [11:17:07] ~/Projects/python-broadlink/cli$ ./broadlink_discovery 
Discovering...
Traceback (most recent call last):
  File "./broadlink_discovery", line 14, in <module>
    devices = broadlink.discover(timeout=args.timeout, local_ip_address=args.ip, discover_ip_address=args.dst_ip)
  File "/Users/choi/Envs/broadlink/lib/python2.7/site-packages/broadlink-0.13.0-py2.7.egg/broadlink/__init__.py", line 122, in discover
    checksum = adler32(packet, 0xbeaf) & 0xffff
TypeError: adler32() argument 1 must be string or read-only buffer, not bytearray
```

hmagrini, 2020-04-01:
>when you run the cli tools you need to prefix them with python to make sure that the python interpreter from the virtualenv is called.

so for example: `python ./broadlink_discovery --ip 'current_pc_IP' --dst-ip 'broadlink_device_IP'`

I had to explicetly add the IP addresses as well BTW

ChristianChoi, 2020-04-01:
Oh missed that part, however it doesn't change anything for me:
```
(broadlink) [12:05:46] ~/Projects/python-broadlink/cli$ python ./broadlink_discovery --ip '127.0.0.1' --dst-ip '192.168.1.122'
Discovering...
Traceback (most recent call last):
  File "./broadlink_discovery", line 14, in <module>
    devices = broadlink.discover(timeout=args.timeout, local_ip_address=args.ip, discover_ip_address=args.dst_ip)
  File "/Users/choi/Envs/broadlink/lib/python2.7/site-packages/broadlink-0.13.0-py2.7.egg/broadlink/__init__.py", line 122, in discover
    checksum = adler32(packet, 0xbeaf) & 0xffff
TypeError: adler32() argument 1 must be string or read-only buffer, not bytearray
```

Danielhiversen, 2020-04-13:
Tried with python3?

felipediel, 2020-04-19:
Fixed by https://github.com/mjg59/python-broadlink/pull/345.

closed 2020-04-19 by Danielhiversen

knobo, 2020-04-20:
Confirmed :)