mjg59/python-broadlink #335: TypeError: adler32() argument 1 must be string or read-only buffer, not bytearray

jhrdzbanda, 2020-04-08:
I'm trying to use Broadlink Library on PowerShell @Windows as follows, but getting an error on broadlink.setup.

Any Idea, what colud be wrong?

Please Advise.

=========
python
Python 2.7.13 (v2.7.13:a06454b1afa1, Dec 17 2016, 20:53:40) [MSC v.1500 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.

>>> broadlink.setup('InterCable9214649', 'm207355a86ce2', 3)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'broadlink' is not defined
>>> import broadlink
>>> broadlink.setup('InterCable9214649', 'm207355a86ce2', 3)
Traceback (most recent call last):
 File "<stdin>", line 1, in <module>
  File "broadlink\__init__.py", line 998, in setup
    checksum = adler32(payload, 0xbeaf) & 0xffff
TypeError: adler32() argument 1 must be string or read-only buffer, not bytearray
>>> exit()

eschava, 2020-04-08:
Have you tried with Python3?

Danielhiversen, 2020-04-13:
https://github.com/mjg59/python-broadlink/issues/330

closed 2020-04-13 by Danielhiversen