mjg59/python-broadlink #336: compilation broken for python 3.7.7

tor-m6, 2020-04-08:
line 250 and 251
        packet[0x24] = self.devtype & 0xff
        packet[0x25] = self.devtype >> 8

devtype has value like 'RM2' and type which is not allowed for such operations, output of pdb

(Pdb) 
> /usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/broadlink/__init__.py(248)send_packet()
-> packet[0x24] = self.devtype & 0xff
(Pdb) 
TypeError: unsupported operand type(s) for &: 'str' and 'int'

felipediel, 2020-04-21:
@tor-m6 devtype must be an integer, your code is assigning a string

jack747474, 2020-04-29:
Hi,

I solved the problem with the scripts attached.

I'm using the last python broadlink library 0.13.2

The scripts are written for domoticz

[Broadlink by LG.zip](https://github.com/mjg59/python-broadlink/files/4553791/Broadlink.by.LG.zip)

felipediel, 2021-02-17:
Closing, since it is not a problem to be solved in our code base.

closed 2021-02-17 by felipediel