mjg59/python-broadlink #338: Broadlink RM mini3 broken on 0.13

robal, 2020-04-11:
Hello. Updating to latest broadlink module has broken my Domoticz broadlink plugin.
The error is:
> Error Connecting to Broadlink device....

This may be Domoticz plugin specific...  Sorry, I don't have time to investigate this thoroughly.

Works OK on 0.12.0, broken on 0.13.0
Using Broadlink RM mini3

[domoticz.br12.log](https://github.com/mjg59/python-broadlink/files/4464851/domoticz.br12.log)
[domoticz.br12.piplist.txt](https://github.com/mjg59/python-broadlink/files/4464852/domoticz.br12.piplist.txt)
[domoticz.br13.log](https://github.com/mjg59/python-broadlink/files/4464853/domoticz.br13.log)
[domoticz.br13.piplist.txt](https://github.com/mjg59/python-broadlink/files/4464854/domoticz.br13.piplist.txt)

baldfox, 2020-04-13:
I too am struggling with my RM Mini3. I set it up perfectly about a year ago in home-assistant and it's been working without issue. In the past few weeks the commands haven't been going through on it, so I just checked and found this: https://github.com/mlfunston/node-red-contrib-broadlink-control/issues

I am checking my node-red logs, and am receiving the following error:
"Broadlink Timeout Received - Closing Device Connection"

I believe this is all part of the same issue with a recent firmware upgrade from Broadlink to use their cloud (or so i've read).

Danielhiversen, 2020-04-13:
Could you test 0.13 without these changes: https://github.com/mjg59/python-broadlink/commit/8e7446f4108db748e82a68e41b254e80b01bd51b ?
That is the change from 0.12, that I think most likely think causes the problem.

felipediel, 2020-04-13:
@Danielhiversen I can confirm that it was my fault. The new padding algorithm for CBC mode has a bug. I'll fix it tonight. Sorry for the inconvenience.

robal, 2020-04-13:
Many thanks for looking into that.
I'm so sorry I don't have time to test the code.

closed 2020-04-14 by mjg59

renatothome, 2020-04-14:
Hi guys I can see you already identifyed the issue, is this already being fixed in any stable version of 108.x? Or there are any workarouds necessary?

Thank you all!

Regards,
Renato

baldfox, 2020-04-15:
> Hi guys I can see you already identifyed the issue, is this already being fixed in any stable version of 108.x? Or there are any workarouds necessary?
> 
> Thank you all!
> 
> Regards,
> Renato

I'm using the node-red component rather than the home-assistant integration and i'm still able to use anything. I guess they'll fix the HA component soon enough. If you want more details on my solution for now it's here: https://www.thesmarthomebook.com/blog

canedje, 2020-04-16:
I also using NodeRED, but broadlink is (still) not working in 0.108.5.
Working yes or  no depending of the size of the message (short IR message works, long ones not)