mjg59/python-broadlink #343: Broadlink RM4 Pro not able to integrate in 0.108.6 vs

CarpeDiemRo, 2020-04-18:
I have 2 Broadlink RM Pro. One RM2 PRO and an RM4 Pro. RM2 works perfectly just from the first moment when I used it. After I upgraded HA to the latest version 0.108.6, I am not able to integrate RM4 Pro.
These are my settings from configuration.yaml

for the **switch**
```
  - platform: broadlink
    host: 192.168.1.166
    mac: 24:DF:A7:B9:C6:81
    type: rm4_pro
```
for the **sensor**
```
  - platform: broadlink
    host: 192.168.1.166
    mac: 24:DF:A7:B9:C6:81
    type: rm4_pro
    monitored_conditions:
      - 'temperature'
      - 'humidity'
```
Each time when I press Check Config button I receive the following message:

> Blockquote
Invalid config for [sensor.broadlink]: [type] is an invalid option for [sensor.broadlink]. Check: sensor.broadlink->type. (See ?, line ?). 
Invalid config for [switch.broadlink]: value is not allowed for dictionary value @ data['type']. Got 'rm4_pro'. (See ?, line ?). 

Can you help me please with some advises?

I want to mention that RM4 PRO is already Registered in the Broadlink app and has a fix IP.

[Later Edit]
I comment out type line from both blocks of code and I didn't receive any error when I check the configuration. After reboot, the switch is not available and the sensors return the following state
[image]

felipediel, 2020-04-18:
@CarpeDiemRo You can no longer use the custom_components folder to integrate your device. You now need to replace the files in the original folder. Follow [these steps](https://github.com/home-assistant/core/issues/30215#issuecomment-615897896) to get it up and running for now.

felipediel, 2020-04-19:
Fixed by https://github.com/home-assistant/core/pull/32523.

cobch, 2020-04-19:
@felipediel Did the PR get approved yet?

felipediel, 2020-04-19:
@cobch Yes! :tada:

cobch, 2020-04-19:
So great!!! I Will try to install the integration (-:

CarpeDiemRo, 2020-04-19:
> Fixed by [home-assistant/core#32523](https://github.com/home-assistant/core/pull/32523). From now on, your devices will work out of the box.

I suppose this bug fix will be integrated in the next release.

raptor75, 2020-04-21:
Hi. So just to clarify the fix is not release yet right? Because I have the same issue on 0.108.6 and not sure if I did something wrong..

Thanks.

felipediel, 2020-04-21:
@raptor75 It's already merged and will come in the next release.

raptor75, 2020-04-22:
Thanks @felipediel . Updated to 0.108.7 still same error :(

_Invalid config for [sensor.broadlink]: [type] is an invalid option for [sensor.broadlink]. Check: sensor.broadlink->type. (See ?, line ?)._

felipediel, 2020-04-22:
@raptor75 If you are in a hurry: `pip3 install --upgrade git+git://github.com/home-assistant/home-assistant.git@dev`.

raptor75, 2020-04-22:
@felipediel Not really in a hurry, was only mindful if I was doing anything wrong. If it has not been released yet I can wait until it is. Thanks.

raptor75, 2020-04-23:
Just as an update, this is working in 0.109 beta, but it is giving "false%" as a reading for both temperature and humidity.

CarpeDiemRo, 2020-04-23:
I will try to clone dev branch today to test it.

raptor75, 2020-04-23:
@CarpeDiemRo I'm using Home Assistant v0.109.0b0, HassOS v3.13, Supervisor v219, on beta channel

raptor75, 2020-04-23:
@felipediel sensor false% is this a consequence of something else?

also, switch is not working, says "failed to connect to device"

Is this because I already integrated with the broadlink app?

felipediel, 2020-04-23:
@raptor75 Yes. You have to disable the cloud.

raptor75, 2020-04-23:
@felipediel thanks. Ach this is annoying Broadlink. I pay for NabuCasa integration - is there anyone who can confirm that A/C integration with Google still works through HA?

felipediel, 2020-04-23:
@raptor75 The problem is only with the Broadlink cloud. Here is the full story: the current authentication method has a security flaw. Anyone connected to your local network can control your device. It's possible to send codes and it is possible to sniff the codes you send. That's why Broadlink modified the authentication method, replacing it with a more secure method, triangulated by their cloud.

The devices are coming unlocked from the factory with the old authentication method, but once connected to the cloud with your login and password, your device becomes more secure and can only be accessed if authenticated by the cloud. The only solution for now is to remove the device from the cloud in order to be able to control it locally again.

The problem could be solved if we implemented the authentication method of the Broadlink SDK at the library level. I think it's really something to think about in the future, but it's off my agenda for now, as I'm working on other updates and I still have my whole life to play. For now, you need to decide between their cloud and Home Assistant. I hope it will change in the future.

felipediel, 2020-04-23:
:point_up: That's why no one should ever send IR / RF codes for doors and gates using the local authentication method. They can be captured and sent by a hacker who gains access to your internal network.

raptor75, 2020-04-24:
Aha. There go my plans to eventually setup an RM Pro with an RF signal to automatically open the garage gate when I'm in 2 minute range of home....

felipediel, 2020-04-24:
@raptor75 Yeap. But RF is not very safe for gates anyway, the frequency can also be sniffed. If you are concerned about security, you need to invest more (look for systems that use encryption).

CarpeDiemRo, 2020-04-25:
I made the update to 108.9 and, unfortunately the issue is reproducible. Can you verify at your system, too. Maybe the fix was not merged in this release.

raptor75, 2020-04-27:
@CarpeDiemRo @felipediel said it will come out in the next release, most probably meaning 0.109. I'm using the beta and sensor connectivity is working (except for that false% which is due to me having connected my device to the cloud already)

raptor75, 2020-04-27:
@felipediel the gate is a communcal gate of the block of apartments and is already setup with RF I'm afraid

CarpeDiemRo, 2020-04-29:
see [https://github.com/mjg59/python-broadlink/issues/355](url)

raptor75, 2020-04-30:
@CarpeDiemRo it's working for you now?

CarpeDiemRo, 2020-05-01:
@raptor75 Yes it working. More details here [https://github.com/mjg59/python-broadlink/issues/343](url)

CarpeDiemRo, 2020-05-01:
@felipediel 
I think that issue must be closed because everything works as design [https://github.com/mjg59/python-broadlink/issues/355#issuecomment-622326812](url)

closed 2020-05-01 by CarpeDiemRo