mjg59/python-broadlink #346: device.auth()   File "/usr/local/lib/python3.7/dist-packages/broadlink/__init__.py", line 223, in auth     response = self.send_packet(0x65, payload)   File "/usr/local/lib/python3.7/dist-packages/broadlink/__init__.py", line 253, in send_packet     packet[0x24] = self.devtype & 0xff TypeError: unsupported operand type(s) for &: 'str' and 'int'

jack747474, 2020-04-21:
Hi,

I upgraded boradlink library from 0.12.0 to 0.13.0 and up to 0.13.2, but from the 13.x version I always get the error:

 device.auth()
  File "/usr/local/lib/python3.7/dist-packages/broadlink/__init__.py", line 223, in auth
    response = self.send_packet(0x65, payload)
  File "/usr/local/lib/python3.7/dist-packages/broadlink/__init__.py", line 253, in send_packet
    packet[0x24] = self.devtype & 0xff
TypeError: unsupported operand type(s) for &: 'str' and 'int'

I need it because I have a new RM4 version and it doesn't work with the previous versions.

Please, let me know where is the error...

I'm working with python 3.7 and raspberry pi4 with buster lite edition

Thanks a lot,

Jack

felipediel, 2020-04-22:
@jack747474 Something is assigning a string to devtype, which should be an integer.
How are you using this library? e.g. Home Assistant, CLI, custom code...

tor-m6, 2020-04-29:
this is the same as mine question [https://github.com/mjg59/python-broadlink/issues/336](url) . Problem in Domoticz, in plugin (plugin_send.py) they use previous version with such implementation causing silently this error (in domoticz log it just say "returned with error 512")

jack747474, 2020-04-29:
Hi,

I solved the problem with the scripts attached.

I'm using the last python broadlink library 0.13.2

The scripts are written for domoticz

[Broadlink by LG.zip](https://github.com/mjg59/python-broadlink/files/4553825/Broadlink.by.LG.zip)

closed 2020-04-29 by jack747474