mjg59/python-broadlink #347: Error using broadlink_discovery without parameters

blazoncek, 2020-04-21:
When using ./cli/broadlink_discovery I get an error:
Traceback (most recent call last):
  File "./cli/broadlink_discovery", line 14, in <module>
    devices = broadlink.discover(timeout=args.timeout, local_ip_address=args.ip, discover_ip_address=args.dst_ip)
  File "/usr/local/lib/python2.7/dist-packages/broadlink/__init__.py", line 131, in discover
    cs.sendto(packet, (discover_ip_address, 80))
TypeError: coercing to Unicode: need string or buffer, NoneType found

The fix is simple: Add type=str in parser.add_argument method call.

parser.add_argument("--ip", type=str, default=None, help="ip address to use in the discovery")
parser.add_argument("--dst-ip", type=str, default=None, help="destination ip address to use in the discovery")

ameer1234567890, 2020-05-09:
The fix could be #361

blazoncek, 2020-05-09:
> The fix could be #361

It does not solve missing --ip parameter. You should add "type=str" into argument list.

felipediel, 2021-03-14:
When handle a local_ip_address with None value [here](https://github.com/mjg59/python-broadlink/blob/9ff6b2d48e58f005765088cdf3dc5cc553cdb01a/broadlink/device.py#L28-L33), not necessary in this case.

Thanks for reporting the issue. Fixed with https://github.com/mjg59/python-broadlink/pull/361.

closed 2021-03-14 by felipediel