mjg59/python-broadlink #349: File "BlackBeanControl.py", line 168, in <module>     RM3Device.auth()   File "/usr/local/lib/python2.7/dist-packages/broadlink/__init__.py", line 211, in auth     response = self.send_packet(0x65, payload)   File "/usr/local/lib/python2.7/dist-packages/broadlink/__init__.py", line 261, in send_packet     checksum = adler32(payload, 0xbeaf) & 0xffff TypeError: adler32() argument 1 must be string or read-only buffer, not bytearray

marcol79, 2020-04-23:
File "BlackBeanControl.py", line 168, in <module>
    RM3Device.auth()
  File "/usr/local/lib/python2.7/dist-packages/broadlink/__init__.py", line 211, in auth
    response = self.send_packet(0x65, payload)
  File "/usr/local/lib/python2.7/dist-packages/broadlink/__init__.py", line 261, in send_packet
    checksum = adler32(payload, 0xbeaf) & 0xffff
TypeError: adler32() argument 1 must be string or read-only buffer, not bytearray

Any idea?

Danielhiversen, 2020-04-25:
Use the latest version of library.
(And should use puthon 3)

closed 2020-04-25 by Danielhiversen

marcol79, 2020-04-25:
python3 BlackBeanControl.py -c COND_ON
  File "BlackBeanControl.py", line 197
    if (len(SentCommand) <> 8) or (not all(c in string.hexdigits for c in SentCommand)):

marcol79, 2020-04-25:
sorry but i'm not a programmer