mjg59/python-broadlink #351: HTS2 sensor

sagitt, 2020-04-25:
Hi, i don't know if this is the correct section

new rm4 support this cable with umidity\temperature sensor.

you think is possible integrate it?

https://www.amazon.it/gp/product/B081GDH5TM/ref=ox_sc_act_title_1?smid=A39FSSU0WTOOK8&psc=1

hakana, 2020-04-25:
It works with RM4 mini, don't know if the same for RM4.

felipediel, 2020-05-28:
@sagitt Do you have the HTS2 cable? Would you mind helping me test [these changes](https://github.com/mjg59/python-broadlink/pull/374)?

sagitt, 2020-05-28:
> @sagitt Do you have the HTS2 cable? Would you mind helping me test [these changes](https://github.com/mjg59/python-broadlink/pull/374)?

Yes, with rm4 pro

felipediel, 2020-05-28:
@sagitt Thank you!

sagitt, 2020-05-28:
> @sagitt Thank you!

Tell me how test new version

felipediel, 2020-05-28:
I thought you had already tested :smile:
I just made a custom [debug script](https://drive.google.com/file/d/1EWntL_m3oY4gHqoan9DQo1RRXnv6vq2m/view?usp=sharing) to make your life easier. Run it on your computer with:

`python3 debug-sensor.py`

Then send me debug.txt.

sagitt, 2020-05-28:
> I thought you had already tested 😄
> I just made a custom [debug script](https://drive.google.com/file/d/1EWntL_m3oY4gHqoan9DQo1RRXnv6vq2m/view?usp=sharing) to make your life easier. Run it on your computer with:
> 
> `python3 debug-sensor.py`
> 
> Then send me debug.txt.

So only need broadlink with cable on my netwoek, right!? Tomorrow i’ll do it.

felipediel, 2020-05-28:
Yes.

felipediel, 2020-05-29:
You no longer need to test. The code is working. But thanks anyway.

sagitt, 2020-05-29:
> You no longer need to test. The code is working. But thanks anyway.

I’m happy!

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/321. Thank you!

closed 2021-02-17 by felipediel