mjg59/python-broadlink #355: Broadlink RM4 PRO is not able to learn or to send IR Signal

CarpeDiemRo, 2020-04-29:
@felipediel 

**Prerequisites**
1. Hassio updated to 109
2. RM4 Pro has been initialized and has a static IP
3. configuration.yaml
```
switch:
  - platform: broadlink
    host: 192.168.1.166
    mac: '24:DF:A7:B9:C6:81'
    type: rm4_pro
    timeout: 15
    switches:
      bedroom_tv:
        command_on: 'JgAcAB0dHB44HhweGx4cHR06HB0cHhwdHB8bHhwADQUAAAAAAAAAAAAAAAA='
        command_off: 'JgAaABweOR4bHhwdHB4dHRw6HhsdHR0dOTocAA0FAAAAAAAAAAAAAAAAAAA='
```

**Description of the issue**
1. From States tab I'm trying to switch on and of the entity switch.bedroom_tv
_Actual results_
a No led is blinking on RM4 Pro
b. No signal is send
2. From Service tab I'm trying to send broadlink.send command and press the power button of the Remote control
_Actual results_
a. The TV is power on
b. _No signal is received_ persistent notification is displayed
[image]

felipediel, 2020-04-29:
@CarpeDiemRo Did you [disable the cloud](https://github.com/home-assistant/core/issues/30215#issuecomment-581159499)?

CarpeDiemRo, 2020-04-29:
> @CarpeDiemRo Did you [disable the cloud](https://github.com/home-assistant/core/issues/30215#issuecomment-581159499)?

Do you mean Nabu Casa? I don't use it.

felipediel, 2020-04-29:
No, Broadlink cloud. Follow the [instructions](https://github.com/home-assistant/core/issues/30215#issuecomment-581159499) to disable it.

CarpeDiemRo, 2020-05-01:
I can confirm that everything it's working and I will describe the steps:

1. I delete the RM4 Pro from new Broadlink application
2. I reset the RM4 Pro by pressing the reset button 10 seconds
3.  I initialized RM4 Pro by using e-Control application. I tried with new Broadlink app but the integration with HA was unsuccessfully 
4. I put the required code into configuration.yaml defining as switch and sensor like this
```
switch:
  - platform: broadlink
    friendly_name: RM4 PRO
    host: 192.168.1.166
    mac: '24:DF:A7:B9:C6:81'
    type: rm4_pro
    timeout: 15
```
```
sensor:
  - platform: broadlink
    name: Bedroom
    host: 192.168.1.166
    mac: '24:DF:A7:B9:C6:81'
    type: rm4_pro
    monitored_conditions:
      - 'temperature'
      - 'humidity' 
```
5. I restarted the HA
6. I checked broadlink.learn
7. A valid persistent notification was returned

I will close this issue because everything works as design. Congrats @felipediel!

closed 2020-05-01 by CarpeDiemRo

Virtual-Nomad, 2020-05-04:
I apologies if I should be logging this issue in another manner as I am not familiar with GitHub. None the less I was hoping you might be able to help. Currently I am trying to get my RM4 Pro to work with the HA and having issues with;
- RF learn (IR learns and sends no problem at all)
- sensor (temperature and humidity) - although not that concerned/interested in this side at the moment

Current Setup
Home Assistant 0.109.3
Broadlink RM4 Pro
Added to network using python-broadlink - PASS
Added to HA as switch - configuration.yaml
switch:
  - platform: broadlink
    host: 192.168.1.172
    mac: 24:df:a7:b9:c0:5a
    type: rm4_pro
    timeout: 15
    retry: 5
    switches:
      tv_samsung_power:
        friendly_name: "Samsung TV Power"
        command_on: IRSTRING
        command_off: IRSTRING
When I run broadlink.learn 
- Red LED comes on
- I can capture IR without an issue as I did with the Samsung TV Power
- UNABLE TO CAPTURE ANY RF 

As far as the sensor is concerned
sensor:
  - platform: broadlink
    host: 192.168.1.172
    mac: 24:df:a7:b9:c0:5a
    type: rm4_pro
    monitored_conditions:
      - temperature
      - humidity

Both Temperature and Humidity just show a value of 0

Do you know if anyone has been successful with the RM4 Pro and RF?