mjg59/python-broadlink #358: Broadlink RM4 PRO is not able to learn or to send RF Signal

Virtual-Nomad, 2020-05-04:
Unsure if this is the right place to raise this issue, however, I cam across https://github.com/mjg59/python-broadlink/issues/355 which relates to a similar issue except with IR. Currently, I am trying to get my RM4 Pro to work with the HA and having issues bring able to learn RF (IR learns and sends no problem at all) with both python-broadlink and Home assistant. (I also have an issue with it as a sensor (temperature and humidity) - although not that concerned/interested in this side at the moment)

Current Setup
Home Assistant 0.109.3
Broadlink RM4 Pro
Added to network using python-broadlink - PASS
when I attempt 
devices[0].host[0] > ‘192.168.1.172’ - PASS
devices[0].get_type() > ‘Unknown’ - Cant get the type (so this may be a concern or compatibility issue)
devices[0].enter_learning() > AttributeError: device instance has no attribute ‘enter_learning’ _FAIL

Under Home assistant 
Added to HA as switch - configuration.yaml
switch:
 platform: broadlink
 host: 192.168.1.172
 mac: 24:df:a7:b9:c0:5a
 type: rm4_pro
 timeout: 15
 retry: 5
 switches:
   tv_samsung_power:
   friendly_name: "Samsung TV Power"
     command_on: IRSTRING
     command_off: IRSTRING
When I run broadlink.learn
- Red LED comes on
- I can capture IR without an issue as I did with the Samsung TV Power
- **UNABLE TO CAPTURE ANY RF**

As far as the sensor is concerned
sensor:
 platform: broadlink 
 host: 192.168.1.172
 mac: 24:df:a7:b9:c0:5a
 type: rm4_pro
 monitored_conditions:
    temperature
    humidity
Both Temperature and Humidity just show a value of 0

Do you know if anyone has been successful with the RM4 Pro and RF?

Another point is that I cant get the RM4 Pro to work with e-control (or broadlink manager) so have not been able to get any RF packets to test it broadlink.send command to see if that works.

felipediel, 2020-05-04:
@Virtual-Nomad The RF functionality has not been fully tested because I don't have the device. Would you like to help me by making some captures in Wireshark so I can add this functionality?

Virtual-Nomad, 2020-05-04:
@felipediel more than happy to. Just be aware I am from the windows world and only now trying to get up to speed with all things python, home assistant. But I am familiar with wireshark (in windows world) so with a little guidance would be able to help out.

felipediel, 2020-05-05:
@Virtual-Nomad Do you have an iOS device?

Virtual-Nomad, 2020-05-06:
Yes mate (although I can arrange an android if required). Is there an alternate communications channel you wish to use for all testing results/process etc.  I am in Brisbane, Australia so time zone wise it is UTC +10 with evenings and weekends being the times I can devote to any testing.

felipediel, 2020-05-06:
@Virtual-Nomad Great!
1. You need to run [this program](https://github.com/gh2o/rvi_capture) on Windows and make some captures while controlling the device with the official app on an iOS device. Execute only one type of command per capture and name the file with the name of the command (e.g. learn_rf).
2. Run [this script](https://drive.google.com/open?id=1Gr-Qe_sdVL-Yn2g2ITDFYB1TTpd1pX8Z) using `python3 debug.py`. A file named debug.txt will be created.
3. Send me the files.

Virtual-Nomad, 2020-05-06:
So currently setup 
Windows 10
Installed Wireshark (Windows)
installed itunes
Installed python3
Downloaded  rvi_capture.py
Downloaded debug.py
- Ensured applemobiledeviceService.exe was running

First run of "rvi_capture.py" 
- On first execute it stated downloading libimobiledevice
Second test run rvi_capture.py - | "c:\Program Files\Wireshark\Wireshark.exe" -k -i -
- Opens wireshark in a capture mode so looking good (just need to get home to actually test)

when I run debug.py however I am getting this error ... do I need to download additional components?
C:\Temp\rvi_capture-master>python3 debug.py
Traceback (most recent call last):
  File "debug.py", line 40, in <module>
    from cryptography.hazmat.backends import default_backend
ModuleNotFoundError: No module named 'cryptography'

Virtual-Nomad, 2020-05-06:
[image]
So as per the above (with the exception of debug) I think I have everything running. 
Connected the iphone to the PC via the USB
Installed the Broadlink App
- Added the RM4_Pro into via AP mode
- Added an appliance using the UserDefined, set the room to Office, Named the Device: Office Fan
- Added a function button > Start Learning 
Laptop cmd - Ran the command "rvi_capture.py - | "c:\Program Files\Wireshark\Wireshark.exe" -k -i -" to start capturing packets
Broadlink App
- Started learning (I have a physical remote)  - LED on RM4 Showed Up 
- Long Pressed the Low Fan button on the RF remote - LED turned off (fan turned on low)
- Broadlink App prompted for send short press - LED on RM4 Showed Up 
- Short Press the Low Fan - LED turned off
NB: As per the screen shot wireshark shows No Packets captured 
Laptop - Exited wireshark
Laptop command - python3 debug.py
- Traceback (most recent call last):
-   File "debug.py", line 40, in <module>
-     from cryptography.hazmat.backends import default_backend
- ModuleNotFoundError: No module named 'cryptography'

No files created (although there is an empty  __pycache__ sub directory) 

I assume I am missing something basic (NB I have also removed webroot incase it was an issue)

Virtual-Nomad, 2020-05-06:
@felipediel This may not actually achieve what it is you are after, however, given I wasn't really sure how the requested approach worked. I setup a hotspot AP on my laptop and connected both the iphone and RM4 Pro to the Wifi.
I then used wireshark to record all traffic across the hotspot wifi during the process of learning an RF signal. It will probably have a lot of there traffic captured but it may hold what you are after. in this capture the iphone is 192.168.137.152 and the RM4 is  192.168.137.126
[RFOfficeFan-LearnLowSpeed.zip](https://github.com/mjg59/python-broadlink/files/4586201/RFOfficeFan-LearnLowSpeed.zip)

Virtual-Nomad, 2020-05-06:
Side note ... I just noticed rvi_capture states "NOTE: There has been a major protocol change in iOS 13, so this will only work for iOS 12 and earlier." my iphone is IOS 13. Not sure if that has factored.

felipediel, 2020-05-07:
@Virtual-Nomad Pure Wireshark does not solve. If you did not succeed with rvictl you can install the Broadlink application on an Android emulator and make the captures with Wireshark. I recommend [MEmu emulator](https://www.memuplay.com/). You may find [this comment](https://github.com/home-assistant/core/issues/30215#issuecomment-593896024) useful.

To solve the cryptography error just type:
`pip install cryptography`

Virtual-Nomad, 2020-05-07:
@felipediel wasn't able to install MEmu as just get a run time error. Anyway found an older iphone with IOS12 so managed to revert back to RVICTL. so I have attached both learning and sending captures. 
Not sure about the debug.txt but I have included it
I am correct you simply run python debup.py once you have finished the capture? 
[RFOfficeFan-LearningOff.zip](https://github.com/mjg59/python-broadlink/files/4592188/RFOfficeFan-LearningOff.zip)
[RFOfficeFan-SendingOff.zip](https://github.com/mjg59/python-broadlink/files/4592189/RFOfficeFan-SendingOff.zip)

felipediel, 2020-05-07:
@Virtual-Nomad Great! The captures are good. Now I need you to [disable the cloud](https://github.com/home-assistant/core/issues/30215#issuecomment-581159499), run debug.py again and send me debug.txt.

Virtual-Nomad, 2020-05-07:
Not sure if I am clear what I need to do however;
- I have removed the device from the app (device went into discover mode)
- Ran Debug.py - 
[debug-PostRemovalfromBroadlinkApp.txt](https://github.com/mjg59/python-broadlink/files/4596241/debug-PostRemovalfromBroadlinkApp.txt)

Re-established direct connection/setup using python-broadlink 
>>> import broadlink
>>> broadlink.setup('','',3)
>>> devices = broadlink.discover(timeout=5)
>>> devices[0].auth()
True
>>> devices[0].host
('192.168.137.220', 80)
- Ran Debug.py - 
[debug-post-Python-BoradlinkImport.txt](https://github.com/mjg59/python-broadlink/files/4596256/debug-post-Python-BoradlinkImport.txt)

I did go to the disable the cloud link above but wasn't really sure what was required. So hoping I read between the lines correctly

felipediel, 2020-05-08:
Sorry, I was unable to decrypt the content of the messages because your device was connected to the cloud during the captures. You need to block the device from accessing the internet using a firewall. Here are the [instructions](https://github.com/home-assistant/core/issues/30215#issuecomment-625566467) from someone who has achieved this before.

Virtual-Nomad, 2020-05-08:
@felipediel just so I am clear should I be using the python-broadlink integration to simply establish a connection to the device and run the debug.py. I guess follow what i did above but simply be disconnected from the internet during this the tests?

felipediel, 2020-05-08:
@Virtual-Nomad The debug serves to intercept your local key. It worked, you don't need to run it anymore.

The problem is with the captures. You need to block the IP address of the Broadlink device using a firewall (your router probably has a built-in firewall). If the device connects to the internet all captures will be lost because they will be encrypted with a key that is in the cloud and I have no access to it. Don't block the phone or the laptop, they need access to the internet.

Do this and capture the commands again (yes, sorry about that).

Virtual-Nomad, 2020-05-08:
@felipediel No problem although will be later tonight

Virtual-Nomad, 2020-05-08:
@felipediel Ok so a new set of results for you
Setup 
Iphone ios12 with broadlink app with RM4 Pro added as a device 
- Connected to WIfi with Internet access
RM4 Pro Connected to WfFi 
- Blocked from internet access 
- (unsure how to be 100% positive but as a test added my laptop and iphone to the same rule and they were blocked so assume it would be)

Did a number of Wireshark Captures for
- Sending Off command x 3
- Learning wIth the LongPress phase of the add function
- Learning with the LongPress + short press phases of the add function
- Learning with the LongPress + short press and Send phases of the add function

[RM4ProInternetBlock-RFOfficeFan.zip](https://github.com/mjg59/python-broadlink/files/4598015/RM4ProInternetBlock-RFOfficeFan.zip)

felipediel, 2020-05-08:
@Virtual-Nomad It didn't work, your device is still connected to the cloud. I'll teach you how to check if it worked.

1. Open the captures in Wireshark.
2. In filter, type `ip.addr == 192.168.1.172`.
3. Find a packet that has `07` in the sixth row of the first column (beside `0050`).
4. Check the last byte of that packet. It needs to be `00`.
5. If the last byte is `01`, it didn’t work. You need to stop everything and start over.

felipediel, 2020-05-08:
Try blocking your phone with the firewall.

Virtual-Nomad, 2020-05-08:
I'll get another router that allows me to view packets/traffic and has better capabilities than my current home internet router. 
Any chance you can send me a screenshot of what you mean in the above? I am not sure if my columns are possibly different. Ie my 6th column is Length and I can't seem to find any 0050 (See screenshot)
[image]

For the heck of it I have taken another record and send with the phone blocked as well. So you never know that may work the way you need. 

[RM4Pro+PhoneInternetBlock.zip](https://github.com/mjg59/python-broadlink/files/4601551/RM4Pro%2BPhoneInternetBlock.zip)

Thanks for the patients.

felipediel, 2020-05-08:
[image]

Virtual-Nomad, 2020-05-08:
So I am a little confused. I have just turned my internet router off completely and done a capture, however as per the screenshot it still show 01 at the end of the packet
[image]

I have attached the capture anyway.
[RM4Pro+PhoneInternetBlock-RFBedroomFan-LearnOff-LongPress+ShortPress+Send2.zip](https://github.com/mjg59/python-broadlink/files/4601993/RM4Pro%2BPhoneInternetBlock-RFBedroomFan-LearnOff-LongPress%2BShortPress%2BSend2.zip)

Any thoughts

Virtual-Nomad, 2020-05-08:
SO not sure where I am going wrong. I disabeld the internet router all together but still get the following 

[image]
[RM4Pro+PhoneInternetBlock-RFBedroomFan-LearnOff-LongPress+ShortPress2.zip](https://github.com/mjg59/python-broadlink/files/4602045/RM4Pro%2BPhoneInternetBlock-RFBedroomFan-LearnOff-LongPress%2BShortPress2.zip)

I have done another capture in that state just the same

felipediel, 2020-05-11:
Hey @Virtual-Nomad, good news for you.
My code is working fine for the RF functions, I don't need to torture you anymore.

After [this update](https://github.com/mjg59/python-broadlink/pull/365) you will be able to learn RF from your RM4 Pro using the CLI. Soon I will take this changes to Home Assistant, but we still have other updates in the queue, so don't hold your breath.

felipediel, 2020-05-11:
What worries me is the temperature and humidity. Is it showing zero? Are you using the HTS2 cable?

Virtual-Nomad, 2020-05-11:
@felipediel sounds good ... How do I keep tabs on when it has been included in a beat or release version?

Virtual-Nomad, 2020-05-11:
> What worries me is the temperature and humidity. Is it showing zero? Are you using the HTS2 cable?

Ok so that would be the issue then. I don't and assumed it was part of the RM4 Pro. Thanks

felipediel, 2020-05-11:
> @felipediel sounds good ... How do I keep tabs on when it has been included in a beat or release version?

I will keep you informed here.

Lecturer555, 2020-05-16:
Felipediel, Virtual-Nomad Hello!
 I am also really looking forward to solving this problem with RF. 
Thank you for not leaving this unattended! Waiting for an update at Home Assistents

com-xuonghuynh, 2020-05-20:
Hope this will work soon!

Virtual-Nomad, 2020-05-24:
@felipediel Any chance there may be an update on when we may see this built into a beta or full release I would love to test it as I need to make a call if I continue with the broadlink or move to something else. Thanks in advance.

felipediel, 2020-05-24:
We are in the middle of a refactor with 4 updates in the queue. New features are blocked.
I also need this. Once everything is ready, I will add it, but as I said, it will take a while.

Virtual-Nomad, 2020-05-24:
@felipediel Thanks for the response ... without much experience in code dev is it likely to be weeks or months? I certainly understand you will have priorities so I am not wishing to pressure you.

Thanks as always your skills and efforts are appreciated

com-xuonghuynh, 2020-05-25:
@felipediel Sorry, can it work with broadlink_cli yet ? I just try to pull the new code then try to catch from broadlink_cli with my RF remote but it not catch anything.

felipediel, 2020-05-25:
@Virtual-Nomad Sorry, I can't give you a deadline because there are steps that don't depend on me. But I promise to dedicate all my free time to these updates.

@com-xuonghuynh Are you sure? Did you use `--rfscanlearn`?

com-xuonghuynh, 2020-05-25:
@felipediel I just did it. Press and hold the remote button until the orange light gone but "RF Frequency not found". I can add it by using Broadlink App before.

perfektor2007, 2020-06-05:
@com-xuonghuynh out of curiosity, what is your device type and what firmware are you running (from the broadlink app)?

I tried to figure rfscanlearn for days without success and finally gave up and tried adding the RM4 pro back to the broadlink app. Tried learning RF within the app but also failed there☹️.

At the end I stumbled upon few posts from the community within the app that rf learning is broken in the last updates...
Maybe you can check it out as well.

My device type is 0x61a2 and broadlink app states firmware version v52076.

Cheers

Virtual-Nomad, 2020-06-10:
@felipediel just checking in to see if it has now been incorporated. Or is there a way I can pull down your fix separately?

felipediel, 2020-06-12:
@Virtual-Nomad This will take time. It is not as simple as it looks. This feature requires modifying the database structure. We are in the middle of a refactoring. I can't touch this right now.

You can use the CLI for now.

ryanerwin, 2020-06-16:
> You can use the CLI for now

@felipediel How do use the CLI for now to learn RF signals with the RM4 pro?
Do I need to be running the latest git checkout or is the 0.14.0 latest pip3 release sufficient?

I've tried `broadlink_cli --learn` and I get: `Learning... No data received...`
I've tried `broadlink_cli --rfscanlearn` and I get: `RF Frequency not found`

Here are background details on my systems
<summary>broadlink_discovery
<details>

```
Discovering...
###########################################
RM4
# broadlink_cli --type 0x6026 --host 192.168.1.51 --mac 6c79e4a7df24
Device file data (to be used with --device @filename in broadlink_cli) : 
0x6026 192.168.1.51 6c79e4a7df24
temperature = 30.38
```
</details>
</summary>

<summary>pip3 show broadlink
<details>

```
Version: 0.14.0
Summary: Python API for controlling Broadlink IR controllers
Home-page: http://github.com/mjg59/python-broadlink
Author: Matthew Garrett
Author-email: mjg59@srcf.ucam.org
License: UNKNOWN
Location: /usr/local/lib/python3.8/dist-packages
Requires: cryptography
Required-by: 
```
</details>
</summary>

<summary>neofetch
<details>
<pre> <font color="#3971ED">❯</font> <font color="#198844">neofetch</font>
<font color="#CC342B"><b>            .-/+oossssoo+/-.</b></font>               <font color="#CC342B"><b>ryan</b></font>@<font color="#CC342B"><b>ri9</b></font> 
<font color="#CC342B"><b>        `:+ssssssssssssssssss+:`</b></font>           -------- 
<font color="#CC342B"><b>      -+ssssssssssssssssssyyssss+-</b></font>         <font color="#CC342B"><b>OS</b></font>: Ubuntu 20.04 LTS x86_64 
<font color="#CC342B"><b>    .ossssssssssssssssss</b></font><b>dMMMNy</b><font color="#CC342B"><b>sssso.</b></font>       <font color="#CC342B"><b>Kernel</b></font>: 5.4.0-37-generic 
<font color="#CC342B"><b>   /sssssssssss</b></font><b>hdmmNNmmyNMMMMh</b><font color="#CC342B"><b>ssssss/</b></font>      <font color="#CC342B"><b>Uptime</b></font>: 1 hour, 50 mins 
<font color="#CC342B"><b>  +sssssssss</b></font><b>hm</b><font color="#CC342B"><b>yd</b></font><b>MMMMMMMNddddy</b><font color="#CC342B"><b>ssssssss+</b></font>     <font color="#CC342B"><b>Packages</b></font>: 4642 (dpkg), 7 (snap) 
<font color="#CC342B"><b> /ssssssss</b></font><b>hNMMM</b><font color="#CC342B"><b>yh</b></font><b>hyyyyhmNMMMNh</b><font color="#CC342B"><b>ssssssss/</b></font>    <font color="#CC342B"><b>Shell</b></font>: zsh 5.8 
<font color="#CC342B"><b>.ssssssss</b></font><b>dMMMNh</b><font color="#CC342B"><b>ssssssssss</b></font><b>hNMMMd</b><font color="#CC342B"><b>ssssssss.</b></font>   <font color="#CC342B"><b>Resolution</b></font>: 5120x1440 
<font color="#CC342B"><b>+ssss</b></font><b>hhhyNMMNy</b><font color="#CC342B"><b>ssssssssssss</b></font><b>yNMMMy</b><font color="#CC342B"><b>sssssss+</b></font>   <font color="#CC342B"><b>DE</b></font>: GNOME 
<font color="#CC342B"><b>oss</b></font><b>yNMMMNyMMh</b><font color="#CC342B"><b>ssssssssssssss</b></font><b>hmmmh</b><font color="#CC342B"><b>ssssssso</b></font>   <font color="#CC342B"><b>WM</b></font>: Mutter 
<font color="#CC342B"><b>oss</b></font><b>yNMMMNyMMh</b><font color="#CC342B"><b>sssssssssssssshmmmhssssssso</b></font>   <font color="#CC342B"><b>WM Theme</b></font>: Adwaita 
<font color="#CC342B"><b>+ssss</b></font><b>hhhyNMMNy</b><font color="#CC342B"><b>ssssssssssss</b></font><b>yNMMMy</b><font color="#CC342B"><b>sssssss+</b></font>   <font color="#CC342B"><b>Theme</b></font>: Yaru-light [GTK2/3] 
<font color="#CC342B"><b>.ssssssss</b></font><b>dMMMNh</b><font color="#CC342B"><b>ssssssssss</b></font><b>hNMMMd</b><font color="#CC342B"><b>ssssssss.</b></font>   <font color="#CC342B"><b>Icons</b></font>: Papirus-Dark [GTK2/3] 
<font color="#CC342B"><b> /ssssssss</b></font><b>hNMMM</b><font color="#CC342B"><b>yh</b></font><b>hyyyyhdNMMMNh</b><font color="#CC342B"><b>ssssssss/</b></font>    <font color="#CC342B"><b>Terminal</b></font>: gnome-terminal 
<font color="#CC342B"><b>  +sssssssss</b></font><b>dm</b><font color="#CC342B"><b>yd</b></font><b>MMMMMMMMddddy</b><font color="#CC342B"><b>ssssssss+</b></font>     <font color="#CC342B"><b>CPU</b></font>: Intel i9-9900K (16) @ 5.000GHz 
<font color="#CC342B"><b>   /sssssssssss</b></font><b>hdmNNNNmyNMMMMh</b><font color="#CC342B"><b>ssssss/</b></font>      <font color="#CC342B"><b>GPU</b></font>: AMD ATI Radeon RX 470/480/570/5 
<font color="#CC342B"><b>    .ossssssssssssssssss</b></font><b>dMMMNy</b><font color="#CC342B"><b>sssso.</b></font>       <font color="#CC342B"><b>Memory</b></font>: 5771MiB / 31842MiB 
<font color="#CC342B"><b>      -+sssssssssssssssss</b></font><b>yyy</b><font color="#CC342B"><b>ssss+-</b></font>
<font color="#CC342B"><b>        `:+ssssssssssssssssss+:`</b></font>           <span style="background-color:#1D1F21"><font color="#1D1F21">   </font></span><span style="background-color:#CC342B"><font color="#CC342B">   </font></span><span style="background-color:#198844"><font color="#198844">   </font></span><span style="background-color:#FBA922"><font color="#FBA922">   </font></span><span style="background-color:#3971ED"><font color="#3971ED">   </font></span><span style="background-color:#A36AC7"><font color="#A36AC7">   </font></span><span style="background-color:#3971ED"><font color="#3971ED">   </font></span><span style="background-color:#C5C8C6"><font color="#C5C8C6">   </font></span>
<font color="#CC342B"><b>            .-/+oossssoo+/-.</b></font>
</pre>
</details>
</summary>

felipediel, 2020-06-16:
@ryanerwin I think there is a problem with the latest firmware on these devices. Some users are complaining that the RF doesn't work even with the official app. If confirmed, you have wait for the next firmware update.

ryanerwin, 2020-06-16:
> Some users are complaining that the RF doesn't work even with the official app

I had been avoiding the official app until now, but was concerned that perhaps my devices were not compatible all all... So I connected the official (Android app) to my RM4 pro and I'm able to learn most of the RF signals that I'm concerned about via the official app.

I can now confirm, with the latest firmware (`v52076`) and the [official Android app](https://play.google.com/store/apps/details?id=cn.com.broadlink.econtrol.international) (`v1.3.31`), the RF function does work on the RM4 Pro.

Virtual-Nomad, 2020-07-03:
@felipediel ... that annoying Australian is back just checking up if there has been any movement on this? I will start fitting out the new house in 2 weeks, so probably at the point at which I need to make a call if I will wait for the inclusion of your fix, or changing platforms. What are your thoughts? Should i move to a different HA rf bridge solution?

ryanerwin, 2020-07-03:
@Virtual-Nomad Same question....

> What are your thoughts? Should i move to a different HA RF bridge solution?

Sure wish Broadlink would actually help support @felipediel ... I messaged them to ask what sort of developer support they offer, but they want to hear from the actual project owner, not some random user

felipediel, 2020-07-03:
@Virtual-Nomad We are in the middle of a [refactor](https://github.com/home-assistant/core/pull/36914). Then we have two updates in the queue. And then I can start working on this feature. It won't be ready in two weeks.

You can use `--rfscanlearn` as a temporary solution to obtain RF codes:

```bash
$ ./broadlink_cli --rfscanlearn --device "0x6026 192.168.1.172 24dfa7b9c05a"
Learning RF Frequency, press and hold the button to learn...
```

Press and hold a button on the remote.

```txt
Found RF Frequency - 1 of 2!
You can now let go of the button
Press enter to continue...
```

Press enter.

```txt
To complete learning, single press the button you want to learn
```

Short press the button and you get the code:

```txt
Found RF Frequency - 2 of 2!
b2002c0221000017b1121101121112112211022101221121211011222110110000000
Base64: b'sgAsAhEREhESEQIEhEBEhESERISERIRARIiEQESERISERESEhEAAF3AAAAAAAAA=='
```

Now you can set up a switch with Home Assistant:
```yaml
switch:
  - platform: broadlink
    host: 192.168.1.172
    mac: 24:df:a7:b9:c0:5a
    switches:
        name_of_the_switch:
            friendly_name: "LG Tv Power"
            command_on: sgAsAhEREhESEQIEhEBEhESERISERIRARIiEQESERISERESEhEAAF3AAAAAAAAA==
            command_off: sgAsAhEREhESEQIEhEBEhESERISERIRARIiEQESERISERESEhEAAF3AAAAAAAAA==
```

You can also use broadlink.send:
```yaml
script:
  name_of_the_script:
    sequence:
      - service: broadlink.send
        data:
          host: 192.168.1.172
          packet: sgAsAhEREhESEQIEhEBEhESERISERIRARIiEQESERISERESEhEAAF3AAAAAAAAA==

Virtual-Nomad, 2020-07-04:
@felipediel thanks for the above .... works a treat so I can go ahead as planned. Really appreciate all the help

burakcankose, 2020-07-08:
Hi !

I tried this but discovery OK! but learn have error.

Traceback (most recent call last):
  File "./broadlink_cli", line 8, in <module>
    import broadlink
ModuleNotFoundError: No module named 'broadlink'

felipediel, 2020-07-08:
@burakcankose Try this:
```bash
$ python3 broadlink_cli --rfscanlearn --device "0x6026 192.168.1.172 24dfa7b9c05a"
```
If it does not work you need to install the module:
```bash
$ pip3 install broadlink
```

burakcankose, 2020-07-08:
pip install broadlink
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: broadlink in /usr/local/lib/python2.7/dist-packages (0.14.0)
Requirement already satisfied: cryptography>=2.1.1 in /usr/lib/python2.7/dist-packages (from broadlink) (2.6.1)

felipediel, 2020-07-08:
pip3

Seeker100, 2020-07-13:
Hi,
First time to this forum.

I have the RM4 Pro and Veraplus hub and have the same issue.
I have downloaded on the 9th July the broadlink_cli app and the broadlink module. 
So I can learn IR codes fine but nothing on the RF. The broadlink has the orange learning light come but times out with nothing picked up.

**IR**
./broadlink_cli --type 0x6026 --host redacted --mac 24dfa7d32c05 --learn
Learning…
260c1c39371c1c1c1d….......c1c1c391c1c1c1c381b1d391c1c000d05000000000000
Base64: JgBgABsdH….......BwcOBwcHBw5HBwcHDgbHTkcHAANBQAAAAAAAA==

**RF**
./broadlink_cli --type 0x6026 --host redacted --mac 24dfa7d32c05 --rfscanlearn
Learning RF Frequency, press and hold the button to learn...
RF Frequency not found

From the previous few posts it looks to work, so I wonder what I need to do the same?

Seeker100, 2020-07-20:
Apologies, did not pick up on the link as it stated homebridge. 
I have the veraplus hub and currently run broadlink_cli on my mac, so to get a working RF learning part I need to install the homebridge on my mac?

kiwi-cam, 2020-07-20:
@Seeker100 No, the Homebridge link is just referencing using this Python module to workaround for learning not working in the Homebridge module.

Seeker100, 2020-07-20:
Ah okay, but not sure if I have confused something here. From my first post (see above) I have downloaded on the the 9th July the broadlink_cli and Python module from https://github.com/mjg59/python-broadlink> The RF does not work for me (IR does) and that was what I was trying to get some clarity on how to fix?

felipediel, 2020-07-20:
@Seeker100 Did you hold the button?

Seeker100, 2020-07-21:
if you meant on the remote, yes I hold it down till the timeout (about 30 secs). I have just tried with two different RF remotes (one of them being the amazon firestick remote and the other a roller blind remote). The IR learning is still working.

felipediel, 2020-07-21:
You can only control 315 and 433MHz devices. Amazon Firestick is 2.4GHz.

Seeker100, 2020-07-21:
Ah, okay but the main roller blind remote is 433Mhz and I have managed to learn the RF through the broadlink app (I've been told by broadlink that the e-control and IHC app do not work with the RM4?)

felipediel, 2020-07-21:
I have no idea, but it should work, since it is working for other users with the same device. Have you tried with another remote?

[64 more replies]

closed 2021-02-17 by felipediel