mjg59/python-broadlink #366: RM Pro + causes exception on sweep_frequency()

KhArtNJava, 2020-05-11:
I just downloaded these scripts and trying to understand, how use it with Broadlink RM Pro+.

Now my Python script is:
```
import broadlink
devices = broadlink.discover(timeout=5)
print(devices)
devices[0].auth()
devices[0].enter_learning()
devices[0].sweep_frequency()
```
This script causes exception 

c:\Python_3.7>python br.py
[<broadlink.rm object at 0x000001CEF5EBFE08>]
Traceback (most recent call last):
  File "br.py", line 11, in <module>
    devices[0].sweep_frequency()
  File "c:\Python_3.7\lib\site-packages\broadlink\__init__.py", line 621, in sweep_frequency
    check_error(response[0x22:0x24])
  File "c:\Python_3.7\lib\site-packages\broadlink\exceptions.py", line 97, in check_error
    raise exception(error_code)
broadlink.exceptions.UnknownError: Unknown error: 0xc

P. S. I've already successfully sent 
```
import broadlink
broadlink.setup('Art', '********', 3)
```
Broadlink RM Pro+ configured and connected to Art Wi-Fi and 
`devices = broadlink.discover(timeout=5)` founds it.

File with packages from WireShark is attached to this issue.
[broadlinkSweep.zip](https://github.com/mjg59/python-broadlink/files/4611570/broadlinkSweep.zip)

How to fix this issue?

closed 2020-05-11 by KhArtNJava

GF-Huang, 2021-01-13:
@KhArtNJava Hey guys how did you fix it?

[image]

KTibow, 2021-01-13:
@GF-Huang just some pre-steps: install the latest version of python (3.9) and python-broadlink (0.16.0).

GF-Huang, 2021-01-13:
> @GF-Huang just some pre-steps: install the latest version of python (3.9) and python-broadlink (0.16.0).

Why not python 3.7?

KTibow, 2021-01-13:
Just in case. Also check the dependencies.

GF-Huang, 2021-01-13:
I'm using the latest version of python-broadlink.

[image]

GF-Huang, 2021-01-13:
And my device is RM Pro Plus