mjg59/python-broadlink #367: Daikin Airconditioner

Spanishu, 2020-05-13:
Hi,

I have a broadlink rm pro 3. Works like a charm with Mitsubishi air-conditioner. However, I am having difficulties with my Daikin Air Conditioner. I was so far only able to switch it on. No temperature change or turning it off. 
Has anybody experience? When I use the original remote for the off button it sounds like it is a longer pulse. (the beep is longer)
Thanks for the help!

marc-invalid, 2020-07-24:
It works for me.

I know this is a stupid answer but maybe it helps to know that some Daikin ACs work.  My remote has a sticker on the back which sais ARC417Axx (xx=some numbers).  The design looks dated, it's not a currently sold model.

Point a cellphone camera to the remote while you press the buttons.  You can see the IR LED flash on the cellphone screen.  Compare with the (not working) Broadlink send.  You should be able to diagnose if the pulse is really (a lot) longer.

rodolfodegrande, 2021-02-24:
i can control my daikin. 

i use smartir platform and device_code: 1106