mjg59/python-broadlink #371: RM mini 3 (0x5f36) : "StorageError" in broadlink_discovery

cbichler, 2020-05-25:
Hi all,

i have fiddled around with the RM mini 3 i got last week (new version 0x5f36, firmware 44057) and have tried using python-broadlink for learning IR codes. 

When using broadlink_discovery, the device is recognized as a RM4 (as categorized in `__init__.py`):
```
 ./broadlink_discovery
Discovering...
###########################################
RM4
# broadlink_cli --type 0x5f36 --host 10.0.0.10 --mac e5XXXXX24
Device file data (to be used with --device @filename in broadlink_cli) :
0x5f36 10.0.0.10 e5XXXX24
Traceback (most recent call last):
  File "./broadlink_discovery", line 24, in <module>
    print("temperature = {}".format(device.check_temperature()))
  File "/usr/local/lib/python3.8/dist-packages/broadlink-0.14.0-py3.8.egg/broadlink/__init__.py", line 674, in check_temperature
    return self._read_sensor( 0x24, 4, 100.0 )
  File "/usr/local/lib/python3.8/dist-packages/broadlink-0.14.0-py3.8.egg/broadlink/__init__.py", line 654, in _read_sensor
    check_error(response[0x22:0x24])
  File "/usr/local/lib/python3.8/dist-packages/broadlink-0.14.0-py3.8.egg/broadlink/exceptions.py", line 97, in check_error
    raise exception(error_code)
broadlink.exceptions.StorageError: The device storage is full
```
I had to comment out lines 23-25 in broadlink_discovery, as otherwise temperature was tried to be read back (seemingly an RM4 feature, but RM mini 3 does not have a temperature sensor). 
Also, the MAC address broadlink_discovery detects is backwards (should start with 24)

I still tried to learn IR codes using broadlink_cli with the correct MAC, it works to put the device in learning mode and i do get codes back. I tried sending those to the TV using openHAB2, but they don't work (not sure how to verify or compare the codes as RM Bridge does not support RM3 protocol - any pointers welcome). It does work however to learn the codes correctly with the Broadlink App. 

Happy to provide extra info wherever needed, not sure the IR code issue is connected - will investigate further!

cbichler, 2020-05-25:
Edit: Sorry, just saw now that the MAC endianness is already a known issue: https://github.com/mjg59/python-broadlink/pull/370

felipediel, 2020-05-26:
Fixed in https://github.com/mjg59/python-broadlink/pull/369.

cbichler, 2020-05-26:
Unfortunately not, i was already at current HEAD (commit 65810f5) including #369.
The StorageErrors i see are during discovery, not learning

felipediel, 2020-05-26:
So this exception is also raised when reading a non-existent sensor. Interesting. Could you confirm?

`broadlink_cli --type 0x5f36 --host 10.0.0.10 --mac e5XXXXX24 --temperature`

cbichler, 2020-05-26:
Yeah, exactly. Same result:

`./broadlink_cli --type 0x5f36 --host 10.0.0.10 --mac 24XXXXXe5 --temperature
Traceback (most recent call last):
  File "./broadlink_cli", line 116, in <module>
    print(dev.check_temperature())
  File "/usr/local/lib/python3.8/dist-packages/broadlink-0.14.0-py3.8.egg/broadlink/__init__.py", line 674, in check_temperature
    return self._read_sensor( 0x24, 4, 100.0 )
  File "/usr/local/lib/python3.8/dist-packages/broadlink-0.14.0-py3.8.egg/broadlink/__init__.py", line 654, in _read_sensor
    check_error(response[0x22:0x24])
  File "/usr/local/lib/python3.8/dist-packages/broadlink-0.14.0-py3.8.egg/broadlink/exceptions.py", line 97, in check_error
    raise exception(error_code)
broadlink.exceptions.StorageError: The device storage is full`

closed 2020-06-01 by cbichler