mjg59/python-broadlink #377: Do we need to choose between using this library or the official app exclusively?

w35l3y, 2020-06-01:
Here my steps that made me think about the question in title of the issue...
1. I had my devices added to the official app.
2. I tried to use `broadlink_discovery` and it was failing the authorization process.
3. I followed some steps that I found somewhere in old closed issues to remove the devices from the app and reset them.
4. I ran `deploy.py` and both devices autheticated successfully.
5. I re-added the devices to the official app and the library stopped working.

My devices are `0x6026` and `0x62be`.

**So my question is:** Do we need to choose between using this library or the official app exclusively?

felipediel, 2020-06-06:
@w35l3y Yes. The official app blocks the authentication.

ryanerwin, 2020-06-16:
I wanted to see if my RF devices work in the offical app (some do, some don't) but after using the official app, how do we re-connect via `python-broadlink`.

Now I'm getting:

```
broadlink.exceptions.AuthenticationError: Authentication failed
```

felipediel, 2020-06-16:
@ryanerwin I am working on an update that will make this better. But for now...
1. Factory reset the device.
2. Open Broadlink app and connect the device to your local network.
4. STOP. Do not finish the setup. Close the app.

ryanerwin, 2020-07-10:
Is it possible to use `python-broadlink` to send the initial SSID and SSID password to the broadlink device after reset?

The Broadlink app hates my Unifi WiFI network and I have to run it 100x to eventually get an IP each time I do a factory reset.

li-ruijie, 2020-07-20:
> 
> 
> Is it possible to use `python-broadlink` to send the initial SSID and SSID password to the broadlink device after reset?
> 
> The Broadlink app hates my Unifi WiFI network and I have to run it 100x to eventually get an IP each time I do a factory reset.
Disable bandwidth steering and switch off 5ghz will do the trick.

w35l3y, 2020-07-20:
> Is it possible to use `python-broadlink` to send the initial SSID and SSID password to the broadlink device after reset?
> 
> The Broadlink app hates my Unifi WiFI network and I have to run it 100x to eventually get an IP each time I do a factory reset.

This is a subject for another issue.

tstevens, 2020-08-26:
@felipediel I see a merge added set_lock for devices but how do we use it? In the PR #387 you mentioned that a hello request was needed before calling set_lock but I don't see any documentation of how to actually achieve that.

felipediel, 2020-08-26:
```python3
>>> import broadlink as blk
>>> devs = blk.discover(timeout=5)
>>> {(d.host[0], hex(d.devtype), d.mac.hex()) for d in devs}  # Print devices
>>> d = devs[0]  # Example device
>>> d.auth()
>>> d.set_lock(False)
```

If you use Home Assistant, in the next update there will be a configuration flow to assist with unlocking.

tstevens, 2020-08-26:
That doesn't seem to be working on my side once the device is added to the app. Maybe Broadlink changed something in a recent firmware? Just setup the device this afternoon and there was an update during that process. I have been able to get it controlling devices through homebridge when I only associate it with my wifi and don't complete the setup in app so that certainly still works.

```
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink as blk
>>> devs = blk.discover(timeout=5)
>>> {(d.host[0], hex(d.devtype), d.mac.hex()) for d in devs}
{('10.0.1.XX', '0x648d', '24...a4')}
>>> d = devs[0]
>>> d.auth()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.7/site-packages/broadlink/__init__.py", line 271, in auth
    check_error(response[0x22:0x24])
  File "/home/pi/.local/lib/python3.7/site-packages/broadlink/exceptions.py", line 97, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthenticationError: Authentication failed
```

felipediel, 2020-08-26:
You need to be authenticated to unlock. Your device needs a [factory reset](https://github.com/mjg59/python-broadlink/issues/377#issuecomment-644516238).

tstevens, 2020-08-26:
Ok just to avoid some back and forth.

1. Factory Reset of Device
2. Connect to network and have broadlink app configure wifi
3. Leave app after wifi is setup, don't add device
4. Run unlock commands from above.
5. Can I now add to app and the lock state stays unlocked?

Maybe I'm confused about the lock. I have previously had the RM4 attached to my local wifi network. I was then able to run commands to the RM4 without the app and also discover it from python (including getting the device lock state which was False at the time and then became True after adding to app). Is it possible to unlock and have both app & local network control? Or was the lock just preventing local control even before adding to the broadlink app?

felipediel, 2020-08-26:
If d.is_locked is False your device is unlocked and you can control it with this module. The Broadlink app will lock the device if you add it and finish the setup. If you want to use both, I think that is worth trying this:
```python
>>> import broadlink as blk
>>> devs = blk.discover(timeout=5)
>>> d = devs[0]  # Example
>>> d.auth()
```
Authenticate to the device and do not close the terminal. Open the app, add the device, finish the setup. Go back to the terminal and unlock the device:
```python3
>>> d.set_lock(False)
```
I don't know if this will work. I don't have the device to test.

fielux, 2020-08-26:
> You need to be authenticated to unlock it. Your device needs a [factory reset](https://github.com/mjg59/python-broadlink/issues/377#issuecomment-644516238).

Ehm apologies for the dumb question... not clear to me how to perform the FACTORY reset... And... is it unlocking the device? In case the answer is no, how do I unlock it? Thx for patience

felipediel, 2020-08-26:
There is a reset button on the side of the device. You need to hold it for a few seconds to put the device in AP mode (the LED will blink when you succeed). Then you open the app, connect the device to your local network and STOP. Do not finish the setup. Close the app.

fielux, 2020-08-26:
> There is a reset button on the side of the device. You need to hold it for a few seconds to put the device in AP mode (the LED will blink when you succeed). Then you open the app, connect the device to your local network and STOP. Do not finish the setup. Close the app.

Thx for your kind answer. At the beginning i did the mistake to use broadlink app and finish the segup. I then tried to fix it multiple times (I mean resetting the RM mini 3 device and providing to it the wifi credentials through ihc for eu app without completing the setup). Unfortunately E.g. Learn is still not working. Sending the same command to the other 2 old RM mini 3, they work perfectly. I’m getting mad

felipediel, 2020-08-26:
```python3
>>> import broadlink as blk
>>> devs = blk.discover(timeout=5)
>>> [(d.host[0], hex(d.devtype), d.mac.hex()) for d in devs]  # Print devices
```
What is the output?

fielux, 2020-08-26:
all the 3 devices are properly recognized with their own ip and mac, but only the most recent one is not working

felipediel, 2020-08-26:
I need to see the type.

fielux, 2020-08-26:
[('192.168.1.125', '0x2737', 'deeab234ea34'), ('192.168.1.189', '0x2737', '3a7b8342f7c8'), ('192.168.1.42', '0x5f36', '3a7f7aa7df24')]

The one with 0x5f36 is the one that is not working

felipediel, 2020-08-26:
```python3
>>> d = devs[2]
>>> d.is_locked
```
What is the output?

fielux, 2020-08-26:
False

fielux, 2020-08-26:
In order to provide you the type (as per your request above) I had to upgrade from plugin 0.12 (that was working perfectly for the other 2 RM Mini 3) to 0.14 (that I understand is mandatory for device type 0x5f36). On 0.14 unfortunately all the devices (all unlocked) give me the error "Error Connecting to Broadlink device...."

felipediel, 2020-08-26:
You cannot update the library without updating Home Assistant. You need to roll back to the previous version and wait for the release.

fielux, 2020-08-26:
I use Domoticz. If I roll back to 0.12, how can I just add to 0.12 only the support for 0x5f36 devices?

felipediel, 2020-08-26:
You need to ask them to support the latest version of this library.

fielux, 2020-08-26:
I thought it was just needed to manually update file "__init__.py"....

felipediel, 2020-08-26:
0.12 to 0.14 is a big leap, you need to deal with the breaking changes.

Ircama, 2020-09-10:
> I don't know if this will work. I don't have the device to test.

Thanks for this trick.
It works (at least with manufacturer=Broadlink, model='RM4C mini', devtype=0x62be).
I can now concurrently control the device with this module and with the Broadlink app.

felipediel, 2020-09-10:
@Ircama Great! Thank you!

Ircama, 2020-09-10:
> 1. Factory reset the device.

This is not strictly required from my test with an *RM4C mini* in order to remove an existing lock made by the Broadlink app [without availability of authorized and active handle](https://github.com/mjg59/python-broadlink/issues/377#issuecomment-680720416): the lock can be reset by removing the device from the Broadlink app (through the app). After removal, it can be newly reconnected via the app,  interrupting the process before registering its name (e.g., not finishing the setup). Then we need [an active handle](https://github.com/mjg59/python-broadlink/issues/377#issuecomment-680679782) via successful `d.auth()`, then a name can be given with the Broadlink app, then the lock can be removed via `d.set_lock(False)` (having kept the handle active).

tstevens, 2020-09-10:
@felipediel I can also confirm that trick works on my RM4 mini

rabin-io, 2020-09-10:
@felipediel I can also confirm that trick works on my RM mini 3

felipediel, 2020-09-10:
Great! I will consider taking this hack to Home Assistant in the future.

jbaarends, 2020-12-12:
Is the authentication issue only with the broadlink universal remote? Or also with the IHC app?

felipediel, 2020-12-13:
@jbaarends It is a feature of the firmware. When the device is locked - this is done automatically after authentication in the Broadlink app - it does not authenticate external applications, including this one. You now have a button to unlock the device in the Broadlink app, you don't need to factory reset the device anymore.

felipediel, 2021-02-18:
The Broadlink app now has a button to unlock the device. This issue no longer exists. Closing.

closed 2021-02-18 by felipediel