mjg59/python-broadlink #385: Support for Broadlink RM3 Pro Plus v2 ?

rephus, 2020-06-16:
Hi, I got a `Broadlink RM3 Pro Plus` with RF and IR and  I can find it using `broadlink_discovery`, I can even enter in learn mode using `broadlink_cli --learn`

The discovery command tells me it is a device type `0x27a9` which corresponds to   `RM2 Pro Plus_300` based on this code https://github.com/mjg59/python-broadlink/blob/e9e14e82d2d11dd85c97f0e94b0035c222a77da6/broadlink/__init__.py#L41

But the problem is that once I enter learn mode (the orange light turns on in the device) it can't learn any code, I tried several times, with different buttons. I know my remote is compatible because I got it working with the official broadlink Android App.

As an experiment, I tried all the different RM PRo device types, all of them seem to trigger the Learn mode, but none of them  are able to learn the remote code.

My final goal is to learn the RF code of my remote so I can activate devices from a script. I've tried different approaches and even different repos, but most of them seem to be using this library or a copy...

I appreciate your time, thanks.

felipediel, 2020-06-16:
Which one is not working? IR or RF?

felipediel, 2020-06-17:
Use `--rfscanlearn` for RF.

rephus, 2020-06-17:
Awesome, `--rfscanlearn` worked, thanks a lot, I got my RF code.

closed 2020-06-17 by rephus