mjg59/python-broadlink #390: Support for ELECTROLUX_OEM 0x4f9b

Artanicus, 2020-06-27:
This is on an Electrolux WP71-265WT portable AC unit. Identified as broadlink affiliated by pcap and the app calling out to the broadlink cloud. App is so so so bad I'd rather build my own.

What doesn't work:
- discovery over broadcast
- detection as a supported device
- any functionality such as turning the unit on/off

What works:
- Discovery if specifying direct ip:
```
%> python3 ./cli/broadlink_discovery --dst-ip 10.0.10.125
Discovering...
###########################################
Unknown
# broadlink_cli --type 0x4f9b --host 10.0.10.125 --mac 85f0bb42f7c8
Device file data (to be used with --device @filename in broadlink_cli) : 
0x4f9b 10.0.10.125 85f0bb42f7c8
```
- Auth
- device.name: `ELECTROLUX_OEM`
- dev.get_fwversion(): 45027
- dev.devtype: 20379

I'm just looking for the ability to turn the unit on but there's much more possible functionality:
- temp set point adjust
- mode
- detailed scheduling

Anything else I should dig into? Not afraid to get dirty and I am comfortable with Python but just not sure where to start.

felipediel, 2020-07-08:
You can try the [hysen class](https://github.com/mjg59/python-broadlink/blob/daebd806fd8529b9c29b4d70f82a036f30ea5847/broadlink/__init__.py#L712).

ilyakamens, 2022-05-09:
@Artanicus, did you end up building your own app?

Artanicus, 2022-05-09:
> @Artanicus, did you end up building your own app?

The summer ended before I got around to it and hasn't gotten hot enough since then to remember this issue. So no prior work has been done on my part to support this but I'll have another look this summer once the heat provides enough motivation!

felipediel, 2024-04-21:
Now we have the `hvac` class, it's worth giving it a try