mjg59/python-broadlink #391: TypeError: unsupported operand type(s) for &: 'str' and 'int'

salmorejos, 2020-07-05:
I was using version 9 of broadlink and it worked fine for me. Due to some computer problems I had to reinstall broadlink and put version 14. Now, when I try to execute the code that previously worked, now these messages appear. Any easy tips for newbies? THANK YOU!

This is the messages that appear:

Call Stack information:
   File "C: \ Program Files (x86) \ National Instruments \ LabVIEW 2020 \ SOLAR REGULATOR PYTHON 2020 \ BROADLINK A1.py", line 7, in function
     devices.auth ()
   File "C: \ Python27 \ broadlink \ __ init__.py", line 224, in auth
     response = self.send_packet (0x65, payload)
   File "C: \ Python27 \ broadlink \ __ init__.py", line 251, in send_packet
     packet [0x24] = self.devtype & 0xff

TypeError: unsupported operand type (s) for &: 'str' and 'int'
[image]

felipediel, 2020-07-08:
Your are defining devtype as a string, but it should be a positive integer. Use `0x2714` for A1 devices.

salmorejos, 2020-07-23:
Thanks, felipediel.

closed 2020-07-25 by salmorejos