mjg59/python-broadlink #393: Support for Bestcon MCB1 devices

geprieto, 2020-07-10:
Any chance on adding support? I have a couple of them, and considering size/output/price, they're going to be very popular...

Mister-Slowhand, 2020-11-19:
I'd like to know that too, I'm considering it as it's the replacement for SC1.
Maybe the SP4L integration could work as SC1 was working as a SP2 ?

KTibow, 2020-11-20:
@felipediel how hard would this be? Is the device class already there?

felipediel, 2020-11-20:
It is very likely that it can be controlled with the SP4 or SP4B class. Users need to test. Please give me the devtype (PID) so I can start the investigation.

Mister-Slowhand, 2020-11-20:
I haven't got one yet, but if needed I can order one or two. Expect 10 days delay...

There are 2 devices after the SC1 : 
- the SCB1 that seems older, and has 3 wire (Neutral, LIn, LOut) and energy monitor
- the MCB1, smaller and with 4 sockets (2 neutral linked, LIn, LOut) without energy monitor

I guess the MCB1 is the most interesting one to investigate ?

felipediel, 2020-11-20:
I think so. This device is really good value for money.

Mister-Slowhand, 2020-11-20:
Bought 2 MCB1, 1 SCB1 along with the 2 SP4L that I needed anyway.
Should have those around Dec 10, I'll make tests if no one has some already at home to provide information.

I bet SCB1 is a SP3 (with energy monitor) and MCB1 is SP4 ;-)

rinne, 2020-11-22:
> It is very likely that it can be controlled with the SP4 or SP4B class. Users need to test. Please give me the devtype (PID) so I can start the investigation.

Hi

I did some investigation in the last spring and this is what I found then.  Unfortunately later, the issue was buried under other stuff.

My MCB1 has device id 0x6111.

My findings:
- It can be found with eControl and configuration can be done (i.e. the device connects to correct WiFi and gets an address from DHCP and so on). However eControl does not find it any longer.
- Even though eControl doesn't find it, the device indeed answers to the discovery broadcast, at least if it's sent directly to the device ip (I don't remember if it actually answers to real broadcasts, but most likely it does) and it yields the encryption key.
- If configured by newer Broadlink application, it seems to go to "private" mode and does not answer to broadcasts.
- My recollection is that trying to command the unit with SP3 compatible power on/off commands or probing the status, caused an error packet to be sent back, but I can't remember what it was.

I can try to find the switch somewhere and test it further, if someone has suggestions.

rinne, 2020-11-22:
Actually I tried it again.  I was able to join the device to network with

```
$ ./broadlink_cli --joinwifi SSID wifi-password
```

and simply by using device id 0x7583 instead of the real one (0x6111), I was able to turn the device on and off and also --check works as expected.

```
$ ./broadlink_cli --device '0x7583 192.168.2.18 c8f74275f000' --turnon
== Turned * ON * ==
$ ./broadlink_cli --device '0x7583 192.168.2.18 c8f74275f000' --turnoff
== Turned * OFF * ==
```

Patch below:

```
diff --git a/broadlink/__init__.py b/broadlink/__init__.py
index 74ae1fa..677b846 100644
--- a/broadlink/__init__.py
+++ b/broadlink/__init__.py
@@ -44,6 +44,7 @@ SUPPORTED_TYPES = {
     0x7579: (sp4, "SP4L-EU", "Broadlink"),
     0x7583: (sp4, "SP mini 3", "Broadlink"),
     0x7D11: (sp4, "SP mini 3", "Broadlink"),
+    0x6111: (sp4, "MCB1", "Broadlink"),
     0x648B: (sp4b, "SP4M-US", "Broadlink"),
     0x2712: (rm, "RM pro/pro+", "Broadlink"),
     0x272A: (rm, "RM pro", "Broadlink"),
```

rinne, 2020-11-22:
Now that I test this specimen I found, I get the device id as 0x756f instead of 0x6111.  I'm not sure, if the discrepancy is different devices or maybe my earlier test was somewhat tainted because I tried to configure the device using eControl.  So, I guess the 0x6111 is just wrong.  In the code, there is already 0x756c and using it as a parameter to broadlink_cli, everything seems to work.  So, most probably the actually working patch is:

```
diff --git a/broadlink/__init__.py b/broadlink/__init__.py
index 74ae1fa..baddedd 100644
--- a/broadlink/__init__.py
+++ b/broadlink/__init__.py
@@ -41,6 +41,7 @@ SUPPORTED_TYPES = {
     0x9479: (sp2, "SP3S-US", "Broadlink"),
     0x947A: (sp2, "SP3S-EU", "Broadlink"),
     0x756C: (sp4, "SP4M", "Broadlink"),
+    0x756F: (sp4, "MCB1", "Broadlink"),
     0x7579: (sp4, "SP4L-EU", "Broadlink"),
     0x7583: (sp4, "SP mini 3", "Broadlink"),
     0x7D11: (sp4, "SP mini 3", "Broadlink"),
```

Sorry about the sloppiness above.

felipediel, 2020-11-22:
Let me see if I understand. You have two devices: 0x6111 and 0x756F, and the second works with the SP4 class? If so, does the first work with SP4B class?

rinne, 2020-11-22:
> Thank you! Let me see if I understand. You have two devices: 0x6111 and 0x756F, and the second works with the SP4 class? If so, does the first work with SP4B class?

Hi

No, I seem to have only 0x756F and the other code was just an error caused by my earlier half-hearted reverse engineering effort.  It works only with sp4 class, and _not_ with sp4b class.

Mister-Slowhand, 2020-12-03:
Ok, I received my devices and I didn't get the expected behavior. Long story short, MCB1 and SCB1E work (almost), SP4L don't !

I'm putting here the results of my tests for all devices, but tell me if you want separate threads.
**MCB1**
1 - connect laptop to BroadLink_WiFi_Device
2 - ./broadlink_cli --joinwifi iOTNetwork iOTPass!
3 - disconnected from Wifi, LED blue fix
4 - Connect to normal wifi.
5 - see the device in my network
6 - authorize and set DHCP address with MAC (WiFi is TPLink Omada with Mac Filter, and Fortinet Firewall as DHCP server)
7 - Force reconnection, see new IP
8 - python tests
```
>>> dev=blk.hello("192.168.7.60")
>>> dev
<sp4: Broadlink MCB1 (0xa56a) at 192.168.7.60:80 | a0:43:b0:2b:c6:87 | 42346 | Unlocked>
>>> dev.auth()
True
>>> dev.get_state()
{'pwr': 0, 'indicator': 1, 'maxworktime': 0, 'childlock': 0}
>>> dev.is_locked
False
>>> dev.get_fwversion()
57052
```

7 - CLI tests
```
$ ./broadlink_cli --type 0xa56a --host 192.168.7.60 --mac a043b02bc687 --switch
* Switch to OFF *
$ ./broadlink_cli --type 0xa56a --host 192.168.7.60 --mac a043b02bc687 --turnon
== Turned * ON * ==
$ ./broadlink_cli --type 0xa56a --host 192.168.7.60 --mac a043b02bc687 --turnoff
== Turned * OFF * ==
```

So this one is perfect, once I added the "0xa56a: (sp4, "MCB1", "Broadlink")" line in the \_\_init\_\_.py file.

**SCB1E**

0 - Was in smart mode, turned it into Noapp mode
1 - connect laptop to BroadLink_WiFi_Device
2 - ./broadlink_cli --joinwifi iOTNetwork iOTPass!
3 - disconnected from Wifi, LED off
4 - Connect to wifi.
5 - see the device in my network
6 - authorize and set DHCP address with MAC
7 - Force reconnection, see new IP
8 - python tests
```
>>> import broadlink as blk
>>> dev=blk.hello("192.168.7.51")
>>> dev
<sp4b: Broadlink SCB1E (0x6113) at 192.168.7.51:80 | 24:df:a7:18:93:24 | SCB1E | Unlocked>
>>> dev.auth()
True
>>> dev.is_locked
False
>>> dev.get_fwversion()
57052
>>> dev.get_state()
{'pwr': 0, 'indicator': 1, 'maxworktime': 0, 'current': 0, 'volt': 236600, 'power': 0, 'totalconsum': 0, 'overload': 0, 'childlock': 0}
```

7 - CLI tests
```
$ ./broadlink_cli --type 0x6113 --host 192.168.7.51 --mac 24dfa7189324 --turnoff
== Turned * OFF * ==
$ ./broadlink_cli --type 0x6113 --host 192.168.7.51 --mac 24dfa7189324 --turnon
== Turned * ON * ==
$ ./broadlink_cli --type 0x6113 --host 192.168.7.51 --mac 24dfa7189324 --switch
* Switch to OFF *
$ ./broadlink_cli --type 0x6113 --host 192.168.7.51 --mac 24dfa7189324 --energy
Traceback (most recent call last):
  File "./broadlink_cli", line 117, in <module>
    print(dev.get_energy())
AttributeError: 'sp4b' object has no attribute 'get_energy'
```

First I added it as a SP2 (SP3S) but didn't work, then as a SP4, and finally as a SP4b and it works fine, except for the "get_energy" that doesn't exists in the SP4b profile.
    0x6113: (sp4b, "SCB1E", "Broadlink"),

**Now for the SP4L-EU**
Same configuration method, I see it on my network, I can ping, send a few requests but they go back to init mode when they do not have internet access (which is blocked on my OIT network)
With SP4, I get this :
```
>>> dev.get_state()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/marc/.local/lib/python3.8/site-packages/broadlink/switch.py", line 280, in get_state
    return self._decode(response)
  File "/home/marc/.local/lib/python3.8/site-packages/broadlink/switch.py", line 297, in _decode
    check_error(response[0x22:0x24])
  File "/home/marc/.local/lib/python3.8/site-packages/broadlink/exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.DataValidationError: [Errno -6] Structure is abnormal
```

With SP4b : 
```
>>> dev=blk.hello("192.168.7.25")
>>> dev.auth()
True
>>> dev.get_state()
{'pwr': 1, 'ntlight': 0, 'indicator': 1, 'usbpwr': 0, 'maxworktime': 0, 'usbmaxworktime': 0, 'ntlbrightness': 100, 'current': -1, 'volt': -1, 'power': -1, 'totalconsum': -1, 'overload': -1, 'childlock': 0}
>>> dev.get_fwversion()
57209

```
Then it disconnect and reset. Both SP4L-EU I got behave the same.

felipediel, 2020-12-03:
@Mister-Slowhand Awesome! Thank you very much for the detailed tests! I will add support later.

About blocking the devices in the firewall, it makes them reboot every 3 minutes. It's not healthy. We are working on a local heartbeat to keep them up without the cloud. For now, you can call device.keep_alive() every two minutes to avoid the reboots if you want to block them.

Mister-Slowhand, 2020-12-03:
Thanks for adding, and thanks for the information about the reboot.
But, probing the device every X seconds, Home Assistant is already doing that ?! So I've tried and configured my 2 SP4L, modified the init.py in the broadlink library, and added the 2 plugs. Let's see if they stay connected ! At least, it works, I can switch them.
Also, it's a SP4B, but with the nightlight. Would this need a new profile ?

MCB1 don't need that access, they do not reboot.

felipediel, 2020-12-03:
What are the PIDs of SP4 devices?

felipediel, 2020-12-04:
About SCB1E's get_energy(), I think this information is available with:
```python3
>>> dev.get_state()["power"]
```
I will add it as SP4B and improve the interface later. Could you please plug something on this device and call dev.get_state() so we can see the JSON with the data types?

Mister-Slowhand, 2020-12-04:
My SP4L-EU show as 0x618b. (and work as a SP4M !!)

You're right, the power/current/volt are readable, even with the SP4L. They do not work with broadlink_cli neither are they attributes in home assistant.
I'm using SP4L since last evening and everything is fine. I'll plug the SCB1E this week-end as it's gonna need a little more work. I'll tell you the values, but volt seems in mV (see the tests before, 236600 for 236V )

`>>> d
<sp4b: Broadlink SP4L-EU (0x618b) at 192.168.7.25:80 | a0:43:b0:00:29:db | 24971 | Unlocked>
>>> d.auth()
True
>>> d.get_state()
{'pwr': 1, 'ntlight': 0, 'indicator': 1, 'usbpwr': 0, 'maxworktime': 0, 'usbmaxworktime': 0, 'ntlbrightness': 100, 'current': -1, 'volt': -1, 'power': -1, 'totalconsum': -1, 'overload': -1, 'childlock': 0}
>>> d.get_state()['volt']
-1
>>> d.get_state()['current']
-1
`

felipediel, 2020-12-04:
Thank you! I added all the types, now its just a matter of time.

Mister-Slowhand, 2020-12-06:
Here's what the SCB1E answers while the dishwasher is working :
- just power on :
`{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'current': 11, 'volt': 233500, 'power': 0, 'totalconsum': 0, 'overload': 0, 'childlock': 0}
`
- when it starts :
`{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'current': 100, 'volt': 233800, 'power': 9200, 'totalconsum': 0, 'overload': 0, 'childlock': 0}
`

- while washing :
`{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'current': 8169, 'volt': 234600, 'power': 1916480, 'totalconsum': 10, 'overload': 0, 'childlock': 0}
`

So power consumption is also in mW ?

felipediel, 2020-12-07:
Yes. Thank you!

amosrohe, 2021-01-08:
hi 
trying to work with the MCB1 and it looks like  it recognizes it but not the commands

>>> dev
<device: None None (0xa56a) at 192.168.0.184:80 | a0:43:b0:2b:ee:d7 | 42346 | Unlocked>
>>> dev.get_state()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'device' object has no attribute 'get_state'

same for power and other instructions 

tnx

amosrohe, 2021-01-11:
anyone???

felipediel, 2021-01-11:
Hi. It's merged. If you want to use it before the release:
```bash
pip3 install git+https://github.com/mjg59/python-broadlink.git@master --force-reinstall --no-deps
```

amosrohe, 2021-01-12:
tnx felipediel 
this solved it :-)

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/478. Thank you!

closed 2021-02-17 by felipediel

felipediel, 2021-03-23:
@Mister-Slowhand @KTibow Hi friends. I'm bringing new features to Home Assistant and I need to know: does set_state(usbpwr=1) and usbmaxworktime do anything on SP4 devices? What options are not supported?
```python3
    def set_state(
        self,
        pwr: bool = None,
        ntlight: bool = None,
        indicator: bool = None,
        usbpwr: bool = None,
        maxworktime: int = None,
        usbmaxworktime: int = None,
        ntlbrightness: int = None,
        childlock: bool = None,
    ) -> dict:
```

KTibow, 2021-03-23:
@felipediel I don't know, but for the SP4B, these are the only things available in `get_state`:
```python3
{'pwr': 0, 'indicator': 1, 'maxworktime': 0, 'childlock': 0}
```

felipediel, 2021-03-23:
Thank you KT! I will create a separate class to improve typing.

felipediel, 2021-03-23:
@KTibow Sorry to abuse your goodwill, could you please see which of these commands work? I need to make a cut in the class to implement new features.
```python3
>>> import broadlink as blk
>>> d = blk.hello("192.168.0.16")  # Example device

# Test 1
>>> print(d.send_packet(0x6a, bytes([1, 0, 0, 0])))
>>> print(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0])))

# Test 2
>>> print(d.send_packet(0x6a, bytes([5, 0, 0, 0])))
>>> print(d.send_packet(0x6a, bytes([4, 0, 5, 0, 0, 0])))

# Test 3
>>> print(d.send_packet(0x68, b''))
```

Mister-Slowhand, 2021-03-23:
Hi,

I didn't know what you were considering 'SP4' devices, so here is the result from the latest devices, SP4L, MCB1 and SC1 !

`
>>> import broadlink as blk
>>> d=blk.hello("192.168.7.25")
>>> d
<sp4s: Broadlink SP4L-EU (0x618b) at 192.168.7.25:80 | a0:43:b0:00:29:db | 24971 | Unlocked>
>>> d.auth()
True
>>> d.get_state()
{'pwr': 1, 'ntlight': 0, 'indicator': 1, 'usbpwr': 0, 'maxworktime': 0, 'usbmaxworktime': 0, 'ntlbrightness': 100, 'childlock': 0}
>>> print(d.send_packet(0x6a, bytes([1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00(\xcc\xf9\xff\x8ba\xee\x03\xe5\xbe\xdb)\x00\xb0C\xa0\x00\x00\x00\x00\xb0\xbe\x00\x00'
>>> print(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00-\xcc\xf9\xff\x8ba\xee\x03\xe6\xbe\xdb)\x00\xb0C\xa0\x00\x00\x00\x00\xb4\xbe\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x6a, bytes([5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00.\xcc\xf9\xff\x8ba\xee\x03\xe7\xbe\xdb)\x00\xb0C\xa0\x00\x00\x00\x00\xb4\xbe\x00\x00'
>>> print(d.send_packet(0x6a, bytes([4, 0, 5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x003\xcc\xf9\xff\x8ba\xee\x03\xe8\xbe\xdb)\x00\xb0C\xa0\x00\x00\x00\x00\xb8\xbe\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x68, b''))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00,\xcc\xfc\xff\x8ba\xec\x03\xe9\xbe\xdb)\x00\xb0C\xa0\x00\x00\x00\x00\xaf\xbe\x00\x00'
>>> 
>>> 
>>> 
>>> d=blk.hello("192.168.7.52")
>>> d
<sp4: Broadlink MCB1 (0xa56a) at 192.168.7.52:80 | a0:43:b0:2c:16:43 | 42346 | Unlocked>
>>> d.auth()
True
>>> d.get_state()
{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'childlock': 0}
>>> print(d.send_packet(0x6a, bytes([1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc8\xcb\xf9\xffj\xa5\xee\x03\xba\xe5C\x16,\xb0C\xa0\x00\x00\x00\x00\xb0\xbe\x00\x00'
>>> print(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xcd\xcb\xf9\xffj\xa5\xee\x03\xbb\xe5C\x16,\xb0C\xa0\x00\x00\x00\x00\xb4\xbe\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x6a, bytes([5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xce\xcb\xf9\xffj\xa5\xee\x03\xbc\xe5C\x16,\xb0C\xa0\x00\x00\x00\x00\xb4\xbe\x00\x00'
>>> print(d.send_packet(0x6a, bytes([4, 0, 5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd3\xcb\xf9\xffj\xa5\xee\x03\xbd\xe5C\x16,\xb0C\xa0\x00\x00\x00\x00\xb8\xbe\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x68, b''))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xcc\xcb\xfc\xffj\xa5\xec\x03\xbe\xe5C\x16,\xb0C\xa0\x00\x00\x00\x00\xaf\xbe\x00\x00'
>>> 
>>> 
>>> 
>>> d=blk.hello("192.168.7.51")
>>> d
<sp4s: Broadlink SCB1E (0x6113) at 192.168.7.51:80 | 24:df:a7:18:93:24 | SCB1E | Unlocked>
>>> d.auth()
True
>>> d.get_state()
{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'childlock': 0, 'current': 0.011, 'volt': 240.2, 'power': 0.0, 'totalconsum': 59.98, 'overload': 0.0}
>>> print(d.send_packet(0x6a, bytes([1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xf6\xca\xf9\xff\x13a\xee\x03_\xa8$\x93\x18\xa7\xdf$\x00\x00\x00\x00\xb0\xbe\x00\x00'
>>> print(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfb\xca\xf9\xff\x13a\xee\x03`\xa8$\x93\x18\xa7\xdf$\x00\x00\x00\x00\xb4\xbe\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x6a, bytes([5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfc\xca\xf9\xff\x13a\xee\x03a\xa8$\x93\x18\xa7\xdf$\x00\x00\x00\x00\xb4\xbe\x00\x00'
>>> print(d.send_packet(0x6a, bytes([4, 0, 5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\xcb\xf9\xff\x13a\xee\x03b\xa8$\x93\x18\xa7\xdf$\x00\x00\x00\x00\xb8\xbe\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x68, b''))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfa\xca\xfc\xff\x13a\xec\x03c\xa8$\x93\x18\xa7\xdf$\x00\x00\x00\x00\xaf\xbe\x00\x00'
>>> 
`

felipediel, 2021-03-23:
Oh god, I forgot to mention t the authentication step. All responses came with AuthorizationErrors 😆 
I updated the instructions. Could you please do it again? Sorry!

Mister-Slowhand, 2021-03-23:
My mistake, did make the d.auth(), but in fact, I just made the get_state() part for all devices, before cycling back to the 3 tests...

Is this ok ? 
`
>>> d=blk.hello("192.168.7.25")                    
>>> d.auth()
True
>>> d.get_state()
{'pwr': 1, 'ntlight': 0, 'indicator': 1, 'usbpwr': 0, 'maxworktime': 0, 'usbmaxworktime': 0, 'ntlbrightness': 100, 'childlock': 0}
>>> d
<sp4s: Broadlink SP4L-EU (0x618b) at 192.168.7.25:80 | a0:43:b0:00:29:db | 24971 | Unlocked>
>>> 
>>> print(d.send_packet(0x6a, bytes([1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00`\xd2\xfb\xff\x8ba\xee\x03m7\xdb)\x00\xb0C\xa0\x01\x00\x00\x00\xaf\xbe\x00\x00s1\x05\xb4:i\x9c\nj\x97\xa1\xc4\x93\xe3\x11\xa2'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([1, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00b\xd2\xfb\xff\x8ba\xee\x03o7\xdb)\x00\xb0C\xa0\x01\x00\x00\x00\xaf\xbe\x00\x00s1\x05\xb4:i\x9c\nj\x97\xa1\xc4\x93\xe3\x11\xa2'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x6a, bytes([5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00d\xd2\xfb\xff\x8ba\xee\x03q7\xdb)\x00\xb0C\xa0\x01\x00\x00\x00\xaf\xbe\x00\x00s1\x05\xb4:i\x9c\nj\x97\xa1\xc4\x93\xe3\x11\xa2'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([5, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> 
>>> print(d.send_packet(0x6a, bytes([4, 0, 5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00f\xd2\xfb\xff\x8ba\xee\x03s7\xdb)\x00\xb0C\xa0\x01\x00\x00\x00\xaf\xbe\x00\x00s1\x05\xb4:i\x9c\nj\x97\xa1\xc4\x93\xe3\x11\xa2'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([4, 0, 5, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x68, b''))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x002\xcb\xfc\xff\x8ba\xec\x03u7\xdb)\x00\xb0C\xa0\x01\x00\x00\x00\xaf\xbe\x00\x00'
>>> print(d.decrypt(d.send_packet(0x68, b'')[0x38:]))
b''
>>> 
>>> 
>>> 
>>> 
>>> d=blk.hello("192.168.7.52")
>>> d.auth()
True
>>> d.get_state()
{'pwr': 1, 'indicator': 1, 'maxworktime': 0, 'childlock': 0}
>>> d
<sp4: Broadlink MCB1 (0xa56a) at 192.168.7.52:80 | a0:43:b0:2c:16:43 | 42346 | Unlocked>
>>> print(d.send_packet(0x6a, bytes([1, 0, 0, 0])))                        
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb6\xca\xfb\xffj\xa5\xee\x03f%C\x16,\xb0C\xa0\x01\x00\x00\x00\xaf\xbe\x00\x00'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([1, 0, 0, 0]))[0x38:]))      
b''
>>> 
>>> print(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0])))                  
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb8\xca\xfb\xffj\xa5\xee\x03h%C\x16,\xb0C\xa0\x01\x00\x00\x00\xaf\xbe\x00\x00'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0]))[0x38:]))
b''
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x6a, bytes([5, 0, 0, 0])))                        
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xba\xca\xfb\xffj\xa5\xee\x03j%C\x16,\xb0C\xa0\x01\x00\x00\x00\xaf\xbe\x00\x00'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([5, 0, 0, 0]))[0x38:]))      
b''
>>> 
[... 28 lines cut ...]
>>> print(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00g\xd2\xfb\xff\x13a\xee\x03@\x06$\x93\x18\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x006W\xe7\xc6\x02\xa6\xefj4I\xd2\\\xd4\x80\xbc:'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x6a, bytes([5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00i\xd2\xfb\xff\x13a\xee\x03B\x06$\x93\x18\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x006W\xe7\xc6\x02\xa6\xefj4I\xd2\\\xd4\x80\xbc:'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([5, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> 
>>> print(d.send_packet(0x6a, bytes([4, 0, 5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00k\xd2\xfb\xff\x13a\xee\x03D\x06$\x93\x18\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x006W\xe7\xc6\x02\xa6\xefj4I\xd2\\\xd4\x80\xbc:'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([4, 0, 5, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> 
>>> 
>>> 
>>> print(d.send_packet(0x68, b''))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00<\xca\xfc\xff\x13a\xec\x03F\x06$\x93\x18\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x00'
>>> print(d.decrypt(d.send_packet(0x68, b'')[0x38:]))
b''
>>> 
`

felipediel, 2021-03-23:
Thank you very much for the detailed information, it was really helpful!

KTibow, 2021-03-23:
@felipediel actually it's a `SP4M-US`, and being classified as a `sp4b`. Here's everything run in IDLE:
```python3
Python 3.9.2 (tags/v3.9.2:1a79785, Feb 19 2021, 13:44:55) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license()" for more information.
>>> import broadlink
>>> devs = broadlink.discover()
>>> dev = devs[0]
>>> dev
<sp4b: Broadlink SP4M-US (0x648b) at 192.168.1.29:80 | 24:df:a7:36:f1:4e | 25739 | Unlocked>
>>> d = dev
>>> print(d.send_packet(0x6a, bytes([1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xf4\xcb\xf9\xff\x8bd\xee\x03;\xa9N\xf16\xa7\xdf$\x00\x00\x00\x00\xb0\xbe\x00\x00'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([1, 0, 0, 0]))[0x38:]))
b''
>>> print(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfa\xcb\xf9\xff\x8bd\xee\x03=\xa9N\xf16\xa7\xdf$\x00\x00\x00\x00\xb4\xbe\x00\x00'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0]))[0x38:]))
b''
>>> print(d.send_packet(0x6a, bytes([5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfc\xcb\xf9\xff\x8bd\xee\x03?\xa9N\xf16\xa7\xdf$\x00\x00\x00\x00\xb4\xbe\x00\x00'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([5, 0, 0, 0]))[0x38:]))
b''
>>> # wait nvm forgot to auth
>>> d.auth()
True
>>> print(d.send_packet(0x6a, bytes([1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xcf\xd0\xfb\xff\x8bd\xee\x03B\xa9N\xf16\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x00\x16\xd8bO\x0fr\x08\x11l\x9d\x15D4e{#'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([1, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> print(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd1\xd0\xfb\xff\x8bd\xee\x03D\xa9N\xf16\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x00\x16\xd8bO\x0fr\x08\x11l\x9d\x15D4e{#'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([4, 0, 1, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> # Test 2
>>> print(d.send_packet(0x6a, bytes([5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd3\xd0\xfb\xff\x8bd\xee\x03F\xa9N\xf16\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x00\x16\xd8bO\x0fr\x08\x11l\x9d\x15D4e{#'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([5, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> print(d.send_packet(0x6a, bytes([4, 0, 5, 0, 0, 0])))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd5\xd0\xfb\xff\x8bd\xee\x03H\xa9N\xf16\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x00\x16\xd8bO\x0fr\x08\x11l\x9d\x15D4e{#'
>>> print(d.decrypt(d.send_packet(0x6a, bytes([4, 0, 5, 0, 0, 0]))[0x38:]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> # Test 3
>>> print(d.send_packet(0x68, b''))
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\xcc\xfc\xff\x8bd\xec\x03J\xa9N\xf16\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x00'
>>> print(d.decrypt(d.send_packet(0x68, b'')[0x38:]))
b''
>>> 
```

felipediel, 2021-03-23:
Thank you very much!

Mister-Slowhand, 2021-03-24:
> Thank you very much for the detailed information, it was really helpful!

You're welcome, if someone is helpful here, it's you !