mjg59/python-broadlink #397: --rfscanlearn does not work on python2.7 (always timeout error in phase 1)

marc-invalid, 2020-07-24:
The --rfscanlearn command does not work on python2.7.  It always fails with a timeout error in phase 1 (frequency sweep).  The reason is a comparison of two incompatible types (string and number) which always evaluates to False.

Please consider accepting this patch:

```
diff --git a/broadlink/__init__.py b/broadlink/__init__.py
index a09b813..79bcba0 100644
--- a/broadlink/__init__.py
+++ b/broadlink/__init__.py
@@ -662,7 +662,11 @@ class rm(device):
         response = self.send_packet(0x6a, packet)
         check_error(response[0x22:0x24])
         payload = self.decrypt(bytes(response[0x38:]))
-        if payload[len(self._request_header) + 4] == 1:
+        r = payload[len(self._request_header) + 4]
+        if isinstance(r, int):
+            if r == 1:
+                return True
+        elif ord(r) == 1:
             return True
         return False
 
@@ -672,7 +676,11 @@ class rm(device):
         response = self.send_packet(0x6a, packet)
         check_error(response[0x22:0x24])
         payload = self.decrypt(bytes(response[0x38:]))
-        if payload[len(self._request_header) + 4] == 1:
+        r = payload[len(self._request_header) + 4]
+        if isinstance(r, int):
+            if r == 1:
+                return True
+        elif ord(r) == 1:
             return True
         return False

```

steffenu, 2020-09-15:
KUDOS man this works in py 2.7 ... Thanks man

felipediel, 2021-02-17:
Sorry, we no longer support Python 2. Closing.

closed 2021-02-17 by felipediel