mjg59/python-broadlink #399: RM2 does not send anything

salmorejos, 2020-07-25:
RM2 does not send anything
When updating the version of broadlink python now I can't get RM2 to send any code (which previously worked with another older version of broadlink python).
I leave you a screenshot of the file I am trying to run and the error I receive.
Any suggestion of what happens to the code, where
 can I be making the mistake? .... THANK YOU.
[image]

[image]

felipediel, 2021-02-17:
Probably an indexing error because Python 2 treats bytes differently. Sorry, we no longer support Python 2. Closing.

closed 2021-02-17 by felipediel

KTibow, 2021-02-17:
@felipediel maybe you should add a label to issues as for why they were closed?