mjg59/python-broadlink #401: Help requested; edit the length of 2nd RF signal capturing (RM Broadlink 4 Pro)

Frederik840, 2020-07-27:
So, I have been struggling with an issue that I would like to get your input on.
I have successfully captured the RF signals of several remote controls in my home. One of these remotes is related to my gas fire place. This remote controls a valve which opens if the 'on' button on the remote. When I hold the button down, the valve continues to open.

However, when running this script, I can only capture the 'on' signal itself and I am unable to capture the signal for a longer period of time. I tried repeating the signal, but that causes an overload with the valve (instead of peeping one time and opening, it now beeps continuously, so that is not ideal).

Then, I tried editing the captured signal itself, but that results in the code being broken and the fireplace is not responding to those signals. I tried several edits of the signal, but unfortunately they are all unsuccessful. 

So, now my idea is to try to edit the time in the Python script to record the 2nd time the RF signal is sent. 
Can anyone point me in the right direction to do this?
Thanks!!

(FYI, see below the captured RF signal. If you have any other ideas, that would also be appreciated!):
b1c0ac00149f06000c0d0f181b0d0f180f190f190e190f190e191a0e0e191a0e1a0002510c0c0f191a0e0e190f190f180f190e190f191a0e0e191a0e190002510c0d0f191a0d0f190f190e190f190e190f191a0e0e191a0e190002520c0d0f191a0d0f190f190e190f190e190f19190f0e191a0e190002510c0d0f191a0d0f190e190f190e190f190f19190e0f19190e1a0002510c0d0f181a0e0f180f190f190e190f190e191a0e0e1a190e190005dc0000000000000000000000000000000000000000000000000000000000000000000000000"

(some thoughts on the RF signal)
b1c0ac00149f0600 (Identifier / Key?) 
0c0d0f181b0d0f180f190f190e190f190e191a0e0e191a0e1a0002510
c0c0f191a0e0e190f190f180f190e190f191a0e0e191a0e190002510
c0d0f191a0d0f190f190e190f190e190f191a0e0e191a0e190002520
c0d0f191a0d0f190f190e190f190e190f19190f0e191a0e190002510
c0d0f191a0d0f190e190f190e190f190f19190e0f19190e1a0002510
c0d0f181a0e0f180f190f190e190f190e191a0e0e1a190e19 (message body)
0005dc (close signal)
0000000000000000000000000000000000000000000000000000000000000000000000000" (nothing basically)

marc-invalid, 2020-07-27:
I think the python code does not enforce any timeout.  It is the broadlink device itself that decides when the learning is finished and it reports this state change back to the python module.  If there was a way to change the timeout, then it must be a parameter to set/change before or while starting to learn.  Others may be more qualified to tell if such a parameter exists?

However, assuming that it does not exist, then you are on the right track with editing the signal.  Probably your remote sends an initial code (button down event) and then a stream of followup codes (button still down).  There's probably a flag bit somewhere to distinguish the two, and maybe also a sequential number that is shared (or different) in the two packet types.

You need to capture the followup packet to continue your work.  For example, don't release the button when the software asks you to (after scanning the frequency).  Just start learning directly, and it will learn a followup code.

If it's just a flag bit that is different, you might be able to directly use this code, just like you already did with the first button (causing the beeps).  For example send the initial code once, and the followup code 5 times.

If there is a sequential mark in the packet, then this might not work.  You then need to understand the packet contents.  Decode manually in a text editor for example, first drawing out all the 0 and 1 bits in a row, then identify the bit cells, then try to make sense of it (simple on/off, manchester encoding, etc).  It's not difficult if you have a lowlevel programming or electronics background.

It also helps if you have an RTLSDR to view the original signal, especially the timing between the packets.

Good luck!

Frederik840, 2020-07-27:
Hi Marc, thank you, that was helpful!
So, while holding down the 'on' button, I learned that the Broadlink continues to capture the RF signal, up to 2.420 characters.
Will do some diging into the code as suggested, thanks for getting me started!