mjg59/python-broadlink #402: CommandNotSupportedError for get_energy

shouya, 2020-07-29:
Hi,

I just got two SP2 devices which claims to have power consumption monitoring mode.

After setup, I can indeed see the energy information from the app (see top right corner).
[image]

In the app it shows the current version of firmware is V29 and up to date.

However, I failed to get the energy info using this library:

```
./broadlink_cli --type '0x2711' --host 192.168.86.186 --mac 'b4430daa8f59' --energy
Traceback (most recent call last):
  File "./broadlink_cli", line 118, in <module>
    print(dev.get_energy())
  File "/Users/shou/.asdf/installs/python/3.7.3/lib/python3.7/site-packages/broadlink-0.14.0-py3.7.egg/broadlink/__init__.py", line 573, in get_energy
    check_error(response[0x22:0x24])
  File "/Users/shou/.asdf/installs/python/3.7.3/lib/python3.7/site-packages/broadlink-0.14.0-py3.7.egg/broadlink/exceptions.py", line 97, in check_error
    raise exception(error_code)
broadlink.exceptions.CommandNotSupportedError: Command not supported
```

Other commands, however, for example '--check' and '--switch' works perfectly.  Any hints on how to get it working?

shouya, 2020-07-29:
I saw this PR (https://github.com/mjg59/python-broadlink/pull/375) which may be related. I am not sure if the device I had is SP2-CL, at least from the label it only says "SP2".

Also, unlike what @felipediel said in the PR, my switch obviously have some type of power meter, since the app is able to retrieve them.

felipediel, 2020-11-01:
[This](https://github.com/mjg59/python-broadlink/issues/413) might work.

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/521. Thank you!

closed 2021-02-17 by felipediel

shouya, 2021-02-17:
@felipediel Confirmed that get energy is working on my device, thank you!

There is a problem though. With `./cli/boradlink_discovery` it shows that my device have a devtype of '0x2711', but which corresponds to sp2[*](https://github.com/mjg59/python-broadlink/pull/521/files#diff-a29a838665bc485155f9126b5a9c8616d588def9ca2468095d112f58e8650c35R18) instead of sp2s. Using the given devtype to query for energy will still get the `AttributeError: 'sp2' object has no attribute 'get_energy'` error.

But when I manually specify a sp2s devtype (0x271D, 0x2716, 0x2736) it works perfectly.

I am wondering if 0x2711 should be categorized as sp2s or is my device simply reporting a confusing devtype.

felipediel, 2021-02-17:
Yes, I had forgotten, thanks for remembering. I will update the class right now.

felipediel, 2021-02-17:
I create the PR: https://github.com/mjg59/python-broadlink/pull/538. I will merge it later.
Could you send me a link to your device? Is the name SP2 correct?

felipediel, 2021-04-08:
Hi @shouya. I'm pinging you as promised. The update is finally released. You will be able to monitor power consumption after updating Home Assistant.

shouya, 2021-04-08:
Hooray! Thanks for the update @felipediel!

felipediel, 2021-04-08:
You're welcome!