mjg59/python-broadlink #404: StorageError: The device storage is full while learning and workaround

chazeon, 2020-08-07:
I recently got a new Broadlink RM mini 3 from Amazon US, the device type is `0x5F36`. With `python-broadlink`, I am able to "learn" a few IR code, it is working fine. After a few codes learned, the device starts to complain "[`The device storage is full`][1]" error. This problem cannot be resolved by resetting the device, and I am pretty desperate at this point.

[1]: https://github.com/mjg59/python-broadlink/blob/3216b069f0a13fb08295b303219710b6cafbf599/broadlink/exceptions.py#L74

On deciding whether I will need to return the device, I realize that: after using `enter_learning()` function, the device has white indicator on, which means it indeed reads the IR code, but the error only comes when I try to use `check_data()` function. Further tests indicate that as long as I commented out the [`check_error()`][1] statement, the `check_data` function can actually correctly reporting IR code.

[1]: https://github.com/mjg59/python-broadlink/blob/fe4aa5b991e033d865e956d7a96526998e917e0f/broadlink/__init__.py#L631

So I assume the "learning" functionality is actually writing IR codes into the device's memory, which, from what I see, is very much limited. Since there is no "erase" option given, I assume a lot of users are going to encounter the same problem. And, since most users of `python-broadlink` only need to read out the IR code instead of really using the learning function, I think it would be great if an option for temporarily disabling the error check for full device storage could be given.

winchester5050, 2020-08-08:
I'm having the same issue here. I tried commenting out the line you reference and the script still crashes.

chazeon, 2020-08-09:
> I'm having the same issue here. I tried commenting out the line you reference and the script still crashes.

Which script are you using? `broadlink_cli`? I have issues with `broadlink_cli` as well.
But I can work with the most basic commands. Try the following lines in the Python interactive shell:

First find your device and enter learning mode
```python
import broadlink

device = broadlink.discover()
devices.auth()
device.enter_learning()
```
Now the LED indicator should be white, press the button now. After that, run:
```python
packet = devices.check_data()
print(packet.hex())
```
This should show the IR packet in hex format.

digitlength, 2020-08-25:
> > > But I can work with the most basic commands. Try the following lines in the Python interactive shell:
> > 
> > 
> > Can you confirm whether to try these commands in the Python console _after_ commenting out [line 74](https://github.com/mjg59/python-broadlink/blob/3216b069f0a13fb08295b303219710b6cafbf599/broadlink/exceptions.py#L74) in _exceptions.py_, or doesn't it require commenting out that line at all?
> 
> Here I mean after commenting out [line 74](https://github.com/mjg59/python-broadlink/blob/3216b069f0a13fb08295b303219710b6cafbf599/broadlink/exceptions.py#L74).
> 
> But actually, in my test, the first portion works even without commenting out the line. The `check_data` method in the second part is giving out `StorageError`.

So just to check: are you commenting out **only** line 74, or also the whole _check_error()_ function?

mjeshurun, 2020-08-28:
Hello friends,
Good to know I'm not the only one suffering from the  "`The device storage is full`" issue.
I have an RM4 Pro and when I try to learn RF codes using `python-broadlink` or `broadlink-mqtt ` I'm seeing the "device storage is full" error for both learning methods.

By the way, I don't know if it's a problem, but my RM4 Pro is showing a red LED when its in learning mode.

Here is another thread which I tried using to troubleshoot the issue (with the kind help of @hafniumzinc), but with no success:
[https://community.openhab.org/t/broadlink-bestcon-rm4c-mini-via-mqtt/98268](url) 

Have you found any new leads to solve the problem?

mwargan, 2020-09-18:
+1 on facing this issue. Interestingly I noticed that when you learn the command from Home Assistant - works fine. When using the broadlink_cli here, it returns the full storage error.

felipediel, 2020-09-19:
This is not a bug, this is the expected behavior. The applications that import this library should handle this exception using a try except clause. Check the Broadlink integration in Home Assistant for more implementation details.

chazeon, 2020-09-22:
> This is not a bug, this is the expected behavior. The applications that import this library should handle this exception using a try except clause. Check the Broadlink integration in Home Assistant for more implementation details.

Yes, I understand it is reasonable in the API. But the [`boradlink_cli`][1] has not implemented the try-except clause. Which makes the learning functionality the CLI totally unusable for me. The `boradlink_cli` is supposed to be working for end-user right?

[1]: https://github.com/mjg59/python-broadlink/blob/master/cli/broadlink_cli

felipediel, 2020-09-22:
Sorry, I read this more carefully now. Yes, I think it would be reasonable to ignore this error (in a targeted way).

mcgheffe, 2020-10-05:
Is the the error "The device storage is full" supposed to be fixed now?  I have the pulled the latest which includes the fix to broadlink_cli to handle the StorageError exception, but I still encounter the error when I try to do anything with my RM3 mini.

felipediel, 2020-10-05:
Please make sure your code is correct. [This](https://github.com/home-assistant/core/blob/8e6a616846efb2b77b1f176928de6db05187ecfd/homeassistant/components/broadlink/remote.py#L312-L343) is our implementation in Home Assistant. You can use it as a template.

mcgheffe, 2020-10-05:
I see the difference now.  In the current broadlink_cli python, the call to dev.check_data() is wrapped in an 'try', but I'm getting the exception from **dev.enter_learning()**, which is not wrapped in exception handling:

```
Traceback (most recent call last):
  File "./broadlink_cli", line 132, in <module>
    dev.enter_learning()
  File "/usr/local/lib/python3.6/dist-packages/broadlink-0.15.0-py3.6.egg/broadlink/remote.py", line 38, in enter_learning
    check_error(response[0x22:0x24])
  File "/usr/local/lib/python3.6/dist-packages/broadlink-0.15.0-py3.6.egg/broadlink/exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.StorageError: [Errno -5] The device storage is full
```

However, wrapping dev.enter_learning() in a 'try' block doesn't help.  It must fail to enter learning mode since it just times out later reading data:

```
### Ignoring Storage Error ###
Learning...
No data received...
```

I also can't send an IR code I've already learned, so there's a fundamental problem executing any useful operation due to the storage error.

felipediel, 2020-10-05:
You are probably using the wrong class. Try `rm4`. You should also consider using `discover()` or `gendevice()` to generate the device correctly.

mcgheffe, 2020-10-05:
Thanks for the pointer.  You are right.  Changing from using the "rm" class to using the "rm4" class works!  Glad to finally get this working!

mcgheffe, 2020-10-05:
One more note just to complete my view of things.  My "old" RM3 mini still requires the old "rm" class (old firmware being "v57").  The new RM3 minis I have purchased recently do not work with the "rm" class, they require the use of the "rm4" class (new firmware being "v44057").  Thanks for the help sorting this out.

felipediel, 2020-10-05:
Yes. RM mini 3 0x5f36 requires special payload headers. It communicates like RM4 series devices.

mjeshurun, 2020-11-18:
Hi friends, 
Based on the recent posts in this thread am I correct to assume that the "device storage is full" issue has been fixed?
I own an RM4 Pro and tried to learn RF codes, but I'm still getting the "storage full" error message. 
I could use your help in solving this issue, since I'm quite a noob in these coding matters 🙏

mjeshurun, 2020-11-18:
> Thanks for the pointer. You are right. Changing from using the "rm" class to using the "rm4" class works! Glad to finally get this working!

Hi friends,
Based on the recent posts in this thread am I correct to assume that the "device storage is full" issue has been fixed?
I own an RM4 Pro and tried to learn RF codes, but I'm still getting the "storage full" error message.
I could use your help in solving this issue, since I'm quite a noob in these coding matters 🙏

KTibow, 2020-11-18:
@mjeshurun what device class are you using? Try a different one, like `rm4` instead of `rm`?

mjeshurun, 2020-11-18:
> @mjeshurun what device class are you using? Try a different one, like `rm4` instead of `rm`?

Thank you for your reply @KTibow , 
Where do I change the device class to rm4? 

When I ran the following command python-broadlink discovered my device as an rm4:
```
pi@raspberrypi:~/python-broadlink/cli $ ./broadlink_discovery --timeout 10 --dst-ip 192.168.1.6
Discovering...
###########################################
RM4
# broadlink_cli --type 0x61a2 --host 192.168.1.6 --mac 24dfa7b9c689
Device file data (to be used with --device @filename in broadlink_cli) : 
0x61a2 192.168.1.6 24dfa7b9c689
temperature = 0.0
```

felipediel, 2020-11-18:
What command are you using to learn the code?

mjeshurun, 2020-11-18:
> What command are you using to learn the code?

I'm following this set of commands:
```
python3

>>> import broadlink

>>> devices = broadlink.discover(timeout=5)

>>> print(devices)

>>> devices[0].auth()

>>> devices[0].enter_learning()
```

After the last "enter_learning" command I get the error message, so I cant continue to the next commands that are supposed to be: 
```
>>> packet = devices[0].check_data()

>>> print(packet.hex())

>>> devices[0].send_data(packet)
```

felipediel, 2020-11-18:
When you call `devices[0].check_data()`, you need to catch the exceptions with a `try: except` clause. Check the [CLI](https://github.com/mjg59/python-broadlink/blob/29345a129ff6c98962fccba8ada1ea049fd9ec16/cli/broadlink_cli#L222-L232) implementation for more details.

mjeshurun, 2020-11-18:
> When you call `devices[0].check_data()`, you need to catch the exceptions with a `try: except` clause. Check the [CLI](https://github.com/mjg59/python-broadlink/blob/29345a129ff6c98962fccba8ada1ea049fd9ec16/cli/broadlink_cli#L222-L232) implementation for more details.

Thank you for the suggestion, @felipediel .
Can you please explain what you mean by catching the exceptions with a `try: except` clause?
I don't really know how to code, so I don't know what to do.  I can follow detailed guides, but I don't know how to understand coding functions that might be naturally understood by a developer. 
If you are able, I would really appreciate a clarification of what steps I need to take to catch the exceptions 🙏🙏

felipediel, 2020-11-18:
The `try: except` clause is similar to an `if statement`, but we ask for forgiveness instead of asking for permission in order to improve performance.

#### If statement (asking for permission)
```python3
if code_received:
    device.check_data()
else:
    print("No code received")
```
#### Try except (asking for forgiveness)
```python3
try:
    device.check_data()
except ReadError:
    print("No code received")
```
[This code snippet](https://github.com/mjg59/python-broadlink/blob/29345a129ff6c98962fccba8ada1ea049fd9ec16/cli/broadlink_cli#L134-L141) is exactly what you are looking for.

mjeshurun, 2020-11-18:
> The `try: except` clause is similar to an `if statement`, but we ask for forgiveness instead of asking for permission in order to improve performance.
> 
> #### If statement (asking for permission)
> ```python
> if code_received:
>     device.check_data()
> else:
>     print("No code received")
> ```
> 
> #### Try except (asking for forgiveness)
> ```python
> try:
>     device.check_data()
> except ReadError:
>     print("No code received")
> ```
> 
> [This code snippet](https://github.com/mjg59/python-broadlink/blob/29345a129ff6c98962fccba8ada1ea049fd9ec16/cli/broadlink_cli#L134-L141) is exactly what you are looking for.

Thank you 🙏 
To double check I understand what you mean. 
My next step is to open the broadlink_cli file with a text editor and check if the commands you wrote (and those highlighted in the code snippet) are in the broadlink_cli file verbatim?

mjeshurun, 2020-11-18:
Or do I need to add the commands you wrote somewhere into the broadlink_cli file?

felipediel, 2020-11-19:
If you just want to learn a code, you can use the CLI:

`broadlink_cli --device  "0x2712 192.168.0.16 aaaaaaaaaa" --learn`

If you want to create a Python script, you can refer to the CLI to understand how the library works and how its functions should be implemented.

mjeshurun, 2020-11-19:
@felipediel 
Do you mean in python3 I need to to use this learn command:
`broadlink_cli --device "0x61a2 192.168.0.6 24:df:a7:b9:c6:89" --learn` 

And not use the earlier learn command I mentioned before?  ( `devices[0].enter_learning()` )

Because doing this didn't work. I got a `SyntaxError: invalid syntax` message.

KTibow, 2020-11-19:
@mjeshurun they mean to run it from the command line, possibly with `python3 -m ` before it.

mjeshurun, 2020-11-20:
> @mjeshurun they mean to run it from the command line, possibly with `python3 -m ` before it.

Thanks @KTibow . 
I tried running this command `python3 -m broadlink_cli --device "0x61a2 192.168.0.6 24:df:a7:b9:c6:89" --learn`
in terminal at the python-broadlink folder and also in the cli folder, but it didn't work.
Then I tried running a `python3` shell and run the learn command in the shell in two ways:
`python3 -m broadlink_cli --device "0x61a2 192.168.0.6 24:df:a7:b9:c6:89" --learn`
and 
`-m broadlink_cli --device "0x61a2 192.168.0.6 24:df:a7:b9:c6:89" --learn`
But both commands gave me the `SyntaxError: invalid syntax`message. 
I guess I'm still doing something wrong due to not understanding something.

KTibow, 2020-11-20:
As for the first one, what do you mean? Did it give any errors? You need to replace the `0x61a2 192.168.0.6 24:df:a7:b9:c6:89` with your `0xdevicesig deviceip devicemac`

mjeshurun, 2020-11-20:
> As for the first one, what do you mean? Did it give any errors? You need to replace the `0x61a2 192.168.0.6 24:df:a7:b9:c6:89` with your `0xdevicesig deviceip devicemac`

This is the command I entered in terminal and the error message I got.
```
pi@raspberrypi:~/python-broadlink/cli $ python -m broadlink_cli --device "0x61a2 192.168.0.6 24:df:a7:b9:c6:89" --learn
/usr/bin/python: No module named broadlink_cli
```

Thanks for double checking. Yes, my device info (type, ip, mac) are mentioned in the command `0x61a2 192.168.0.6 24:df:a7:b9:c6:89`.

felipediel, 2020-11-20:
`./broadlink_cli --device "0x61a2 192.168.0.6 24dfa7b9c689" --learn`

or

`python3 broadlink_cli --device "0x61a2 192.168.0.6 24dfa7b9c689" --learn`

Do not use `:` in the MAC address.

mjeshurun, 2020-11-21:
> `./broadlink_cli --device "0x61a2 192.168.0.6 24dfa7b9c689" --learn`
> 
> or
> 
> `python3 broadlink_cli --device "0x61a2 192.168.0.6 24dfa7b9c689" --learn`
> 
> Do not use `:` in the MAC address.

Thank you. I tried both commands, but I got the same error message.

```
pi@raspberrypi:~/python-broadlink/cli $ python3 broadlink_cli --device "0x61a2 192.168.0.6 24dfa7b9c689" --learn
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/broadlink/__init__.py", line 369, in send_packet
    response = cs.recvfrom(2048)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "broadlink_cli", line 105, in <module>
    dev.auth()
  File "/usr/local/lib/python3.7/dist-packages/broadlink/__init__.py", line 270, in auth
    response = self.send_packet(0x65, payload)
  File "/usr/local/lib/python3.7/dist-packages/broadlink/__init__.py", line 374, in send_packet
    raise exception(0xfffd)
broadlink.exceptions.DeviceOfflineError: The device is offline

pi@raspberrypi:~/python-broadlink/cli $ ./broadlink_cli --device "0x61a2 192.168.0.6 24dfa7b9c689" --learn
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/broadlink/__init__.py", line 369, in send_packet
    response = cs.recvfrom(2048)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./broadlink_cli", line 105, in <module>
    dev.auth()
  File "/usr/local/lib/python3.7/dist-packages/broadlink/__init__.py", line 270, in auth
    response = self.send_packet(0x65, payload)
  File "/usr/local/lib/python3.7/dist-packages/broadlink/__init__.py", line 374, in send_packet
    raise exception(0xfffd)
broadlink.exceptions.DeviceOfflineError: The device is offline
```

felipediel, 2020-11-21:
I think the IP address is wrong.

mjeshurun, 2020-11-21:
> I think the IP address is wrong.

Wonderful! You were right!!! 
I didn't notice I made a small mistake in the IP address. Sorry for this foolish mistake.

My next step was to record my appliances wireless remotes' codes. However, the RM4 Pro didn't recognize any RF codes, and, as such, it returned a "No data received..." message.
```
pi@raspberrypi:~/python-broadlink/cli $ python3 broadlink_cli --device "0x61a2 192.168.1.6 24dfa7b9c689" --learn
Learning...
No data received...
```
On the other hand, when I tried to record IR codes the RM4 Pro properly recorded them:
```
pi@raspberrypi:~/python-broadlink/cli $ python3 broadlink_cli --device "0x61a2 192.168.1.6 24dfa7b9c689" --learn
Learning...
2600680112000249610001270e130e330f1111101110110f1110110f111011301110110f1130111110100f320f330f32113011301110110f1110110f11101110110f1110111010110e120f120e1210111110110f1110110f1110110f120f1110110f1110110f1111101011100e130e120e130f110f32113111301130116162000125113011101110111010110e120e130f110f1211301110110f1130111011301130113110310f320f330e120f12110f11100f12110f1110110f11100f1111100f12110f0f130e120f120e120f120f11101111100f1110110f1110110f120f110f120f1110110f120f120e130e120e130e120f636100012710310f1110110f1110110f120f110f120f1110320f120e120f330e330f1110320f110f32103110320f320f320f330e330e330f120f1110110f320f320f3210110f120f110f120f1110320f320f330e120f3210110f1110320f320f1110110f120f110f120f120e130e330e330e330f3210000d050000000000000000000000000000
Base64: b'JgBoARIAAklhAAEnDhMOMw8RERAREBEPERARDxEQETAREBEPETARERAQDzIPMw8yETARMBEQEQ8REBEPERAREBEPERAREBARDhIPEg4SEBEREBEPERARDxEQEQ8SDxEQEQ8REBEPEREQEBEQDhMOEg4TDxEPMhExETARMBFhYgABJREwERAREBEQEBEOEg4TDxEPEhEwERARDxEwERARMBEwETEQMQ8yDzMOEg8SEQ8REA8SEQ8REBEPERAPEREQDxIRDw8TDhIPEg4SDxIPERARERAPERARDxEQEQ8SDxEPEg8REBEPEg8SDhMOEg4TDhIPY2EAAScQMQ8REBEPERARDxIPEQ8SDxEQMg8SDhIPMw4zDxEQMg8RDzIQMRAyDzIPMg8zDjMOMw8SDxEQEQ8yDzIPMhARDxIPEQ8SDxEQMg8yDzMOEg8yEBEPERAyDzIPERARDxIPEQ8SDxIOEw4zDjMOMw8yEAANBQAAAAAAAAAAAAAAAAAA'
```

Am I missing a necessary step before recording RF codes?

felipediel, 2020-11-21:
You can use `python3 broadlink_cli --device "0x61a2 192.168.1.6 24dfa7b9c689" --rfscanlearn`.

mjeshurun, 2020-11-21:
> You can use `python3 broadlink_cli --device "0x61a2 192.168.1.6 24dfa7b9c689" --rfscanlearn`.

YAYAYAYAYAY!!! This command did the trick!!! :)))))
Now I need to learn the RF codes and configure the OpenHAB binding :)))
I'm so glad I'm able to continue to the next step. 
Thank you so much for your help @felipediel @KTibow 🙏🙏🙏

felipediel, 2021-02-17:
Closing, since it is not a problem to be solved in our code base.

closed 2021-02-17 by felipediel

rodrigogml, 2021-02-22:
Sorry I reopened this thread. The problem still happens, but I think I can help to replicate. And I believe it is a code problem.

Hope this help to solve. I spent my afternoon trying to make it work and getting this error. I notice when we started to learn any code, and don't press any button on remote, the error popups in screen (about 5 seconds after the command) while the RM4 Mini still with the white led on.

But, if you press the remote control button just after entered the learning mode, it works beautifully.

**So to replicate the problem, just enter the learn mode and just wait 5 seconds.**

I'm a developer but zero skills with python. So, I suppose, the problemas has something about the delay between entering the learning mode and "grab" the code from the device while it still not ready yet.

And the error about full storage is just a "bad catch" somewhere, or the "bad code" returned crashes the write method.

Hope it drives for the solution, and btw REALLY THANKS for all the effort spent.

felipediel, 2021-02-22:
Hi. We are catching the StorageError [here](https://github.com/mjg59/python-broadlink/blob/b33bbdbbcf1270fb77f346f0e003273247d3ff2f/cli/broadlink_cli#L138). The problem does not exist in our code base. It is an implementation problem in client applications, which should use the CLI as a starting point to implement our functions correctly.

rodrigogml, 2021-02-22:
You're right! My mistake, I'm sorry!

I was able to detect the problem using the library directly on python console, but the treatment is indeed in the other project.

*By problem I meant ask to check_data() before pressing the button.