mjg59/python-broadlink #413: Support for Ankuoo NEO devices

tottka, 2020-09-04:
I've been using the Ankuoo NEO wall switches & plugs in Home Assistant for a couple years and they have worked well.  Thank you for your efforts.  Recently I wanted to use an Ankuoo NEO Pro plug which monitors electricity usage and found that the current_power_w attribute isn't returned to Home Assistant for these devices.   Funny because the original NEO's have the attribute returned but aren't capable of doing the usage.  The attribute is always 0 which I just ignore.

After looking thru the code I realized the the NEO's aren't defined in your devtype table and fall thru to the exception logic to work.    I have solved the usage calculation for the NEO Pro and would like to understand if it's possible to include the code in the master broadlink.  My experience with github has been limited to cloning so not sure how to move forward or if you would consider this change.

Here's what I have working in my environment:
The original NEO's are 0x2717 and the NEO Pro is a 0x2716.
Here's the code needed to calculate the enery usage (change in SP2)
    def get_energy(self):
        **if self.devtype == 10006:
           packet = bytearray(16)
           packet[0] = 4
           response = self.send_packet(0x6a, packet)
           err = response[0x22] | (response[0x23] << 8)
           if err == 0:
              payload = self.decrypt(bytes(response[0x38:]))
              energy =  round(float(payload[0x04]/100.0 + payload[0x05])/4.1, 2)
              return energy**
      else:
          packet = bytearray([8, 0, 254, 1, 5, 1, 0, 0, 0, 45])
          response = self.send_packet(0x6a, packet)
          check_error(response[0x22:0x24])
          payload = self.decrypt(bytes(response[0x38:]))
          if isinstance(payload[0x7], int):
              energy = int(hex(payload[0x07] * 256 + payload[0x06])[2:]) + int(hex(payload[0x05])[2:]) / 100.0
          else:
              energy = int(hex(ord(payload[0x07]) * 256 + ord(payload[0x06]))[2:]) + int(
              hex(ord(payload[0x05]))[2:]) / 100.0
         return energy

felipediel, 2020-11-04:
Hi @tottka! Thanks for the contribution! I want to add this code to the library, but first I need to understand something:
```python3
energy = round(float(payload[0x04]/100.0 + payload[0x05])/4.1, 2)
```
Why `/4.1`? Are you sure about this?

tottka, 2020-11-05:
Hey Felipe thanks for the reply,  This week, I updated to HA 0.117.3 & python 3.8.6 and noticed the formula is wrong.  Famous last words, it was working when I posted this request.  I'm gonna look into it today to see if I can figure out what changed0.  

Wish I could take credit for the code but originally I found it here https://github.com/mjg59/python-broadlink/issues/200    

I'll let you know when I get it working again.  Thanks

tottka, 2020-11-05:
Hi @ energy = (float(payload[0x06]) * 256 + payload[0x05] + float(payload[0x04])/100.0)/3.9093

felipediel, 2020-11-06:
Thanks for the response. `3.9093` still look like a magic number. How about this formula:
```python3
energy = int((payload[0x06] + payload[0x05] / 100) * 100) + payload[0x04] / 100
```
How does it work for you? (I don't have the device to test)

tottka, 2020-11-08:
Works Perfectly, thank you so much.

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/424 and https://github.com/mjg59/python-broadlink/pull/521. Thank you!

closed 2021-02-17 by felipediel