mjg59/python-broadlink #434: Broadlink LB1 (0x60c7) SmartBulb problem

forty76, 2020-09-27:
Hi, I have 1 bulb controlled by Broadlink with code lb1 0x60c7. I configured the home network wifi settings using broadlink_cli.
I tried broadlink-mqtt setting it as "bg1", it recognizes the light bulb and returns the power and brightness status but it is not possible to change the values ​​of the device.
Broadlink_cli not have valid command (--check, --turnon, --turnoff)
ex: broadlink_cli --type...... --host...... --mac......  --switch
     Traceback (most recent call last):
  File "./broadlink_cli", line 193, in <module>
    if dev.check_power():
AttributeError: 'lb1' object has no attribute 'check_power'

I tried directly in python3 this is the sequence:

import broadlink
devices = broadlink.discover(timeout=5)
devices[0].auth() ---> output :  <lb1: Broadlink LB1 (0x60c7) at 197.168.1.73:80 | c8:f7:42:8c:06:18 | Smart Bulb | Unlocked>
devices[0].get_type()  --> SmartBulb
devices[0].get_fwversion()  --> 57053
result = devices[0].get_state()  --->   OUTPUT:    []        is empty?

I change light.py file and print payload
first decode
b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00i\xd4\xfb\xff\xc7`\xee\x03\x00\xeb\x18\x06\x8cB\xf7\xc8\x01\x00\x00\x00\xaf\xbe\x00\x006W\xebF\xe4\xb2\x9f\x81\xab\xb9\x88\xb8+Z\xed\x1e'

after decode
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Can help me? thanks

forty76, 2020-09-27:
I change __init__.py to assign 0x60c7 to bg1
This is output
length packet bg1    16
 packet bg1      0xe 0x0 0xa5 0xa5 0x5a 0x5a 0xb3 0xc1 0x1 0xb 0x2 0x0 0x0 0x0 0x7b 0x7d  
length response gb1   216
response 
0x5a 0xa5 0xaa 0x55 0x5a 0xa5 0xaa 0x55 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd9 0x18 0x0 0x0 0xc7 0x60 0xee 0x3 0xff 0xf3 0x18 0x6 0x8c 0x42 0xf7 0xc8 0x1 0x0 0x0 0x0 0xb9 0xec 0x0 0x0 0xc9 0x25 0x1f 0x4c 0x49 0x2e 0xb1 0x5f 0x65 0xf6 0x43 0x80 0xb0 0xb0 0xe4 0xe5 0x5b 0x78 0x77 0x9 0x37 0xfb 0x65 0x5b 0xb4 0xdd 0x3a 0x53 0xb2 0xc2 0x99 0x95 0x26 0xb5 0x18 0x4f 0xab 0x90 0x6d 0x9 0xe3 0x7b 0x77 0xa3 0x64 0xdd 0x65 0x72 0xcf 0xd9 0x5c 0x3f 0x30 0x6b 0xa0 0xc5 0x56 0x17 0x39 0xc3 0x40 0xfb 0xb4 0xe9 0x9a 0x8f 0x7c 0x2c 0xe2 0x9 0x84 0xad 0xd0 0x2a 0x7e 0x2 0x69 0x23 0x42 0x5c 0x48 0xd4 0x1a 0x61 0xb2 0xa1 0x5f 0xdb 0xc3 0xef 0xdd 0xc8 0xe2 0x4b 0xd4 0x2d 0xdb 0x40 0x61 0xd3 0x83 0x8 0xc 0x3e 0x81 0x65 0x5b 0xb7 0x15 0xdf 0x84 0x3a 0x20 0xdb 0xfe 0x2 0x67 0xb5 0x6b 0x18 0x90 0x31 0xbf 0x4b 0x35 0x52 0xb9 0x9d 0x4 0x24 0x2 0xd1 0xbc 0x4c 0xf5 0x64 0x94 0xf3 0x4d 0x9 0xa6 0x1c 0xfb 0x3e 0x7 0xad 0x11 0xa1 0x8c 0xc8 0xbd 0x76 0xcd 0xd5 0xcc 0x59 0x8d 0x7d 0x13 0x51  

payload    0x9b 0x0 0xa5 0xa5 0x5a 0x5a 0x9a 0xea 0x3 0xb 0x8f 0x0 0x0 0x0 0x7b 0x22 0x70 0x77 0x72 0x22 0x3a 0x31 0x2c 0x22 0x62 0x72 0x69 0x67 0x68 0x74 0x6e 0x65 0x73 0x73 0x22 0x3a 0x31 0x30 0x30 0x2c 0x22 0x62 0x75 0x6c 0x62 0x5f 0x63 0x6f 0x6c 0x6f 0x72 0x6d 0x6f 0x64 0x65 0x22 0x3a 0x31 0x2c 0x22 0x62 0x75 0x6c 0x62 0x5f 0x73 0x63 0x65 0x6e 0x65 0x73 0x22 0x3a 0x22 0x5b 0x5c 0x22 0x31 0x30 0x30 0x2c 0x31 0x30 0x30 0x2c 0x32 0x34 0x30 0x30 0x2c 0x31 0x2c 0x31 0x30 0x30 0x2c 0x32 0x34 0x30 0x30 0x2c 0x30 0x5c 0x22 0x2c 0x5c 0x22 0x31 0x30 0x30 0x2c 0x30 0x2c 0x30 0x2c 0x30 0x5c 0x22 0x5d 0x22 0x2c 0x22 0x62 0x75 0x6c 0x62 0x5f 0x73 0x63 0x65 0x6e 0x65 0x22 0x3a 0x22 0x22 0x2c 0x22 0x62 0x75 0x6c 0x62 0x5f 0x73 0x63 0x65 0x6e 0x65 0x69 0x64 0x78 0x22 0x3a 0x32 0x35 0x35 0x7d 0x0 0x0 0x0  
state payload  {'pwr': 1, 'brightness': 100, 'bulb_colormode': 1, 'bulb_scenes': '["100,100,2400,1,100,2400,0","100,0,0,0"]', 'bulb_scene': '', 'bulb_sceneidx': 255}

I can use   
result = devices[0].set_state(True/False)    to switch on/off

how can use set_state to change brightness?
def set_state(
        self,
        pwr: bool = None,
        pwr1: bool = None,
        pwr2: bool = None,
        maxworktime: int = None,
        maxworktime1: int = None,
        maxworktime2: int = None,
        idcbrightness: int = None,
    ) -> dict:

KTibow, 2020-10-22:
@felipediel any way for you to implement this?

forty76, 2020-10-24:
> @felipediel any way for you to implement this?

I have change light.py, I use your bg1 code. 
this work fine for me, I can set pwr and brightness

import json
import struct

from .device import device
from .exceptions import check_error

class lb1(device):
    """Controls a Broadlink LB1."""

    def __init__(self, *args, **kwargs) -> None:
        """Initialize the controller."""
        device.__init__(self, *args, **kwargs)
        self.type = "SmartBulb"

    def get_state(self) -> dict:
        """Return the power state of the device.

        Example: `{"pwr":1,"brightness":50}`
        """
        packet = self._encode(1, b'{}')
        response = self.send_packet(0x6a, packet)
        check_error(response[0x22:0x24])
        return self._decode(response)

    def set_state(
        self,
        pwr: bool = None,        
        idcbrightness: int = None,
    ) -> dict:
        """Set the power state of the device."""
        data = {}
        if pwr is not None:
            data['pwr'] = int(bool(pwr))
        if idcbrightness is not None:
            data['brightness'] = idcbrightness
        js = json.dumps(data).encode('utf8')
        packet = self._encode(2, js)
        response = self.send_packet(0x6a, packet)
        check_error(response[0x22:0x24])
        return self._decode(response)

    def _encode(self, flag: int, js: str) -> bytes:
        """Encode a message."""
        #  The packet format is:
        #  0x00-0x01 length
        #  0x02-0x05 header
        #  0x06-0x07 00
        #  0x08 flag (1 for read or 2 write?)
        #  0x09 unknown (0xb)
        #  0x0a-0x0d length of json
        #  0x0e- json data
        packet = bytearray(14)
        length = 4 + 2 + 2 + 4 + len(js)
        struct.pack_into('<HHHHBBI', packet, 0, length, 0xa5a5, 0x5a5a, 0x0000, flag, 0x0b, len(js))
        for i in range(len(js)):
            packet.append(js[i])

        checksum = sum(packet[0x08:], 0xc0ad) & 0xffff
        packet[0x06] = checksum & 0xff
        packet[0x07] = checksum >> 8
        return packet

    def _decode(self, response: bytes) -> dict:
        """Decode a message."""
        payload = self.decrypt(response[0x38:])
        js_len = struct.unpack_from('<I', payload, 0x0a)[0]
        state = json.loads(payload[0x0e:0x0e+js_len])        
        return state

KTibow, 2020-10-24:
Can you please select all of your code, and hit the edit button?

felipediel, 2020-10-31:
I will rework the LB1 class when I find some time, but first I need to complete my pending PRs to avoid a "rebase hell".

baloghd, 2020-11-30:
> I will rework the LB1 class when I find some time, but first I need to complete my pending PRs to avoid a "rebase hell".

Hi, based on the approach from @forty76 I was able to modify the code to get most of the functionality of the bulbs working, including: 

- power on/off
- brightness
- changing colormodes and setting colors (RGB-based; for my sunrise setup I also used the webcolors library to set the colors by HTML name)
- setting the duration for these transitions
- setting hue and saturation do work, but only on every second command - more investigation is needed why that happens

Should I send a PR now, or are you planning a complete rewrite of the ` lb1` class?

felipediel, 2020-11-30:
Hi @baloghd! Please send the PR, we can make the necessary changes during the review. Thanks!

syntacticsugarglider, 2020-12-21:
I've created my own (asynchronous) implementation of this for use in a personal project (https://github.com/syntacticsugarglider/lights-broadlink), but I can't figure out how to set color temperature. @baloghd were you able to do so? passing the "colortemp" property as listed in the JSON returned from the device has no effect and doesn't even update that field in the next returned state

kylemd, 2021-01-11:
> Hi, based on the approach from @forty76 I was able to modify the code to get most of the functionality of the bulbs working, including:
>
>     * power on/off
>
>     * brightness
>
>     * changing [...]

Sorry to clog up the thread, but would you be able to make this available for the time being if possible? Trying to get my lights to work with Home Assistant and this would be very helpful!

felipediel, 2021-03-12:
Fixed with https://github.com/mjg59/python-broadlink/pull/517. Thank you @jbsky!

closed 2021-03-12 by felipediel