mjg59/python-broadlink #435: 0.15.0 Seems to have broke the cli?

timothystewart6, 2020-09-28:
Hi!  When scanning for devices in `0.14.1` I could discover 2 devices.
Now in `0.15.0` it returns no devices when discovering.  

Do the examples need to be updated?

forty76, 2020-09-28:
Can you try one device at a time?

I'm at the beginning in the use of broadlink devices. At the moment I am trying some bulbs code 0x60c7 "SmartBulb". I was able to manage them via mqtt (python-broadlink + mqtt-broadlink)

Now I have problems if I turn on two at the same time

timothystewart6, 2020-09-28:
I've tried with one device connected to the network and two.  When the call comes back from `.discover()` it's an empty list.

forty76, 2020-09-28:
What is your device? You can see on router lan device ip ? 
When I connect two devices it happens that one of the two loses data to connect to the network and I have to restore via broadlink_cli --joinwifi

timothystewart6, 2020-09-28:
They are on my network, I can ping them.  They are the RM3 and the RM4.
Like I mentioned, if I downgrade to version `0.14.1` they appear when discovering, after upgrading to `0.15.0` and using the cli in the docs (`devices = broadlink.discover(timeout=5)`) the list of devices comes back empty.

Thank you.

timothystewart6, 2020-09-29:
any ideas @felipediel?

felipediel, 2020-09-29:
1. Does it work If you specify a `local_ip_address`?
2. Does it work with `broadlink.hello("192.168.0.17")`? (replace with your device IP address)

timothystewart6, 2020-09-29:
```
import broadlink as blk

# This generates a device.
d = blk.hello("192.168.0.238")  # Device IP address.

print (d)

# This updates the device.
d.hello()  # Send hello again and update name and lock status with the response.
```

`python test.py`
<rm4: Broadlink RM mini 3 (0x5f36) at 192.168.0.238:80 | redacted | 智能遥控 | Unlocked>

@felipediel 

It does, but discover still returns an empty list.
Thank you!

felipediel, 2020-09-29:
What about `discover(timeout=10, local_ip_address="192.168.0.21")`? (replace with your local IP address)

timothystewart6, 2020-09-29:
I think I found a way to discover.  

I changed:
`parser.add_argument("--dst-ip", default="255.255.25.255", help="destination ip address to use in the discovery")`

`parser.add_argument("--dst-ip", default="192.168.0.255", help="destination ip address to use in the discovery")`

Was this related to the bind change?

timothystewart6, 2020-09-29:
Funny, I had to do something similar to get this to run in a docker container, but this worked previously on my local machine without needing to pass this into the cli.

timothystewart6, 2020-09-29:
Thanks for the help.  I will close this. More than happy to open a PR for the docs if this is not an edge case.  If someone stumbles upon this and wants me to add it to the docs and I more than happy to.

felipediel, 2020-09-29:
This is interesting. Thanks for informing this.

I don't know if this is another RM4-related issue. Normally when these devices are locked they can only be reached using broadlink.hello(). But yours is unlocked.

Please give that chance to `local_ip_address`. I need to know if this issue is related to socket.bind(), which I made optional recently to fix discover in Home Assistant.

timothystewart6, 2020-09-29:
Sure thing!  Will do.  Funny my RM4

timothystewart6, 2020-09-29:
Sure thing!  Will do.  Funny my RM3 has a `.enter_learning()` hanging off but my RM4 does not (I have 2 devices).  I will test now with just the ip.

felipediel, 2020-09-29:
Check if the devtype is supported. The rm4 class should have .enter_learning().

timothystewart6, 2020-09-29:
`[Errno 99] Cannot assign requested address`

`blk.discover(timeout=10, local_ip_address="192.168.0.238")`

timothystewart6, 2020-09-29:
```
import broadlink as blk

d = blk.hello("192.168.0.238")  # Device IP address.
print (d)
d.hello() 
devices = blk.discover(timeout=10, local_ip_address="192.168.0.150")

for device in devices:
    print(f'device.enter_learning: {device.enter_learning}')
    if device.auth():
        device.enter_learning()
```
`RM3`
```
<rm4: Broadlink RM mini 3 (0x5f36) at 192.168.0.238:80 | mac.add.ress | 智能遥控 | Unlocked>
^[[21~device.enter_learning: <bound method rm.enter_learning of <rm4: Broadlink RM mini 3 (0x5f36) at 192.168.0.238:80 | mac.add.ress. | 智能遥控 | Unlocked>>
```

`RM4`
```
<rm4: Broadlink RM4 mini (0x648d) at 192.168.0.98:80 | mac.add.ress | 智能遥控 | Unlocked>
device.enter_learning: <bound method rm.enter_learning of <rm4: Broadlink RM4 mini (0x648d) at 192.168.0.98:80 | mac.addre.ss | 智能遥控 | Unlocked>>
```

Looks good with IP and `enter_learning` on both.  So, it looks like IP is mandatory now?

felipediel, 2020-09-29:
socket.bind() was the default, but I made it optional because it was causing lots of problems. If we bind() to the wrong interface not even hello() is possible. I thought that skipping this step would work for everyone, but now we know it doesn't. I'm glad I left this as an optional parameter, so we can treat side cases, but your discover_ip_address solution seems even more interesting than using a local_ip_address, as we don't need to bind().

timothystewart6, 2020-09-29:
Nice.  Let me be sure it works.  That's how I use it inside of a docker container.  I am building a web API and a simple front end all containerized.  It's not pretty but it works.

timothystewart6, 2020-09-29:
```
import broadlink as blk

d = blk.hello("192.168.0.98")  # Device IP address.
print (d)
d.hello() 
devices = blk.discover(timeout=10, discover_ip_address="192.168.0.255")

for device in devices:
    print(f'device.enter_learning: {device.enter_learning}')
    if device.auth():
        device.enter_learning()
```

```
<rm4: Broadlink RM4 mini (0x648d) at 192.168.0.98:80 | mac.addr.ess | 智能遥控 | Unlocked>
device.enter_learning: <bound method rm.enter_learning of <rm4: Broadlink RM4 mini (0x648d) at 192.168.0.98:80 | mac.addr.ess | 智能遥控 | Unlocked>>
device.enter_learning: <bound method rm.enter_learning of <rm4: Broadlink RM mini 3 (0x5f36) at 192.168.0.238:80 | mac.addr.ess | 智能遥控 | Unlocked>>
```

Yeah, it works by only setting this. Without `discover_ip_address="192.168.0.255"`  or `local_ip_address="192.168.0.150"` it can't discover and devices because I am assuming it's bound to the wrong interface or no interface and can't broadcast.

closed 2020-10-18 by mjg59