mjg59/python-broadlink #437: Newbe: Installation not working !? Why ?

S2Revo, 2020-09-28:
Hello,
today i get a RM4 pro and i want to integrate it into my fhem... but the installation doesnt work.... 

fhem runs on a IntelNUC with Ubuntu 18..

I follow this: https://wiki.fhem.de/wiki/Broadlink

Could you please Help me ??? Thanks in advance !

I copy the Installtion:

running install
running bdist_egg
running egg_info
writing requirements to broadlink.egg-info/requires.txt
writing broadlink.egg-info/PKG-INFO
writing top-level names to broadlink.egg-info/top_level.txt
writing dependency_links to broadlink.egg-info/dependency_links.txt
reading manifest file 'broadlink.egg-info/SOURCES.txt'
writing manifest file 'broadlink.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/light.py -> build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/sensor.py -> build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/alarm.py -> build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/climate.py -> build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/device.py -> build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/remote.py -> build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/helpers.py -> build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/exceptions.py -> build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/switch.py -> build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/cover.py -> build/bdist.linux-x86_64/egg/broadlink
copying build/lib.linux-x86_64-2.7/broadlink/__init__.py -> build/bdist.linux-x86_64/egg/broadlink
byte-compiling build/bdist.linux-x86_64/egg/broadlink/light.py to light.pyc
  File "build/bdist.linux-x86_64/egg/broadlink/light.py", line 24
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/broadlink/sensor.py to sensor.pyc
  File "build/bdist.linux-x86_64/egg/broadlink/sensor.py", line 15
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/broadlink/alarm.py to alarm.pyc
  File "build/bdist.linux-x86_64/egg/broadlink/alarm.py", line 15
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/broadlink/climate.py to climate.pyc
  File "build/bdist.linux-x86_64/egg/broadlink/climate.py", line 12
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/broadlink/device.py to device.pyc
  File "build/bdist.linux-x86_64/egg/broadlink/device.py", line 18
    timeout: int = 10,
           ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/broadlink/remote.py to remote.pyc
  File "build/bdist.linux-x86_64/egg/broadlink/remote.py", line 9
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/broadlink/helpers.py to helpers.pyc
  File "build/bdist.linux-x86_64/egg/broadlink/helpers.py", line 5
    def calculate_crc16(input_data: bytes) -> int:
                                  ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/broadlink/exceptions.py to exceptions.pyc
byte-compiling build/bdist.linux-x86_64/egg/broadlink/switch.py to switch.pyc
  File "build/bdist.linux-x86_64/egg/broadlink/switch.py", line 12
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/broadlink/cover.py to cover.pyc
  File "build/bdist.linux-x86_64/egg/broadlink/cover.py", line 11
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/broadlink/__init__.py to __init__.pyc
  File "build/bdist.linux-x86_64/egg/broadlink/__init__.py", line 94
    dev_type: int,
            ^
SyntaxError: invalid syntax

creating build/bdist.linux-x86_64/egg/EGG-INFO
copying broadlink.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying broadlink.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying broadlink.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying broadlink.egg-info/not-zip-safe -> build/bdist.linux-x86_64/egg/EGG-INFO
copying broadlink.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying broadlink.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
creating 'dist/broadlink-0.15.0-py2.7.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing broadlink-0.15.0-py2.7.egg
removing '/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg' (and everything under it)
creating /usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg
Extracting broadlink-0.15.0-py2.7.egg to /usr/local/lib/python2.7/dist-packages
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg/broadlink/__init__.py", line 94
    dev_type: int,
            ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg/broadlink/alarm.py", line 15
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg/broadlink/climate.py", line 12
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg/broadlink/cover.py", line 11
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg/broadlink/device.py", line 18
    timeout: int = 10,
           ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg/broadlink/helpers.py", line 5
    def calculate_crc16(input_data: bytes) -> int:
                                  ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg/broadlink/light.py", line 24
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg/broadlink/remote.py", line 9
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg/broadlink/sensor.py", line 15
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg/broadlink/switch.py", line 12
    def __init__(self, *args, **kwargs) -> None:
                                        ^
SyntaxError: invalid syntax

broadlink 0.15.0 is already the active version in easy-install.pth

Installed /usr/local/lib/python2.7/dist-packages/broadlink-0.15.0-py2.7.egg
Processing dependencies for broadlink==0.15.0
Searching for cryptography==2.7
Best match: cryptography 2.7
Adding cryptography 2.7 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Searching for six==1.11.0
Best match: six 1.11.0
Adding six 1.11.0 to easy-install.pth file

Using /usr/lib/python2.7/dist-packages
Searching for cffi==1.14.3
Best match: cffi 1.14.3
Adding cffi 1.14.3 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Searching for ipaddress==1.0.17
Best match: ipaddress 1.0.17
Adding ipaddress 1.0.17 to easy-install.pth file

Using /usr/lib/python2.7/dist-packages
Searching for enum34==1.1.6
Best match: enum34 1.1.6
Adding enum34 1.1.6 to easy-install.pth file

Using /usr/lib/python2.7/dist-packages
Searching for asn1crypto==0.24.0
Best match: asn1crypto 0.24.0
Adding asn1crypto 0.24.0 to easy-install.pth file

Using /usr/lib/python2.7/dist-packages
Searching for pycparser==2.20
Best match: pycparser 2.20
Adding pycparser 2.20 to easy-install.pth file

Using /usr/local/lib/python2.7/dist-packages
Finished processing dependencies for broadlink==0.15.0

felipediel, 2020-09-29:
We no longer support Python 2. If you want to use it you should pick a version < 0.14.

edif30, 2020-10-21:
> We no longer support Python 2. If you want to use it you should pick a version < 0.14.

I must be missing where the installation instructions are that says this and what you need to do with pip3?

felipediel, 2020-10-21:
https://github.com/mjg59/python-broadlink/blob/bbab292bbe286ab2fb04d37aac66f91794dfac67/broadlink/__init__.py#L1

edif30, 2020-10-21:
pip3 produces this...

```$ sudo pip3 install broadlink
Collecting broadlink
  Using cached https://files.pythonhosted.org/packages/73/88/a82faf3d7fb6a94c284839a7b409a7396ad157618d721758a6530f7118cf/broadlink-0.15.0-py3-none-any.whl
Requirement already satisfied: cryptography>=2.1.1 in /usr/local/lib/python3.7/dist-packages (from broadlink) (2.9.2)
Requirement already satisfied: six>=1.4.1 in /usr/lib/python3/dist-packages (from cryptography>=2.1.1->broadlink) (1.12.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /usr/local/lib/python3.7/dist-packages (from cryptography>=2.1.1->broadlink) (1.14.0)
Requirement already satisfied: pycparser in /usr/local/lib/python3.7/dist-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.1->broadlink) (2.20)
Installing collected packages: broadlink
Successfully installed broadlink-0.15.0
$
$
$ ./broadlink_discovery
Traceback (most recent call last):
  File "./broadlink_discovery", line 5, in <module>
    import broadlink
ImportError: No module named broadlink
$

felipediel, 2020-10-21:
The shebang is outdated for broadlink_discovery. I made a [fix](https://github.com/mjg59/python-broadlink/pull/455) for this. Thanks for reporting the issue.

closed 2020-10-24 by mjg59