mjg59/python-broadlink #440: Add support for Broadlink SP4M

paulie-g, 2020-10-05:
I've picked up the following Broadlink SP4M-US device: https://www.amazon.com/gp/product/B08732M6TG

I initially set it up via the NOAPP process in Google Home (connect to it in AP mode, fill in SSID/password/email on walled garden page, receive code, enter code in Google Home).

My attempts to get it working with this module have yielded the following results:
```
>>> d=broadlink.discover(2)
>>> d
[<device: None None (0x648b) at 192.168.1.149:80 | 24:df:a7:36:f3:21 | 25739 | Unlocked>]

>>> s=broadlink.sp4(d.host, d.mac, d.devtype)
>>> s
<sp4: None None (0x648b) at 192.168.1.149:80 | 24:df:a7:36:f3:21 | None | Unlocked>

TypeError: 'NoneType' object is not callable
>>> s.hello()
True
>>> s.get_state()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/paul/sw/broadlink/bl/lib/python3.8/site-packages/broadlink/switch.py", line 278, in get_state
    return self._decode(response)
  File "/home/paul/sw/broadlink/bl/lib/python3.8/site-packages/broadlink/switch.py", line 306, in _decode
    check_error(response[0x22:0x24])
  File "/home/paul/sw/broadlink/bl/lib/python3.8/site-packages/broadlink/exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthorizationError: [Errno -7] Control key is expired
>>> s.get_fwversion()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/paul/sw/broadlink/bl/lib/python3.8/site-packages/broadlink/device.py", line 223, in get_fwversion
    check_error(response[0x22:0x24])
  File "/home/paul/sw/broadlink/bl/lib/python3.8/site-packages/broadlink/exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthorizationError: [Errno -7] Control key is expired
>>> s.auth()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/paul/sw/broadlink/bl/lib/python3.8/site-packages/broadlink/device.py", line 185, in auth
    check_error(response[0x22:0x24])
  File "/home/paul/sw/broadlink/bl/lib/python3.8/site-packages/broadlink/exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthenticationError: [Errno -1] Authentication failed
```

I've seen conflicting info reading other issues. Otoh, it is recommended to factory reset; however, it was also stated that as long as device shows as unlocked, all is well. There's also some mention of an update that fixes 'Control key is expired' in a home assistant issue somewhere that I've read but can't find again.

Does this device need to have support added, or am I missing something (like factory reset/setup via broadlink.setup() despite device showing as Unlocked)? As I don't understand how Broadlink authentication works (would be useful as a readme so ppl like me can RTFM), I thought I'd ask before going down a rabbithole.

forty76, 2020-10-06:
Hi, I am newbie to python broadlink.

First:
In file     __init__.py  is declared all type. Your SP4M-US is not declared, but you can look SP4-EU
       0x7579: (sp4, "SP4L-EU", "Broadlink"),
You can try to add line
      0x648b: (sp4, "SP4L-EU", "Broadlink"),      
many broadlink have different ID code but work in the same way

Second:
The right sequence to comunicate to device is

import broadlink
devices = broadlink.discover(timeout=5)
devices[0].hello()     ---->>>>>   Direct device discovery.   Useful if the device is locked. This module not give you device access

devices[0].auth()     ----->>>>>  without this "get/set" not work, you need this step
devices[0].get_state()

In file switch.py you can look all SP4 module you can use

I hope I have been helpful

felipediel, 2020-10-09:
`devices[0].hello()` is for pings and updates, you can omit this part.

closed 2020-10-18 by mjg59