mjg59/python-broadlink #442: Unable to extract RF Codes

dfarrall, 2020-10-07:
Posting here as learning RF codes isn't supported in Home assistant, and unsure where the issue lies.
I've been trying to get RF sending working in HA via a broadlink RM Pro 4. 

I've got broadlink_cli up and running and extracted a few codes. Each time I extract one it's different, despite being the same button press. Is that expected behaviour?
Example of a code i've extracted below.

`Found RF Frequency - 1 of 2!
You can now let go of the button
Press enter to continue...
To complete learning, single press the button you want to learn
Found RF Frequency - 2 of 2!
b0c0a401129c06009d32170c170c170d160d0b18170c170c0b180b180b190b180b18170c170c0b180b180b18170c0b180b180b180c17180c0b180b180b180b180b180b180c170c170c170c180b180b18170c0b180b180c17170d9c32170d160d160d160d0a19160d160d0b180b180b180b180c17170c170d0b180b180b18170c0b180b180b180b18170c0b190b180b180b180b180b180b180b180b180c180b180b18170c0b180b180b18170d9c33160d160d170c170c0b18170c170c0b190a190b180b180b18170c170c0b180b180b18180c0a190b180b180b18170c0b180b180b180b180b190a190b180b180b180b180b180b18170c170c170c170d0b0001089c32170c170c170c170c0b18170d160d0a190b180b180b180b18170c170c0b180b180c18160d0b180b180b180b18170c0b180b190a180c180b180b180b180b180b180b180b180b18170d170c170c170c0b0001079d32170c170c170c170d0a18170d160d0a190a190b180b180b18170c170c0b180b190b18170c0b180b180b180b18170c0b180b180b190b180b180b180b180b180b180b180b180b18170c180c170c170c0b0005dc0000
Base64: b'sMCkARKcBgCdMhcMFwwXDRYNCxgXDBcMCxgLGAsZCxgLGBcMFwwLGAsYCxgXDAsYCxgLGAwXGAwLGAsYCxgLGAsYCxgMFwwXDBcMGAsYCxgXDAsYCxgMFxcNnDIXDRYNFg0WDQoZFg0WDQsYCxgLGAsYDBcXDBcNCxgLGAsYFwwLGAsYCxgLGBcMCxkLGAsYCxgLGAsYCxgLGAsYDBgLGAsYFwwLGAsYCxgXDZwzFg0WDRcMFwwLGBcMFwwLGQoZCxgLGAsYFwwXDAsYCxgLGBgMChkLGAsYCxgXDAsYCxgLGAsYCxkKGQsYCxgLGAsYCxgLGBcMFwwXDBcNCwABCJwyFwwXDBcMFwwLGBcNFg0KGQsYCxgLGAsYFwwXDAsYCxgMGBYNCxgLGAsYCxgXDAsYCxkKGAwYCxgLGAsYCxgLGAsYCxgLGBcNFwwXDBcMCwABB50yFwwXDBcMFw0KGBcNFg0KGQoZCxgLGAsYFwwXDAsYCxkLGBcMCxgLGAsYCxgXDAsYCxgLGQsYCxgLGAsYCxgLGAsYCxgLGBcMGAwXDBcMCwAF3AAA'`

Which i've then added to a script to send the RF signal back out. Clicking this script does absolutely nothing.

`raise_blinds:
  sequence:
    - service: remote.send_command
      data:
        entity_id: remote.broadlink_rm_pro_remote
        command: b64:b'sMCkARKcBgCdMhcMFwwXDRYNCxgXDBcMCxgLGAsZCxgLGBcMFwwLGAsYCxgXDAsYCxgLGAwXGAwLGAsYCxgLGAsYCxgMFwwXDBcMGAsYCxgXDAsYCxgMFxcNnDIXDRYNFg0WDQoZFg0WDQsYCxgLGAsYDBcXDBcNCxgLGAsYFwwLGAsYCxgLGBcMCxkLGAsYCxgLGAsYCxgLGAsYDBgLGAsYFwwLGAsYCxgXDZwzFg0WDRcMFwwLGBcMFwwLGQoZCxgLGAsYFwwXDAsYCxgLGBgMChkLGAsYCxgXDAsYCxgLGAsYCxkKGQsYCxgLGAsYCxgLGBcMFwwXDBcNCwABCJwyFwwXDBcMFwwLGBcNFg0KGQsYCxgLGAsYFwwXDAsYCxgMGBYNCxgLGAsYCxgXDAsYCxkKGAwYCxgLGAsYCxgLGAsYCxgLGBcNFwwXDBcMCwABB50yFwwXDBcMFw0KGBcNFg0KGQoZCxgLGAsYFwwXDAsYCxkLGBcMCxgLGAsYCxgXDAsYCxgLGQsYCxgLGAsYCxgLGAsYCxgLGBcMGAwXDBcMCwAF3AAA
        device: Living Room Blinds`

Using the native broadlink app i can learn and send codes for this device fine.
Not sure if i'm doing something wildly wrong here?

Thanks

CyrisXD, 2020-10-09:
I had the same issue, but got it working sending the command as: 

`command: b64:"sMCkARKcBgCdMhcMFwwXDRYNCxgXDBcMCxgLGAsZCxgLGBcMFwwLGAsYCxgXDAsYCxgLGAwXGAwLGAsYCxgLGAsYCxgMFwwXDBcMGAsYCxgXDAsYCxgMFxcNnDIXDRYNFg0WDQoZFg0WDQsYCxgLGAsYDBcXDBcNCxgLGAsYFwwLGAsYCxgLGBcMCxkLGAsYCxgLGAsYCxgLGAsYDBgLGAsYFwwLGAsYCxgXDZwzFg0WDRcMFwwLGBcMFwwLGQoZCxgLGAsYFwwXDAsYCxgLGBgMChkLGAsYCxgXDAsYCxgLGAsYCxkKGQsYCxgLGAsYCxgLGBcMFwwXDBcNCwABCJwyFwwXDBcMFwwLGBcNFg0KGQsYCxgLGAsYFwwXDAsYCxgMGBYNCxgLGAsYCxgXDAsYCxkKGAwYCxgLGAsYCxgLGAsYCxgLGBcNFwwXDBcMCwABB50yFwwXDBcMFw0KGBcNFg0KGQoZCxgLGAsYFwwXDAsYCxkLGBcMCxgLGAsYCxgXDAsYCxgLGQsYCxgLGAsYCxgLGAsYCxgLGBcMGAwXDBcMCwAF3AAA"

dfarrall, 2020-10-09:
> I had the same issue, but got it working sending the command as:
>
> `command: [...]

Thanks, i had tried enclosing with quotes if that's what your highlighting? No luck unfortuntely.

CyrisXD, 2020-10-09:
@dfarrall did you remove the **b'** and trailing single quote at the end? If it's still not working it may be a rolling code. Capture the code a few times and see if it changes at all.

dfarrall, 2020-10-10:
> @dfarrall did you remove the **b'** and trailing single quote at the end? If it's still not working it may be a rolling code. Capture the code a few times and see if it changes at all.

Gave that a go and no joy.

It may be arolling code yes, mentioned that it's different each capture in the original description.
Using the broadlink App i can capture and relay without issue though, it's purely translating that into HA.

felipediel, 2021-02-17:
You need to remove the quotes:

### Before
```yaml
command: b64:"sMCkARKcBgCdMhcMFwwXDRYNCxgXDBcMCxgLGAsZCxgLGBcMFwwLGAsYCxgXDAsYCxgLGAwXGAwLGAsYCxgLGAsYCxgMFwwXDBcMGAsYCxgXDAsYCxgMFxcNnDIXDRYNFg0WDQoZFg0WDQsYCxgLGAsYDBcXDBcNCxgLGAsYFwwLGAsYCxgLGBcMCxkLGAsYCxgLGAsYCxgLGAsYDBgLGAsYFwwLGAsYCxgXDZwzFg0WDRcMFwwLGBcMFwwLGQoZCxgLGAsYFwwXDAsYCxgLGBgMChkLGAsYCxgXDAsYCxgLGAsYCxkKGQsYCxgLGAsYCxgLGBcMFwwXDBcNCwABCJwyFwwXDBcMFwwLGBcNFg0KGQsYCxgLGAsYFwwXDAsYCxgMGBYNCxgLGAsYCxgXDAsYCxkKGAwYCxgLGAsYCxgLGAsYCxgLGBcNFwwXDBcMCwABB50yFwwXDBcMFw0KGBcNFg0KGQoZCxgLGAsYFwwXDAsYCxkLGBcMCxgLGAsYCxgXDAsYCxgLGQsYCxgLGAsYCxgLGAsYCxgLGBcMGAwXDBcMCwAF3AAA"
```

### After
```yaml
command: b64:sMCkARKcBgCdMhcMFwwXDRYNCxgXDBcMCxgLGAsZCxgLGBcMFwwLGAsYCxgXDAsYCxgLGAwXGAwLGAsYCxgLGAsYCxgMFwwXDBcMGAsYCxgXDAsYCxgMFxcNnDIXDRYNFg0WDQoZFg0WDQsYCxgLGAsYDBcXDBcNCxgLGAsYFwwLGAsYCxgLGBcMCxkLGAsYCxgLGAsYCxgLGAsYDBgLGAsYFwwLGAsYCxgXDZwzFg0WDRcMFwwLGBcMFwwLGQoZCxgLGAsYFwwXDAsYCxgLGBgMChkLGAsYCxgXDAsYCxgLGAsYCxkKGQsYCxgLGAsYCxgLGBcMFwwXDBcNCwABCJwyFwwXDBcMFwwLGBcNFg0KGQsYCxgLGAsYFwwXDAsYCxgMGBYNCxgLGAsYCxgXDAsYCxkKGAwYCxgLGAsYCxgLGAsYCxgLGBcNFwwXDBcMCwABB50yFwwXDBcMFw0KGBcNFg0KGQoZCxgLGAsYFwwXDAsYCxkLGBcMCxgLGAsYCxgXDAsYCxgLGQsYCxgLGAsYCxgLGAsYCxgLGBcMGAwXDBcMCwAF3AAA
```

Closing, since it is not a problem to be solved in our code base.

closed 2021-02-17 by felipediel