mjg59/python-broadlink #449: Broadlink device unlock

astra1, 2020-10-10:
I think broadlink heard my feature request and officially added unlock device toggle in version 1.5.03. Now unlocking will be simpler :)

[image]

Beendiis, 2020-10-31:
I have version 1.5.10 and i can't find "lock device " :(  Maybe is only on your version ?

KTibow, 2020-10-31:
Nope, I have it too. What kind of device do you have?

Beendiis, 2020-10-31:
RM4 pro. On which menu is "device info" , "Firmware update", .....,

KTibow, 2020-11-01:
@Beendiis I have a SP4-US, and it shows up. Anyway wrong place, create a new issue, and @astra1 you might want to consider closing this issue so it doesn't have 
[image]
issues.

closed 2020-11-05 by astra1

CyborgVette, 2024-02-10:
I found the unlock toggle but it still doesn't really unlock. The Home Assistant and the Hubitat won't connect. Do I use the AP mode or the Smart mode install?