mjg59/python-broadlink #450: understanding ir packet

alonizyl, 2020-10-10:
hi, im trying to use the remote as a trigger in my code, every time i press the button i get different results is there and way to decrypt the ir packet and know if it the same button?

[image]

alonizyl, 2020-10-10:
> hi, im trying to use the remote as a trigger in my code, every time i press the button i get different results is there and way to decrypt the ir packet and know if it the same button?
> 
> [image]

felipediel, 2021-02-17:
Hi. This is normal. Some codes have a toggle bit, so the device is able to differentiate a short press from a long press for example. The codes are not encrypted in the image you sent, they are in raw byte format. If you want to understand how this works you can read the [protocol](https://github.com/mjg59/python-broadlink/blob/master/protocol.md).

Not something to be fixed. Closing.

closed 2021-02-17 by felipediel