mjg59/python-broadlink #452: temperature reading always returns 0.0 on rm4 pro

hi-flyer, 2020-10-15:
I am on Ubuntu 18.04 with a RM4 pro unit. To get discovery to work I had to use 
sudo iptables -A INPUT -p udp --sport 80 -j ACCEPT (form one of the thias comments post).
all is working well except that the temperature reading always comes back as 0.0. I am using python3 with the following output:
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink
>>> devices = broadlink.discover(timeout=5)
>>> print (devices[0])
<rm4: Broadlink RM4 pro (0x61a2) at 192.168.1.231:80 | 24:df:a7:e4:7c:43 | 智能遥控 | Unlocked>
>>>devices[0].auth()
True
>>> devices[0].check_temperature()
0.0
>>> 
any suggestion on how getting the temperature from the device?

when I used the broadlink phone app, it showed the temperature correctly.

hi-flyer, 2020-10-15:
forgot using version = '0.15.0'

hi-flyer, 2020-10-24:
According to Iphone Broadlink app, I am using Firmware V52078, SDK 2.15, Plug-in version 1.5.4.1

anyone got ideas why I am not getting the temperature and Humidity but the Broadlink app is providing temp and humidity figures?

hi-flyer, 2020-10-27:
I was confused. I needed to have the special cord with the temp and humidity sensor in it. I bought without the sensors. Apparently the Broadlink app  is using a local temp & humidity for the internet.

closed 2020-10-27 by hi-flyer

pyrite357, 2023-07-06:
> I was confused. I needed to have the special cord with the temp and humidity sensor in it. I bought without the sensors. Apparently the Broadlink app is using a local temp & humidity for the internet.

This happened to me as well!! To top it off, Alexa auto-adds the RM4 Pro's "Temp Sensor" as an independent device called "BroadLink Temperature Sensor HTS2" in the Alexa device list. Leaving here to help anyone down the road ...