mjg59/python-broadlink #453: broadlink.discover() returns nothing

chri-nak, 2020-10-16:
HW used:
0x4ead: (hysen, "HY02B05H", "Hysen"), (Beok TDS21wp)

Steps to reproduce:

broadlink.setup('myssid', 'mynetworkpass', 3)
devices = broadlink.discover(timeout=5)
 
devices is empty: []

Communication with the device is possible with:

device = broadlink.hysen((strHost,80),macbytes,0x4ead)
device.auth()
data = device.get_full_status()

full status is beeing received.

felipediel, 2020-10-31:
If you `device.set_lock(False)`, does it show up?

felipediel, 2021-02-17:
Hello. Two possibilities:
1. Your devices are locked. When the devices are locked they cannot be discovered. To unlock:
```python3
import broadlink as blk
>>> d = blk.hello('192.168.0.16')  # Example device
>>> d.auth()
>>> d.set_lock(False)
```
2. You have multiple network interfaces and didn't specify an appropriate local_ip_address or discover_ip_address, so the OS is binding the socket to the wrong interface. The best way to discover is using the L3 broadcast address:
```python3
>>> import broadlink as blk
>>> devs = blk.discover(discover_ip_address='192.168.0.255')
>>> devs
```
I am closing old issues. If it doesn't work, you can reopen it.

closed 2021-02-17 by felipediel