mjg59/python-broadlink #458: Support sending cloud ping packets (packet type 0x01)

marcan, 2020-10-23:
Some Broadlink devices (at least the RM3 Minis running recent firmware) will reboot periodically if they can't establish a connection to the Broadlink cloud. This is undesirable, as it leads to transient communication loss.

This can be preventing by sending packet type 0x01 to the devices (this is the cloud alive ping packet, but they don't care if it comes in via the cloud or not). This works both unicast and broadcast. The only thing the device cares about is the packet type.

Literally, this is all it takes (for the broadcast case):
```
echo -ne '\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\x01\0\0\0\0\0\0\0\0\0' | nc -bu 192.168.7.255 80 -w 0
```

It would be nice if python-broadlink supported sending this unicast to devices, as well as broadcast to reduce traffic on networks with multiple devices.

For the RM3 Minis, the timeout is 3 minutes, though I think there is a secondary timeout on top of that (currently checking).

KTibow, 2020-10-23:
So how would this be implemented? Something like `device.send_keep_alive()`? Should it be sent every time a command is sent?

marcan, 2020-10-23:
It would just be another method. No need to send it with every command; the user would be responsible for sending it periodically. It needs to be sent every 1-2 minutes to stop the device from rebooting and becoming momentarily unavailable.

I guess hypothetically python-broadlink could spawn a thread to take care of this for the user, but this seems like too much automagic behind the scenes to me. Different users will probably have different ways of scheduling things to run on a timer...

KTibow, 2020-10-23:
@felipediel does this look easy to you?

marcan, 2020-10-24:
Thinking further about this, since this works in both broadcast and unicast, and the device type or any other device info does not matter, this probably makes more sense as a global function (which could be also callable from a device method for convenience).

Then a user like Home Assistant could default to sending keepalives to each individual device found/configured, but also let the user explicitly specify a list of keepalive targets, where they could just plug in a broadcast address to reduce WiFi traffic when there are many devices on the same network (this only works when you're on the same subnet as the devices, so it should not be the default/forced for cases where people use routing).

litinoveweedle, 2020-10-24:
Please consider, that you have to choose correct broadcast IP address in situation, when HA has several network interfaces and/or network interface with several IP subnets on them. There were several bugs in the past HA not handling such network setups well.

So if HA is not already aware of IP broadcast IP for each and every Broadlink device (could be different), than it is required to use IP address of all Broadlink devices + corresponding HA network interface IP subnet mask to calculate correct IP broadcast address to sent this keep-alive packet to (for each Broadlink device). Than it should compare/aggregate these IPs (and most of the time these will be same as all Broadlink devices typically reside in same IP subnet, but this could not guaranteed) and only than it is enough to sent one such packet per each used IP subnet.

litinoveweedle, 2020-10-24:
> Thinking further about this, since this works in both broadcast and unicast, and the device type or any other device info does not matter, this probably makes more sense as a global function (which [...]

There  are several HA integration using broadlink library in slightly different ways (I have at least two - Floureon thermostat and SmartIR), so when keepalive packet is sent to broadcast IP you successfully keeping ALL Broadlink devices from rebooting, which is very helpful. ;-)

marcan, 2020-10-24:
It's important to note that you can only determine the broadcast address, and actually send the broadcasts, if you're on the same network segment (i.e. some interface on the host has an address covering that device). Most routers won't let you route to broadcast addresses.

So the "smart" way to do this automatically this would be to collect all target IPs, and try to find local interfaces within the same subnets, and then send packets to all broadcast addresses of those interfaces, plus unicast packets to any target IPs not covered by local subnets.

litinoveweedle, 2020-10-24:
> So the "smart" way to do this automatically this would be to collect all target IPs, and try to find local interfaces within the same subnets, and then send packets to all broadcast addresses of those interfaces, plus unicast packets to any target IPs not covered by local subnets.

More and more I am thinking about the implementation, I would propose to simplify it to:
- provide inside library automatic keep-alive function for each given IP of discovered/configured device automatically using unicast IP of the device
- provide way to use library easily in the HA to create integration with simple configurable service, which user can periodically call with parameter of the required target IP address (could be unicast or broadcast IP)
- in both cases library shall select correct source IP / bind interface (as implemented now for discovery -> such function is already present in the OS networking stack so not defined source IP / bind interface will cause OS to automatically select proper IP and outgoing interface based on local route table)

I hope, that such change should not be hard to implement. ;-)

felipediel, 2020-10-25:
I made a minimalist version for testing: https://github.com/mjg59/python-broadlink/pull/461. We can improve it after the tests.

felipediel, 2020-10-26:
I think that binding to a local_ip_address, setting SO_BROADCAST and sending this packet to ("255.255.255.255", 80) would do the trick for 99.9% of the people. We can handle side cases with a [x] keep me alive option in the UI, which adds the device to a list of subscribers.

felipediel, 2021-03-12:
Added. Closing.

closed 2021-03-12 by felipediel