mjg59/python-broadlink #459: Assigned specific frequency for RF learning

a211212abc, 2020-10-24:
There're two main frequency range - 315 MHz and 433 MHz. 

When I want to learning a 315 MHz RF, it works fine if I assign the frequency in the app. But if I do sweep frequency through this command line tool, it seems never find 315 MHz. 

Could we support letting user have the option to assign the specific frequency? That would be greatly helpful. 

Thanks

felipediel, 2020-10-29:
Are you sure it doesn't not work? Are you holding the button to sweep the frequency, and then pressing the button again to capture the code? I don't have 315MHz remotes to test this.

a211212abc, 2020-10-29:
I'm sure that it couldn't lock to 315MHz automatically using the command line tool, but I finally worked around to get the hex by capturing packages from the official app.

Chanete, 2023-01-24:
> I'm sure that it couldn't lock to 315MHz automatically using the command line tool, but I finally worked around to get the hex by capturing packages from the official app.

How did you capture the packages from the offical app wireshark? . I'm stuck wih 315mhz rf remote. Works with the app, but I'm unable to capture with this api.

thx!

closed 2024-04-21 by felipediel