mjg59/python-broadlink #460: AuthorizationError: Control key is expired - Developing Plugin

steffenu, 2020-10-24:
Greetings ...
Im trying to develop a plugin for Eventghost ...

Using the previous release try to make a rm 3 mini compatible--> 0x5f36 ...
still rocking pyhton 2.7

I can run a single action , like learn a single command or send a single command
but any follow up command throws

AuthorizationError: Control key is expired

Any chance you can lend me a hand figuring out the root of this problem ?

Greetings

felipediel, 2020-10-24:
You need to authenticate first.
```python3
import broadlink as blk

d = blk.hello("192.168.0.17")  # Example device
d.auth()
```

steffenu, 2020-10-25:
> You need to authenticate first.
> 
> ```python
> import broadlink as blk
> 
> d = blk.hello("192.168.0.17")  # Example device
> d.auth()
> ```

Im using the previous release.
For my understanding ... the hello() method is necessary? 

So basically the previous release isnt able at all to communicate with stuff on the newer firmware ?

I still have an older rm pro plus ... which works perfectly ... and it doesnt have any auth stuff .... 
Also was able to use the app and the library simultanisously ... until I updated out of curiousity :D... now it says I need to upgrade the firmware on the app .... but I better not do that I guess...

steffenu, 2020-10-25:
ok I pulled over the `hello()` method  into version 0.1.4.1
which needs` xdiscover()` s and` scan()`  and a slightly modifed gendevice()

now im getting this error running the discover method ;

```
  File "C:\ProgramData\EventGhost\plugins\Broadlink\__init__.py", line 209, in discoverDevices
    discovered = broadlink.discover(timeout=conTimeout)
  File "C:\ProgramData\EventGhost\plugins\Broadlink\broadlink\__init__.py", line 385, in discover
    return [gendevice(*resp) for resp in responses]
  File "C:\ProgramData\EventGhost\plugins\Broadlink\broadlink\__init__.py", line 1414, in scan
    devtype = response[0x34] | response[0x35] << 8
TypeError: unsupported operand type(s) for <<: 'str' and 'int'
```

Do you know what I can do here ?

felipediel, 2020-10-25:
You don't need blk.hello(). This line is just to examplify the instantiation of the controller. You can do it the old way if you want. All you need is d.auth().

If you want to adapt this code to Python 2, it is not only an annotations thing. We reworked all the code. Python 2 treats bytes as strings. You will need a lot of conversions.

steffenu, 2020-10-25:
> You don't need blk.hello(). This line is just to examplify the instantiation of the controller. You can do it the old way if you want. All you need is d.auth().
>
> If you want to adapt this code to [...]

Ok now that makes sense if py3 treats bytes differently ... 
I was wondering how the latest code even works ....

Regarrding the problem. of the zeros shifting from left to right ... 
can you give me a hint on where to start looking ...

I suspect the problem is somewhere going through my code from the check data which I already "patched"  a couple times (looking at open issues):

that is wrongly shifting zeros to the left side of the bytestring

**0000**260046008f941238123813381213121311141213111411391238133811141213111412131213123813381139121312131114121312131312121411141238123812381338123811000d05

-->

260046008f941238123813381213121311141213111411391238133811141213111412131213123813381139121312131114121312131312121411141238123812381338123811000d05**0000**

You could save me a major headache if you could tell me what I can do to / need to understand to figure this out ...

I dont expect you to solve it for me :D ... but im clueless where to look. @felipediel 

```
    def check_data(self):
        packet = bytearray(self._request_header)
        packet.append(0x04)
        response = self.send_packet(0x6a, packet)

        try:
            check_error(response[0x22:0x24])

        except (ReadError, StorageError):
            pass

        err = response[0x22] | (response[0x23] << 8)
        if err == 0:
            payload = self.decrypt(bytes(response[0x38:]))
            return payload[0x04:]

```

I wont try rewriting everything to pyhthon 3 ... thats way too much ...

But I already got the rm pro plus working and ...  and the rm 3 mini is basically also done if it weren't for this 
shifting problem ....  

I mean the recording and all works ... there is just some switcheroo going on that I dont know the cause of

felipediel, 2020-10-25:
We use a padding mechanism to encrypt the message for AES. We append 0's to the payload when necessary so its length will always be multiple of 16.

steffenu, 2020-10-25:
> We use a padding mechanism to prepare the message for AES encryption. We append 0's to the payload when necessary so its length will always be multiple of 16.

Thank you this helped me figure it out .... all working now THANK YOU !!!!!!!!!

closed 2020-10-25 by steffenu