mjg59/python-broadlink #462: Control key is expired

SAMLeex, 2020-10-26:
Hi all, following are my codes:

```
devices = broadlink.discover(timeout=5)
print(devices)
print(devices[1].auth())
devices[1].send_data(<RF code>)
```

Device:
`<rm: Broadlink RM pro+ (0x279d) at 192.168.2.xx:80 | xx:xx:xx:xx:xx | 智能遥控 | Unlocked>`

However, the following error occurs when sending data:
```
  File "/usr/local/lib/python3.8/site-packages/broadlink/remote.py", line 31, in send_data
    check_error(response[0x22:0x24])
  File "/usr/local/lib/python3.8/site-packages/broadlink/exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthorizationError: [Errno -7] Control key is expired
```

Do i need to factory reset my device in this case? Thanks!

steffenu, 2020-10-26:
You are using the latest release from here and the lastest Firmware ?
does the same thing happen if you do it like this 

```
import broadlink as blk

d = blk.hello("192.168.0.17")  # your device ip ?
d.auth()
d.send_data(<RF code>)
```

SAMLeex, 2020-10-26:
The same happens with your above code. @steffenu

KTibow, 2020-10-26:
@SAMLeex Is it unlocked? Try factory resetting and reconnecting, not letting it to register to the cloud.

steffenu, 2020-10-26:
> The same happens with your above code. @steffenu

could you try python 3.7 ... I remember using 3.8 did throw me some random errors that I hadnt previously ... 
One more thing I nottices your are pritinting devices[1] ? you have multiple devices ?

felipediel, 2020-10-29:
You should not receive this error with the device unlocked. I have no idea what's going on. Are you sure you are authenticating before sending the command? This never happened before.

hi-flyer, 2020-10-30:
I notice SAMLeex you are using devices[1] which I think means you have 2 broadlink devices. should you really be using devices[0]?

felipediel, 2021-03-12:
I will consider this case closed. If you still have the problem, reopen it.

closed 2021-03-12 by felipediel