mjg59/python-broadlink #465: Library support for Fan coil Thermostat, Beok TGR87G-WIFI-AC2?

ciafe, 2020-10-30:
Hi, 

I have a Fan coil Thermostat, Beok TGR87G-WIFI-AC2, which is not supported by the library. Is the are any change, or plan, to add support to this Broardlink Thermostat?

I debugged a little bit the library with my Thermostat and “devtype” reported is 0x4F5B. I can provide more debug information from the device, if needed! But I don't know what else should be checked, or done, to add support to it.

Thanks for your support!

ciafe, 2020-11-03:
Hi, 

I'm trying to develope the support for this Thermostat, and currently using the Hysen climade device as a reference. I'm able to send requests with the _send_request()_ and some are working, like _set_power()_, but most of the reading information, like _get_full_status()_, or _get_temp()_ is readback but not valid (temps, times, ...).

I don't know where to get the information on how the _response_payload_ is coded and structured in the frame. Is there any Broadlink documentation available? Could someone guide me on how to get the payload information codification, so I can implemented the decodification and usage in the library? How people does for other Broadlink devices?

Thanks a lot for the support!

felipediel, 2020-11-12:
Hi there. Thanks for your dedication to this.

The [official documentation](https://docs.ibroadlink.com/public/appsdk_en/appsdk_01/) serves us better to understand what their  are doing than how they are doing it. There is no shortcut, you need to look at the bytes and try to understand what they mean.

One thing I usually do is to compare an old message with the new message to understand the changes. It usually doesn't change much. A pattern that I am seeing over and over: in the new devices the payload is being preceded by its length, example:
```python3
b'\x04\x00\x00\x00'
```
Becomes:
```python3
b'\x04\x00\x04\x00\x00\x00'
```
This is what happened to RM2/RM4 and SP4/SP4B, and it will continue to happen because this is how the new version of the protocol works. The payload needs to be preceded by its length now. So that's where I would start looking.

nortnil, 2022-07-29:
lockmode = (payload[3] & 0x0f);
lock = (payload[3] >> 4);
power =  (payload[4] & 0x01);
valve =  ((payload[4] >> 4) & 0x01);
controlMode = payload[5];//1:ventilate;2:cool;3:heat
windSpeed = payload[6];//1:low;2:middle;3:high;4:auto
pv = payload[7];
sv = payload[8];
dif = payload[9];
adj = (payload[10] > 127 ? -((256 - payload[10]) / 10) : payload[10] / 10);
sh1 = payload[11];
sl1 = payload[12];
sh2 = payload[13];
sl2 = payload[14];
fanMode = payload[15];
freMode = payload[16];
hour = payload[17];
minute = payload[18];
second = payload[19];
week = payload[20];
// payload[21];//reserve
loop = payload[22];
openHour1 = payload[23] & 0xff;//enable in bit7
openMinute1 = payload[24];
closeHour1 = payload[25] & 0xff;//enable in bit7
closeMinute1 = payload[26];
openHour2 = payload[27] & 0xff;//enable in bit7
openMinute2 = payload[28];
closeHour2 = payload[29] & 0xff;//enable in bit7
closeMinute2 = payload[30];