mjg59/python-broadlink #467: discovery() found instances

steffenu, 2020-10-31:
So im having a rm3 mini and  rm pro plus running

doing the discovery method returns a bunch of instances ...

`         [<eg.UserPluginModule.Broadlink.broadlink.rm instance at 0x08C80940>, <eg.UserPluginModule.Broadlink.broadlink.rm4 instance at 0x0D559A08>, <eg.UserPluginModule.Broadlink.broadlink.rm instance at 0x08C7BF80>, <eg.UserPluginModule.Broadlink.broadlink.rm4 instance at 0x0D57F0A8>, <eg.UserPluginModule.Broadlink.broadlink.rm instance at 0x08CA5CD8>, <eg.UserPluginModule.Broadlink.broadlink.rm4 instance at 0x0D57F990>, <eg.UserPluginModule.Broadlink.broadlink.rm instance at 0x0D570A08>, <eg.UserPluginModule.Broadlink.broadlink.rm4 instance at 0x08CB63C8>, <eg.UserPluginModule.Broadlink.broadlink.rm instance at 0x08CAAD50>]`

when i loop through them those are all the same 2 devices .... 
`         ('model :', 'RM pro+')
         ('host :', ('192.168.1.108', 80))
         ('model :', 'RM mini 3')
         ('host :', ('192.168.1.112', 80))
         ('model :', 'RM pro+')
         ('host :', ('192.168.1.108', 80))
         ('model :', 'RM mini 3')
         ('host :', ('192.168.1.112', 80))
         ('model :', 'RM pro+')
         ('host :', ('192.168.1.108', 80))
         ('model :', 'RM mini 3')
         ('host :', ('192.168.1.112', 80))
         ('model :', 'RM pro+')
         ('host :', ('192.168.1.108', 80))
         ('model :', 'RM mini 3')
         ('host :', ('192.168.1.112', 80))
         ('model :', 'RM pro+')
         ('host :', ('192.168.1.108', 80))`

I expected to get only 2 instances for each of my device .. now I have to basicly have to filter stuff ?

Can you explain this ? @felipediel 

Just wondering what is going on here

felipediel, 2020-10-31:
After [this PR](https://github.com/mjg59/python-broadlink/pull/438), we broadcast several hello messages and ignore repeated responses. We are using [this filter](https://github.com/felipediel/python-broadlink/blob/06888c9bec5fd1209258b4b2118b1213a04501b5/broadlink/device.py#L89-L90).

steffenu, 2020-11-01:
> After [this PR](https://github.com/mjg59/python-broadlink/pull/438), we broadcast several hello messages and ignore repeated responses. We are using [this filter](https://github.com/felipediel/python-broadlink/blob/06888c9bec5fd1209258b4b2118b1213a04501b5/broadlink/device.py#L89-L90).

Mhh okay ... I it seems in py3 ithat filter works .... but in py2 it doesnt ... do you know why that could be ? 
Guess I need to rewrite it .. but I dont see whats wrong

steffenu, 2020-11-01:
Did a hotfix now that works for me ... but I still dont get why the previous code is not working in py2 

`                # Preventing duplicates in discovery # py2
                wtf = False
                for x in discovered:
                    print(x[0])
                    if host in x:
                        #print("WTF")
                        wtf = True

                if not wtf:
                    discovered.append((host, mac, devtype))
                    print("append", host)
                    name = response[0x40:].split(b'\x00')[0].decode('utf-8')
                    is_locked = bool(response[-1])
                    yield devtype, host, mac, name, is_locked`

felipediel, 2020-11-01:
You need to change [this line](https://github.com/felipediel/python-broadlink/blob/06888c9bec5fd1209258b4b2118b1213a04501b5/broadlink/device.py#L88) to support Python 2.

steffenu, 2020-11-01:
> You need to change [this line](https://github.com/felipediel/python-broadlink/blob/06888c9bec5fd1209258b4b2118b1213a04501b5/broadlink/device.py#L88) to support Python 2. You can use this:
> 
> ```
> mac = response[0x3f:0x39:-1]
> ```

Okay that works ... The explanation would probably take longer I suspect ... and im not really interisted .... so im fine leaving it at that. 
Thx again... 

The road is clear now ;)... I wish you good health and fortune

closed 2020-11-01 by steffenu

felipediel, 2020-11-01:
Thanks! I wish you the best too. Good luck with the plugin!

steffenu, 2020-11-06:
> Thanks! I wish you the best too. Good luck with the plugin!

tiny issue .. i want to print the mac adress of the device ... but I cant decode the bytes object

```
    discovered = broadlink.discover(7)
    print(discovered[0].mac.decode())  << what is the correct codec ?
```

`UnicodeDecodeError: 'utf-8' codec can't decode byte 0x96 in position 5: invalid start byte`

felipediel, 2020-11-06:
```python3
from binascii import hexlify
mac_hex = hexlify(mac_bytes).decode("utf8")

steffenu, 2020-11-06:
> ```python
> from binascii import hexlify
> mac_hex = hexlify(mac_bytes).decode("utf8")
> ```

okay that kinda works ... expect the ":" between the numbers are missing rofl ...

I feel stupid .... but now I ended up using regex instead ... because reformating this is a bitch for me xD

780f77636696 <<< your solution

`x = re.search("(\d+|\w+):(\d+|\w+):(\d+|\w+):(\d+|\w+):(\d+|\w+):(\d+|\w+)(\d+|\w+)", str(data))`
78:0f:77:63:66:96 <<<

first world problems lol ...

felipediel, 2020-11-06:
You can use this:
```python3
':'.join(format(b, '02x') for b in mac)
```

steffenu, 2020-11-06:
> You can use this:
> 
> ```python
> ':'.join(format(b, '02x') for b in mac)
> ```

that does the trick ....  wouldn't have been able to come up with this ;)  ... don't know the bytes stuff ... .. dont know the format stuff ... and haven't used .join either that way :D ....

thx