mjg59/python-broadlink #473: SP4L program

lakeandhubo, 2020-11-11:
#!/usr/bin/python
# -*- coding: UTF-8 -*-

import broadlink
import sys
import time
import json

SwitchInfo={
    "ipLast_0":"155","mac_0":"24dfa7eea735",
}

def switchControl(switchIndex,powerState):
    global SwitchInfo

    device_ip="192.168.1."+SwitchInfo["ipLast_"+str(switchIndex)]
    device_mac = SwitchInfo["mac_"+str(switchIndex)]

    device_port=80
    device_type=0x7d11

    '''SP4L:以下三个码都可以
        0x7579: (sp4l, "SP4L-EU", "Broadlink"),
        0x7583: (sp4l, "SP mini 3", "Broadlink"), 
        0x7d11: (sp4l, "SP mini 3", "Broadlink"),

    '''

    device = broadlink.sp4l(host=(device_ip,device_port), mac=bytearray.fromhex(device_mac),devtype=device_type)

    device.auth()

    if powerState=="off":
        device.set_nightlight(True)
        device.set_power(False)

        #print("\t\t",end="")
        #print(device.check_power())
        #print(device.check_nightlight())
        #print(device.get_state())
    else:
        #print("->on")
        device.set_nightlight(False)
        device.set_power(True)
        
        #print(device.check_power())
        #print(device.check_nightlight())
        #print(device.get_state())

if __name__ == "__main__":

    for i in range(100):
        print("%d:on"%(i))
        switchControl(0,"on")
        time.sleep(10)

        print("\t\t%d:off"%(i))
        switchControl(0,"off")
        time.sleep(10)

    pass

lakeandhubo, 2020-11-11:
test SP4L OK

felipediel, 2021-02-17:
Thank you! These devices are on the master branch now. Closing.

closed 2021-02-17 by felipediel