mjg59/python-broadlink #477: RM4 Mini causes every esp8266 device to become unavailable as soon as it is plugged in

iankaufmann, 2020-11-21:
Sorry if this isn't the place for this but I can't think of anywhere else someone who might have seen this behavior would hang out.

I bought two RM4 Minis back in May and have been using them after setting them up with `python-broadlink` through Home Assistant with zero trouble at all.

I bought a third this week, set it up the same way with `python-broadlink` and as soon as I did, it somehow killed every esp8266 based device on my IOT VLAN (same that the RM4 is connected to) immediately.  (It appears to have no impact on any device connected to the "main" network)

I saw the other thread about how the newer firmware isn't happy when it can't reach home and reboots every 5 minutes.  The timing on this isn't quite every 5 minutes, but if I leave the RM4 connected (and it does actually function via Home Assistant), all ESP devices will go unavailable for 5-10 minutes, then come back for anywhere from 5-45 minutes and then go down again.

I am 100% certain that it stops as soon as I unplug the RM4.  

What in the world is it doing on my network?  It's like it is broadcasting something that is "tricking" the ESPs into talking to it instead of my router.  What's weirder is that Unifi still shows the devices as connected to the Wifi, but all of their MQTT activity stops so Home Assistant says they're unavailable.

I factory reset and tried setup again a few times... and on the last try, set it up through the official app.... and it still does it even with that.  The moment I plug it in, the whole network goes crazy and I can't make any sense of it.

What's worse is that I think since I had the other two (fully working all year) plugged in still when I ran `python-broadlink`, they now do the same thing the moment they're plugged in.

They might as well be advertising these as WiFi jammers...

Does this ring a bell for anybody at all or am I just losing my mind?

KTibow, 2020-11-21:
Does it work at all with the IR? Does it work with the official app?

iankaufmann, 2020-11-21:
Yep, it works in the app and when I only setup via `python-broadlink`...  that is, I can send commands to it via Home Assistant, or by clicking the buttons in the app.  It's currently connected to the app since that is the last thing I have tried.  

However, even the original two (that were connected and working properly at the time I first ran setup in `python-broadlink` yesterday) which have never been paired with the app are now doing it.

I'm guessing that since `discover()` can see all of the eligible devices on the network that it applied my `setup()` to all of them and that is why they "changed"?  (Wondering if they'd still work properly if I'd disconnected them prior to setting up the 3rd one)

Quite literally everything was working perfectly all year until the moment I typed:
```
import broadlink

broadlink.setup('myssid', 'mynetworkpass', 3)
```
... and hit 'enter'.

It sounds unbelievable but it's true... and I spent my entire day yesterday trying to get to the bottom of it.  I honestly didn't even realize that these were causing it at first and thought it was obviously a WiFi issue... even rebuilt my whole Unifi network.  The only thing that fixes it is to make sure none of the RM4s are on the network.  It's not even necessarily a VLAN issue because while it is happening I can still communicate with wired devices on the IoT subnet, and a couple WiFi cameras.  But no ESP devices... smart plugs, D1 minis, NodeMCUs, and my doorbell camera (which I'm not sure what chip it uses but that thing is the first thing to cut out during this)

iankaufmann, 2020-11-21:
So I reset all of them again today and it's still true that if I put them on the IoT VLAN, the network goes wild.

But... If I put them on the "main" VLAN, all is well.  Doesn't seem to have an impact on any other device.

It must be sending out some crazy packets that are over my head to debug that is causing all of the other devices to freak out.  It's just not making much sense to me that it really does seem to only be ESP8266 based devices that it happens to.  

So far, I'm back up and running again.  I just need to make some specific firewall rules to prevent these from talking instead of just having them on the IoT VLAN.

Feel free to close this as I know it's not an issue with `python-broadlink`, but this information may be useful or interesting to somebody.  I lost way too many hours of my life on this.

felipediel, 2021-10-18:
Perhaps something related to multicast? It seems to me that the problem is with the software running on ESPs, not the Broadlink devices. Thanks for informing about this. Closing, not something we can fix here.