mjg59/python-broadlink #484: RM mini 3 doesn't get discovered

Rania-M, 2020-11-27:
Hi,
i'm trying to control an RM mini 3 device via the broadlink api, the device is easily recognized by the mobile app, but folowing the readme steps doesn't get it discovered on the api.
**Steps** : reset the device , set the wifi in the api, launch the discover
**Result** : it returns an empty list

_PS_: i did turn off the device lock on the mobile app
Thank you

KTibow, 2020-11-27:
Don't use the mobile app at all.

Rania-M, 2020-11-27:
> Don't use the mobile app at all.

Thank you for your response,
I didnt usé the mobile app at first and it wasn't working (always empty list)

KTibow, 2020-11-28:
If you install Fing on your phone, and connect it to the same network as your PC, can you find the device?

KTibow, 2020-11-28:
Never mind, it hasn't been merged yet: https://github.com/mjg59/python-broadlink/pull/479

Rania-M, 2020-12-01:
> Never mind, it hasn't been merged yet: #479

thank you, so the device is not supported yet ?

felipediel, 2020-12-01:
@Rania-M If you have multiple network interfaces (eg WiFi, Ethernet), you can try to reach the device by specifying a local_ip_address or discover_ip_address in discover().

Rania-M, 2020-12-02:
> @Rania-M If you have multiple network interfaces (eg WiFi, Ethernet), you can try to reach the device by specifying a local_ip_address or discover_ip_address in discover(). If you don't give these arguments, the OS will bind the socket to the first available interface and the message may end up being transmitted to the wrong network. You can also try direct discovery:
> 
> ```python
> device = broadlink.hello("192.168.0.12")  # Example IP address
> ```

Thank you ! i did this but still didn't work, i, however, changed the computer as i was only tring with a virtual machine first, and now it's working :+1: 
I can now detect the RM mini 3, the only problem is that it is assigned to the RM4 class, and it seems ans i can't find a way to send data to the device ( o function found there)
I can try to use the RM fsend data function , but i was wondering why this has not been impelmented.
Thank you in advance !

felipediel, 2020-12-02:
Some devices use the RM4 class (eg RM mini 3 0x5f36). This class has the same interface of RM2 and all the methods are implemented.

Rania-M, 2020-12-02:
> Some devices use the RM4 class (eg RM mini 3 0x5f36). This class has the same interface of RM2 and all the methods are implemented.

Thank you, i finally used the cli files, i will close thi subjectif if no objections 
Thanks to all 😊

closed 2020-12-02 by Rania-M