mjg59/python-broadlink #492: Incorrect get_energy() values for SP3S

Stormhand, 2020-12-09:
Hello,
I have two SP3S devices with the last firmware version 10006 where the get_energy() method is returning very different values. For the test i started a while loop which prints the payload and the real calculation every 10s and watched the consumption in the app:

payload[0x05] | payload[0x06] | payload[0x07] | calc
------------------- | ------------------ | ------------------- | -------
8 | 102 | 0 | 102.08
104 | 101 | 0 | 102.04
72 | 102 | 0 | 102.72
136 | 100 | 0 | 101.36
146 | 101 | 0 | 102.46
136 | 102 | 0 | 103.36
0 | 102 | 0 | 102.0
72 | 102 | 0 | 102.72
0 | 102 | 0 | 102.0
72 | 102 | 0 | 102.72
50 | 102 | 0 | 102.5
22 | 102 | 0 | 102.22
104 | 101 | 0 | 102.04

During that time the wattage in the app was almost constantly 66.5 with some peaks of 80.

Stormhand, 2020-12-09:
I debugged a bit and found something interesting. Payloads hex(payload[0x06]) and hex(payload[0x05]) are the exact value in the app:
App reading | hex(payload[0x06]) | hex(payload[0x05])
---------------- | -------------------------- | --------------------------
25.68 | 0x25 | 0x68
25.84 | 0x25 | 0x84
25.76 | 0x25 | 0x76
25.76 | 0x25 | 0x76

ill test the payload 0x07 as how its included in the calcs

Stormhand, 2020-12-09:
So the correct value is returned by:
`float("{:x}{:x}.{:x}".format(*payload[0x05:0x08][::-1]))`
because what we need is the numbers without `0x` concatenated.

KTibow, 2020-12-09:
@felipediel Have you seen anything else like this?

Stormhand, 2020-12-09:
What do you mean? I have the device since yesterday so this is my first contact with the code.

KTibow, 2020-12-09:
I was @mentioning a contributor who can often make code changes.

Stormhand, 2020-12-09:
Ah, sorry didn't see the mention. I personally haven't seen such encoding before.

felipediel, 2020-12-09:
It looks strange at first glance, but looking at the data, that's exactly how they're doing it. Thank you, I will fix it.

Stormhand, 2020-12-09:
> It looks strange at first glance, but looking at the data, that's exactly how they're doing it. Thank you, I will fix it.

Thanks for the fix. Yes its one of the strangest encodings ive seen.

Stormhand, 2020-12-10:
There are edge cases which means we need better parsing:
` could not convert string to float: '2e21.aa'`

felipediel, 2020-12-11:
What is the power (W) in the app when you get these values? How often do they appear?

felipediel, 2020-12-13:
We could catch the exception and return 0, how about?

Stormhand, 2020-12-13:
Well it happened to return it when its not OFF. Also i found in some nodejs libs that they are checking for those incorrect values too(and throwing an exception) so its a common behavior. I'm not completely sure what to return: 0, None or Exception? May be 0 will require less checks and coding for the users as it wont mess with the statistics.

felipediel, 2020-12-14:
If there is a possibility that the value is not zero, better to generate an exception so as not to mess up the state. I will do it. Thank you.

bernardoaraujor, 2020-12-18:
not sure if related to this issue, but my SP3s turns off after a few readings.
is that normal?

felipediel, 2020-12-18:
I don't think so. Did you block internet access? Some devices reboot periodically when they don't receive heartbeats from the cloud.

bernardoaraujor, 2020-12-19:
@felipediel I didn't, the device is connected to a WiFi with regular internet access.

closed 2021-01-11 by felipediel