mjg59/python-broadlink #494: Code for connect to rm3 mini without use of discover function?

pfolcato, 2020-12-12:
Hi,
I have a problem when i try to connect to my broadlink.
discover works perfectly but i have more than one device and with discover i find them all. 
How can I establish a connection only with the ip or mac of my device?
I used this but now it doesn't work with rm3 mini:
```
device = broadlink.rm(host=(<ipaddress>,80), mac=bytearray.fromhex(<macaddress>), devtype = 'rm3')
        device.auth()
        device.host    
```

KTibow, 2020-12-12:
Try broadlink.hello(“192.168.1.x”)

pfolcato, 2020-12-12:
> 
> Try broadlink.hello(“192.168.1.x”)

It works well! thanks

pfolcato, 2020-12-12:
So thi is my simple script used in Domoticz for put on or off a device (no error control)
```
#!/usr/bin/python
import broadlink
import os
import sys
device = ""
brohost = "10.0.0.41" #ip of broadlink device
sendCommand = "260050020F0D0E0F0D100D0F0E0F0D00033F713A0F2A0E0F0D100D0F0E2B0E0F0E0F0D0F100D0D2C0D100D2C0D2C0D100D2C0F2A0D2C0E2C0D2C0D0F0E0F0D2C0E0F0D0F0E0F0D100D0F0E0F0D100D0F0D100D0F0F2A0E0F0D2C0F0E0D100D0F0D2C0F2A0E0F0D100D0F0E0F0D100D0F0F0E0D0F0E0F0D100D0F0E0F0F0D100D0D100D0F0E2B0E2C0D2C0D0F0E2C0D0F10290E2C0D000479713B0F2A0D100D0F0D100F2A0D100F0D0E0F0D0F102A0D0F0E2B0E2C0F0D0E2B0E2C0D2C0D2C0D2C0D100D0F102A0D0F0E0F0D0F0E0F0F0E0D0F0E0F0D100F0D0D100D0F0E2B0E0F0F0E0D0F0E0F0F2A0F0E0F0D0E0F0D100D0F0D100D0F0E0F0D100D0F0E0F0D0F100D0D100D0F0E0F0D0F0E0F0D100D2C0D0F0E2C0D0F0E2B0E0F0D000479723A0D2C0E0F0D100D0F0E2B100D0D100D0F0E0F0D2C0D100D2C0F2A0D100D2C0D2C0D2C0D2C0E2C0D0F0F0E0D2C0D100D0F0E0F0D100D0F0D100D0F0E0F0D100D0F0E0F0F0D0E0F0D100D0F0E0F0D0F0E0F0D100D0F0E0F0D2C0D100D0F0E0F0D0F0E0F0D2C0D100D0F0E2C0D2C0D0F0E0F0D100D0F0E0F0D100D0F0D100D0F0E0F0D2C0D2C0E2B0E2C0D2C0D2C0D100F2A0D100D0F0E0F0D0F0E0F0D100D0F0F0E0D0F0E0F0D100D0F0F0E0D0F0E0F0D100D0F0E2B0E2C0D0F0E0F0F0D0E0F0D100D0F0E0F0D100D0F0F0E0D2C0D2C0D100D0F0E0F0D100D0F0D100F0D0E0F0D100D0F0E0F0D0F0E0F0F0E0D0F0E0F0D0F0E2C0D0F0E0F0D0F0E0F0D100D2C0D2C0D100D0F0E0F0D0F0E0F0D100D0F0E2B0E0F0D100D0F100D0D100D0F0D100D0F0E0F0D2C0D100D0F0E2C0D2C0D0F0E2C0D000D050000000000000000"  #this code is for put on a daikin air conditioner
sendCommand = bytes.fromhex(sendCommand) #convert to hex the command
device = broadlink.hello(brohost)   #connection to broadlink device and save parameter to dictionary varaible
device.auth()  #retrive the auth key from the device
device.send_data(sendCommand)   #send data command
```

closed 2020-12-12 by pfolcato