mjg59/python-broadlink #501: broadlink.exceptions.AuthenticationError: [Errno -1] Authentication failed

terrycarlin, 2021-01-09:
$ broadlink_discovery

Running latest code.
Found RM2 Pro+ ok 
Device that failed = RM4 Pro

Discovering...
###########################################
RM2
#broadlink_cli --type 0x279d --host 192.168.0.123 --mac 34ea346fe911
Device file data (to be used with --device @filename in broadlink_cli) : 
0x279d 192.168.0.123 34ea346fe911
temperature = 0.0

Traceback (most recent call last):
  File "./broadlink_discovery", line 17, in <module>
    if device.auth():
  File "/usr/local/lib/python3.8/dist-packages/broadlink/device.py", line 198, in auth
    check_error(response[0x22:0x24])
  File "/usr/local/lib/python3.8/dist-packages/broadlink/exceptions.py", line 137, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthenticationError: [Errno -1] Authentication failed

KTibow, 2021-01-09:
Does it always fail?

terrycarlin, 2021-01-09:
Yes, I have run broadcom_discover many times with the same result. 

I put in some debug code to try to see what’s going on. So far this is what I see:

device_type: RM4
devtype: 0x61a2
device_host: 192.168.0.160
device_mac: 24dfa7f1032a

If I can be of any help, please let me know. I will put in any other debug code necessary.
The RM4 Pro is brand new. I have configured it in the Broadcom app. My target use is with Home Assistant sending out RF codes to fans and patio shades.

I have a RM2 Pro+ that works just fine. This code helped me set up a number of IR and RF devices.

KTibow, 2021-01-10:
Did you unlock it in the Broadlink app?

terrycarlin, 2021-01-10:
That was it!
Thanks!
I didn’t know about that feature.
Now I do.

Terry Carlin

KTibow, 2021-01-10:
Can you close this issue then?

closed 2021-01-10 by terrycarlin

r3a1d3a1, 2021-12-21:
@felipediel Please mention in the README that the device needs to be unlocked through the app. It took me a while to figure it out (as it wasn't authenticating.)

pyrite357, 2023-07-06:
> @felipediel Please mention in the README that the device needs to be unlocked through the app. It took me a while to figure it out (as it wasn't authenticating.)

i second this if it hasn't been. A better idea though might be to include some verbiage about this in the error message instead of just "Authentication failed", you could clue the user in on what it might be e.g. "Authentication failed (is the device unlocked in the Broadlink app?)"